openwrt/openwrt #15013: UNII-2 and UNII-2 EXT channels are not working properlly (something regarding its center frequency)

phpaiva, 2024-03-29:
### Describe the bug

In OpenWrt 23.05.2 I have found this by updating my Archer AX23 v1 to this version. Previously (I do not remember the version, was 22.X or 23.X) **it was working well in Channel 52 / 80 MHz radiating two (2) BSS , but since I upgraded they stoped to work**.

I found that **with UNII-3's and UNII-1's channels the issue is not happening**, and then I saw that **the center frequency is not being measured as the correct one** just in the case of UNII-2 and UNII-2 EXT:

[image]

Since, for channel 52 should be 5260 MHz:

[image]

**All the channels from 52 to 144 shows more than 6 GHz center frequency**, which is crearly a mistake.

The issue also occurs with [another device which I am porting to OpenWRT](https://forum.openwrt.org/t/add-support-to-mediatek-mt7621-reference-board-sold-by-ruxtek/191074/1) which is based almost in the same chipset:

MT7621AT (SoC)
MT7975DN + MT7905DAN (WiFi)
MT7530 (switch)

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer AX23 v1

### Image kind

Official downloaded image

### Steps to reproduce

Just set the channel to any UNII-2 or UNII-2 EXT. Check its reported frequency in LUCI. Check that no SSID is being irradiated.

### Actual behaviour

The WiFi does not work in UNII-2 or UNII-2 EXT

### Expected behaviour

The WiFi should work in UNII-2 and UNII-2 EXT

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-30:
Attach `iw reg get` and `iw list` notably channel list from later.

phpaiva, 2024-03-30:
First `iw reg get`

`root@OpenWrt:~# iw reg get
global
country AR: DFS-FCC
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
`

Well, looking at this here the values seems to be OK, If we omit the "detail" that the country code AR (Argentina) is wrong, since I can assure you as an Argentinian who works in telecommunications (who I am) that ... here UNII-1 is free up to 20 dBm and UNII-2 (not EXT) is also free up to 24 dBm .

Now `iw list` also does not seem to show any issue:

`root@OpenWrt:~# iw list
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 4
        max scan IEs length: 2190 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 2:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-15
                VHT Capabilities (0x339939b1):
                        Max MPDU length: 7991
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        TX STBC
                        SU Beamformer
                        SU Beamformee
                        MU Beamformer
                        MU Beamformee
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
[... 469 lines cut ...]
                * [ RRM ]: RRM
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ FILS_DISCOVERY ]: FILS discovery frame transmission support
                * [ UNSOL_BCAST_PROBE_RESP ]: unsolicated broadcast probe response transmission support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
                * [ BSS_COLOR ]: BSS coloring support
`

Thank you very much.

phpaiva, 2024-03-30:
Well I noticed that with "AR" country code the issue of the "+6 GHz" seems not to be happening. This happens with "world" coutry code, and also I tested India "IN" where the issue happens also. With country code "World" (00), we have:

```
root@OpenWrt:~# iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

```

```
root@OpenWrt:~# iw reg get
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

root@OpenWrt:~# ^C

root@OpenWrt:~# iw list
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 4
        max scan IEs length: 2190 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 2:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
[... 520 lines cut ...]
                        * 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
                        * 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
                        * 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
                        * 5500 MHz [100] (20.0 dBm) (no IR, radar detection)
                        * 5520 MHz [104] (20.0 dBm) (no IR, radar detection)
                        * 5540 MHz [108] (20.0 dBm) (no IR, radar detection)
                        * 5560 MHz [112] (20.0 dBm) (no IR, radar detection)
                        * 5580 MHz [116] (20.0 dBm) (no IR, radar detection)
                        * 5600 MHz [120] (20.0 dBm) (no IR, radar detection)
                        * 5620 MHz [124] (20.0 dBm) (no IR, radar detection)
                        * 5640 MHz [128] (20.0 dBm) (no IR, radar detection)
                        * 5660 MHz [132] (20.0 dBm) (no IR, radar detection)
                        * 5680 MHz [136] (20.0 dBm) (no IR, radar detection)
                        * 5700 MHz [140] (20.0 dBm) (no IR, radar detection)
                        * 5720 MHz [144] (20.0 dBm) (no IR, radar detection)
                        * 5745 MHz [149] (20.0 dBm)
                        * 5765 MHz [153] (20.0 dBm)
                        * 5785 MHz [157] (20.0 dBm)
                        * 5805 MHz [161] (20.0 dBm)
                        * 5825 MHz [165] (20.0 dBm)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)
                        * 5885 MHz [177] (disabled)

```

brada4, 2024-03-30:
LuCI shows same as native iwinfo.
Sometimes it is beacon channel, but try to catch oob case between them.
Gladly no out of spec transmissions.

castillofrancodamian, 2024-03-30:
@brada4 They are not very different from those in other countries, except for the channels that have to do with DFS.

brada4, 2024-03-30:
If you could catch the situation when iwinfo and/or luci shows wildly out of line parameters. Wifi-scripts complete center channel, but some cards report back actual beacon channel instead, similar to upper/lower with wifi-n.

phpaiva, 2024-03-31:
Guys, keep in mind that the incorrect central frequency displayed in LUCI is just a peculiarity that I noted that came hand in hand with the non-functioning of the mentioned channels. The main problem is that the UNII-2 and UNII-2 EXT channels are not working in several country codes, starting with "World".

To add the iwinfo dump and confirm what you are saying:

```
root@OpenWrt:~# iwinfo
phy0-ap0  ESSID: "Pablomagno"
          Access Point: 54:AF:97:C5:C0:84
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HE20
          Center Channel 1: 1 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -91 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "Pablomagno"
          Access Point: 54:AF:97:C5:C0:83
          Mode: Master  Channel: 52 (6.210 GHz)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 24 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

```

You can find there `Channel: 52 (6.210 GHz)` . **In this situation the no BSS is able to operate (hostapd_cli report them as `state=DISABLED`), no beacons are being irradiated and the devices previously associated loose its connection**.

brada4, 2024-03-31:
check logread, if you have some radar near triggering dfs often might be worth adding like 
```
option channels '36 52 100'
```
to have fallback if radar kicks you out.

https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/tree/db.txt#n92

phpaiva, 2024-03-31:
> check logread, if you have some radar near triggering dfs often might be worth adding like
> 
> ```
> option channels '36 52 100'
> ```
> 
> to have fallback if radar kicks you out.
> 
> https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/tree/db.txt#n92

India country code has no DFS in any channel, and the issue still occurs, I could not set any channel between 52 and 144. AR has DFS for UNII-2 and sets channel 52 without any issue. This seems to be related just for some country codes, while "US", "AR", "CN" for example do not show the issue, this happens with "00", "IN", "IM", "IO", "IQ",  country codes.

brada4, 2024-03-31:
00 does not count at all.
Normally you get jut one regdom change per poweron, subsequent changes just go towards adding restrictions.

Since mid-channel has to be set for fixed channel you cannot have full AUTO-BW with fixed channel.
https://github.com/openwrt/openwrt/blob/29cca6cfee4bc732f9bf7eca0cae9e56cd3c0f2f/package/network/config/wifi-scripts/files/lib/netifd/wireless/mac80211.sh#L240