openwrt/openwrt #15023: vim-full: E1187: Failed to source defaults.vim

532910, 2024-03-30:
### Describe the bug

```
root@OpenWrt:/# opkg update
root@OpenWrt:/# opkg install vim-full
root@OpenWrt:/# reboot
root@OpenWrt:/# vim
E1187: Failed to source defaults.vim
Press ENTER or type command to continue
```

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

various

### Device

various

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-30:
How comes your opkg produces no output?

CapitalF, 2024-04-01:
> How comes your opkg produces no output?

He doesn't need to provide output. It's not relevant.

This bug is confirmed.

brada4, 2024-04-01:
Just that the line would clearly show that issue is in ../packages and no change in this repo can correct it.

532910, 2024-04-08:
dup of https://github.com/openwrt/packages/issues/20203

closed 2024-04-08 by 532910