openwrt/openwrt #15028: filogic: wwan doesn't get IP on boot when adding a new wireless AP on same radio.

hashcan, 2024-03-31:
### Describe the bug

Everything works perfect after connecting to another AP to get internet access from WWAN until you reboot the device. 
Then you need yo login to luci and restart the wwan network interface. 

### OpenWrt version

r25717-d7d94a8d91

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Steps to reproduce

Create a wwan by Luci > Wireless > Scan > Join Network > Check "Replace wireless configuration" option and type down the passphrase > submit, save and apply.
Add a new AP wireless network on the same radio.  assign it to "lan" network .

### Actual behaviour

After following the Steps to reproduce, WWAN successfully connects to upstream wifi router and automatically gets IP address from the upstream DHCP server. until you reboot the router and then this happens. 
[image]
[image]

A manual wwan network restart in luci will solve the issue : 

[image]
[image]

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-31:
Probably not the best place to clone MAC address....

hashcan, 2024-04-01:
> Probably not the best place to clone MAC address....

I changed the client's wireless network MAC to random : 
[image]

The client interface looks like this after reboot : 
[image]

[image]

brada4, 2024-04-01:
Probably you need to use rnd mac in AP side to not consume uplink's dhcp pool in no time.
Log should say that - basically set any channel on ap , not auto, since it is forced by supplicant on wwan.

hashcan, 2024-04-01:
> Probably you need to use rnd mac in AP side to not consume uplink's dhcp pool in no time. Log should say that - basically set any channel on ap , not auto, since it is forced by supplicant on wwan.

The current bug is that wwan cannot obtain IP address at boot time. all will work good if I restart the wwan network manually in Luci after boot.

brada4, 2024-04-02:
config/network and config/wireless would be nice.

hashcan, 2024-04-02:
> config/network and config/wireless would be nice.
```bash
root@OpenWrt:~# cat /etc/config/network 

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd71:d3fd:2e92::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.31.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config device
	option name 'wan'
	option macaddr 'a4:XX:XX:XX:XX:YX'

config interface 'wan'
	option device 'wan'
	option proto 'dhcp'

config interface 'wan6'
	option device 'wan'
	option proto 'dhcpv6'

config interface 'wwan'
	option proto 'dhcp'

root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi'
	option channel '1'
	option band '2g'
	option htmode 'HE20'
	option cell_density '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi+1'
	option channel '36'
	option band '5g'
	option htmode 'HE80'
	option disabled '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWRT'
	option encryption 'sae'
	option disabled '0'
	option key '88888888'

config wifi-iface 'wifinet2'
	option device 'radio0'
	option mode 'sta'
	option network 'wwan'
	option ssid 'upstairs'
	option encryption 'sae'
	option key '123456789'
	option macaddr 'random'

config wifi-iface 'wifinet3'
	option device 'radio0'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'sae'
	option key '88888888'

root@OpenWrt:~# 

```

brada4, 2024-04-02:
Config looks generic, like made from luci.
What is in `logread -e upstairs:mac` and so on?
I did aporximately same you did, just that wireless file has sra and ap in reverse order.

hashcan, 2024-04-02:
> What is in `logread -e upstairs:mac` and so on?

at boot I have : 
```
Sun Mar 31 20:16:38 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: SME: Trying to authenticate with e8:xx:xx:xx:xx:xx (SSID='upstairs' freq=2412 MHz)
Sun Mar 31 20:16:38 2024 kern.info kernel: [   20.681701] phy0-sta0: authenticate with e8:xx:xx:xx:xx:xx
Sun Mar 31 20:16:38 2024 kern.info kernel: [   20.689532] phy0-sta0: send auth to e8:xx:xx:xx:xx:xx (try 1/3)
Sun Mar 31 20:16:38 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: SME: Trying to authenticate with e8:xx:xx:xx:xx:xx (SSID='upstairs' freq=2412 MHz)
Sun Mar 31 20:16:38 2024 kern.info kernel: [   20.787288] phy0-sta0: authenticate with e8:xx:xx:xx:xx:xx
Sun Mar 31 20:16:38 2024 kern.info kernel: [   20.792855] phy0-sta0: send auth to e8:xx:xx:xx:xx:xx (try 1/3)
Sun Mar 31 20:16:38 2024 kern.info kernel: [   20.799589] phy0-sta0: authenticated
Sun Mar 31 20:16:38 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: Trying to associate with e8:xx:xx:xx:xx:xx (SSID='upstairs' freq=2412 MHz)
Sun Mar 31 20:16:38 2024 kern.info kernel: [   20.811849] phy0-sta0: associate with e8:xx:xx:xx:xx:xx (try 1/3)
Sun Mar 31 20:16:38 2024 kern.info kernel: [   20.819620] phy0-sta0: RX AssocResp from e8:xx:xx:xx:xx:xx (capab=0x431 status=0 aid=3)
Sun Mar 31 20:16:38 2024 kern.info kernel: [   20.829673] phy0-sta0: associated
Sun Mar 31 20:16:38 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: Associated with e8:xx:xx:xx:xx:xx
Sun Mar 31 20:16:38 2024 daemon.notice netifd: Network device 'phy0-sta0' link is up
Sun Mar 31 20:16:38 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Sun Mar 31 20:16:38 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: Unknown event 37
Sun Mar 31 20:16:38 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: WPA: Key negotiation completed with e8:xx:xx:xx:xx:xx [PTK=CCMP GTK=CCMP]
Sun Mar 31 20:16:38 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: CTRL-EVENT-CONNECTED - Connection to e8:xx:xx:xx:xx:xx completed [id=1 id_str=]
Sun Mar 31 20:16:38 2024 kern.info kernel: [   20.942457] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-sta0: link becomes ready
Sun Mar 31 20:16:38 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: Unknown event 37
Sun Mar 31 20:16:38 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names

```
and if I restart wwan from luci : 

```
Sun Mar 31 20:20:00 2024 daemon.notice netifd: radio0 (3205): sh: out of range
Sun Mar 31 20:20:00 2024 daemon.notice wpa_supplicant[1559]: Set new config for phy phy0
Sun Mar 31 20:20:00 2024 daemon.notice wpa_supplicant[1559]: Update configuration for interface phy0-sta0
Sun Mar 31 20:20:00 2024 daemon.notice netifd: Network device 'phy0-sta0' link is down
Sun Mar 31 20:20:00 2024 kern.info kernel: [  222.828757] phy0-sta0: deauthenticating from e8:xx:xx:xx:xx:xx by local choice (Reason: 3=DEAUTH_LEAVING)
Sun Mar 31 20:20:00 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: CTRL-EVENT-DISCONNECTED bssid=e8:xx:xx:xx:xx:xx reason=3 locally_generated=1
Sun Mar 31 20:20:00 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: CTRL-EVENT-DSCP-POLICY clear_all
Sun Mar 31 20:20:00 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: CTRL-EVENT-DSCP-POLICY clear_all
Sun Mar 31 20:20:00 2024 daemon.notice wpa_supplicant[1559]: nl80211: deinit ifname=phy0-sta0 disabled_11b_rates=0
Sun Mar 31 20:20:00 2024 daemon.err wpa_supplicant[1559]: rmdir[ctrl_interface=/var/run/wpa_supplicant]: Permission denied
Sun Mar 31 20:20:01 2024 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Sun Mar 31 20:20:01 2024 daemon.notice hostapd: Reloaded settings for phy phy0
Sun Mar 31 20:20:01 2024 daemon.notice wpa_supplicant[1559]: Set new config for phy phy0
Sun Mar 31 20:20:01 2024 daemon.notice netifd: Wireless device 'radio0' is now up
Sun Mar 31 20:20:01 2024 daemon.notice netifd: Interface 'wwan' is enabled
Sun Mar 31 20:20:04 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: SME: Trying to authenticate with e8:xx:xx:xx:xx:xx (SSID='upstairs' freq=2412 MHz)
Sun Mar 31 20:20:04 2024 kern.info kernel: [  227.039158] phy0-sta0: authenticate with e8:xx:xx:xx:xx:xx
Sun Mar 31 20:20:04 2024 kern.info kernel: [  227.046823] phy0-sta0: send auth to e8:xx:xx:xx:xx:xx (try 1/3)
Sun Mar 31 20:20:04 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: SME: Trying to authenticate with e8:xx:xx:xx:xx:xx (SSID='upstairs' freq=2412 MHz)
Sun Mar 31 20:20:04 2024 kern.info kernel: [  227.109542] phy0-sta0: authenticate with e8:xx:xx:xx:xx:xx
Sun Mar 31 20:20:04 2024 kern.info kernel: [  227.115054] phy0-sta0: send auth to e8:xx:xx:xx:xx:xx (try 1/3)
Sun Mar 31 20:20:04 2024 daemon.notice wpa_supplicant[1559]: phy0-sta0: Trying to associate with e8:xx:xx:xx:xx:xx (SSID='upstairs' freq=2412 MHz)
Sun Mar 31 20:20:04 2024 kern.info kernel: [  227.122200] phy0-sta0: authenticated
Sun Mar 31 20:20:04 2024 kern.info kernel: [  227.131809] phy0-sta0: associate with e8:xx:xx:xx:xx:xx (try 1/3)
Sun Mar 31 20:20:04 2024 kern.info kernel: [  227.139485] phy0-sta0: RX AssocResp from e8:xx:xx:xx:xx:xx (capab=0x431 status=0 aid=3)
Sun Mar 31 20:20:04 2024 kern.info kernel: [  227.148585] phy0-sta0: associated
Sun Mar 31 20:20:04 2024 daemon.notice netifd: Network device 'phy0-sta0' link is up
Sun Mar 31 20:20:04 2024 daemon.notice netifd: Interface 'wwan' has link connectivity
Sun Mar 31 20:20:04 2024 daemon.notice netifd: Interface 'wwan' is setting up now
[... cut ...]
Sun Mar 31 20:20:08 2024 daemon.notice netifd: wwan (3328): udhcpc: broadcasting select for 192.168.111.155, server 192.168.111.1
Sun Mar 31 20:20:08 2024 daemon.notice netifd: wwan (3328): udhcpc: lease of 192.168.111.155 obtained from 192.168.111.1, lease time 43200
Sun Mar 31 20:20:08 2024 daemon.notice netifd: Interface 'wwan' is now up
Sun Mar 31 20:20:08 2024 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sun Mar 31 20:20:08 2024 daemon.info dnsmasq[1]: using nameserver 192.168.111.1#53
```

brada4, 2024-04-02:
Permission denied - ujail, edit /etc/init.d/wpad and change /sbin/ujail to like /sbin/nonexistant, it will run wifi daemons withour sandbox still under user not root. It is ought to be cosmetic, but check

event 37
sh out of range

i do not have these. Your wifi driver is very common, and really nothing in config/network that looks like a range or even counter. 

no indication it needs country code, but polite to have one per radio.

btw logread is superset of dmesg

hashcan, 2024-04-02:
> Permission denied - ujail, edit /etc/init.d/wpad and change /sbin/ujail to like /sbin/nonexistant, it will run wifi daemons withour sandbox still under user not root.

```bash

root@OpenWrt:~# grep nonex /etc/init.d/wpad 
		[ -x /sbin/nonexistant -a -e /etc/capabilities/wpad.json ] && {
root@OpenWrt:~# 

```

<details><summary>logread</summary>

```
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Sun Mar 31 21:49:24 2024 kern.notice kernel: [    0.000000] Linux version 6.1.82 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.2.0 r25717-d7d94a8d91) 13.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sat Mar 30 00:04:17 2024
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Machine model: Xiaomi Mi Router AX3000T
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Zone ranges:
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000004fffffff]
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000]   DMA32    empty
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000]   Normal   empty
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Early memory node ranges
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043030000-0x0000000047c7ffff]
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000047c80000-0x0000000047ffffff]
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000048000000-0x000000004fffffff]
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000004fffffff]
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.2
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] percpu: Embedded 18 pages/cpu s33896 r8192 d31640 u73728
Sun Mar 31 21:49:24 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s33896 r8192 d31640 u73728 alloc=18*4096
Sun Mar 31 21:49:24 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] CPU features: detected: GIC system register CPU interface
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] alternatives: applying boot alternatives
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64512
Sun Mar 31 21:49:24 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n1 loglevel=8 swiotlb=512 rootfstype=squashfs firmware=0 mtd=ubi uart_en=1
Sun Mar 31 21:49:24 2024 kern.notice kernel: [    0.000000] Unknown kernel command line parameters "firmware=0 mtd=ubi uart_en=1", will be passed to user space.
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Memory: 240276K/262144K available (8512K kernel code, 892K rwdata, 2468K rodata, 448K init, 306K bss, 21868K reserved, 0K cma-reserved)
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] GICv3: 640 SPIs implemented
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] GICv3: 0 Extended SPIs implemented
Sun Mar 31 21:49:24 2024 kern.info kernel: [    0.000000] Root IRQ handler: gic_handle_irq
[... cut ...]
Sun Mar 31 21:50:47 2024 daemon.warn odhcpd[1695]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!

```
</details>

[logread2.log](https://github.com/openwrt/openwrt/files/14834161/logread2.log)

> i do not have these. Your wifi driver is very common, and really nothing in config/network that looks like a range or even counter.

Yes I did everything through luci and tried to keep the config as minimum as possible. 

> event 37
> sh out of range

not so sure what I should do with these.

brada4, 2024-04-02:
I see messages in your logread not present in mine.

hashcan, 2024-04-05:
I redid everything on `OpenWrt SNAPSHOT r25820-46efc77502 / LuCI Master git-24.089.54606~0ecb5ed`  from a vanilla image from https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=mediatek%2Ffilogic&id=xiaomi_mi-router-ax3000t-ubootmod and the issue still exists. like before. 

Version OpenWRT Snapshot r25820-46efc77502

[image]
[image]

<details><summary>opkg list-installed</summary>

```
base-files - 1587~46efc77502
busybox - 1.36.1-r1
ca-bundle - 20240203-r1
cgi-io - 2022.08.10~901b0f04-r21
dnsmasq - 2.90-r2
dropbear - 2022.83-r1
eip197-mini-firmware - 20240220-r1
firewall4 - 2023.11.03~698a5335-r1
fitblk - 1
fstools - 2024.01.22~08cd7083-r1
fwtool - 2019.11.12~8f7fe925-r1
getrandom - 2024.01.24~2c5887cb-r2
hostapd-common - 2024.03.09~695277a5-r1
iw - 5.19-r1
iwinfo - 2024.03.23~79a96150-r1
jansson4 - 2.14-r3
jshn - 2024.03.29~eb9bcb64-r1
jsonfilter - 2024.01.23~594cfa86-r1
kernel - 6.1.82~6c022b6b98b636aec64d494a8c63cde1-r1
kmod-cfg80211 - 6.1.82.6.6.15-r1
kmod-crypto-aead - 6.1.82-r1
kmod-crypto-authenc - 6.1.82-r1
kmod-crypto-ccm - 6.1.82-r1
kmod-crypto-cmac - 6.1.82-r1
kmod-crypto-crc32c - 6.1.82-r1
kmod-crypto-ctr - 6.1.82-r1
kmod-crypto-des - 6.1.82-r1
kmod-crypto-gcm - 6.1.82-r1
kmod-crypto-gf128 - 6.1.82-r1
kmod-crypto-ghash - 6.1.82-r1
kmod-crypto-hash - 6.1.82-r1
kmod-crypto-hmac - 6.1.82-r1
kmod-crypto-hw-safexcel - 6.1.82-r1
kmod-crypto-manager - 6.1.82-r1
kmod-crypto-md5 - 6.1.82-r1
kmod-crypto-null - 6.1.82-r1
kmod-crypto-rng - 6.1.82-r1
kmod-crypto-seqiv - 6.1.82-r1
kmod-crypto-sha1 - 6.1.82-r1
kmod-crypto-sha256 - 6.1.82-r1
kmod-crypto-sha512 - 6.1.82-r1
kmod-gpio-button-hotplug - 6.1.82-r3
kmod-hwmon-core - 6.1.82-r1
kmod-leds-gpio - 6.1.82-r1
kmod-lib-crc-ccitt - 6.1.82-r1
kmod-lib-crc32c - 6.1.82-r1
kmod-libphy - 6.1.82-r1
kmod-mac80211 - 6.1.82.6.6.15-r1
kmod-mt76-connac - 6.1.82.2024.04.03~1e336a85-r1
kmod-mt76-core - 6.1.82.2024.04.03~1e336a85-r1
kmod-mt7915e - 6.1.82.2024.04.03~1e336a85-r1
kmod-mt7981-firmware - 6.1.82.2024.04.03~1e336a85-r1
kmod-nf-conntrack - 6.1.82-r1
kmod-nf-conntrack6 - 6.1.82-r1
kmod-nf-flow - 6.1.82-r1
kmod-nf-log - 6.1.82-r1
kmod-nf-log6 - 6.1.82-r1
kmod-nf-nat - 6.1.82-r1
kmod-nf-reject - 6.1.82-r1
kmod-nf-reject6 - 6.1.82-r1
[... 601 lines cut ...]
Thu Apr  4 23:13:39 2024 daemon.notice netifd: Network device 'phy1-ap0' link is up
Thu Apr  4 23:13:39 2024 daemon.notice hostapd: Reloaded settings for phy phy1
Thu Apr  4 23:13:39 2024 daemon.notice wpa_supplicant[1530]: Set new config for phy phy1
Thu Apr  4 23:13:39 2024 daemon.notice netifd: Wireless device 'radio1' is now up
Thu Apr  4 23:13:39 2024 daemon.notice netifd: Interface 'wwan' is enabled
Thu Apr  4 23:13:39 2024 daemon.notice netifd: Interface 'wwan' has link connectivity
Thu Apr  4 23:13:39 2024 daemon.notice netifd: Interface 'wwan' is setting up now
Thu Apr  4 23:13:39 2024 daemon.notice netifd: wwan (3251): udhcpc: started, v1.36.1
Thu Apr  4 23:13:39 2024 daemon.notice netifd: wwan (3251): udhcpc: broadcasting discover
Thu Apr  4 23:13:39 2024 daemon.notice netifd: wwan (3251): udhcpc: broadcasting select for 192.168.111.211, server 192.168.111.1
Thu Apr  4 23:13:39 2024 daemon.notice netifd: wwan (3251): udhcpc: lease of 192.168.111.211 obtained from 192.168.111.1, lease time 43200
Thu Apr  4 23:13:39 2024 daemon.notice netifd: Interface 'wwan' is now up
Thu Apr  4 23:13:39 2024 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Thu Apr  4 23:13:39 2024 daemon.info dnsmasq[1]: using nameserver 192.168.111.1#53
Thu Apr  4 23:13:39 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Thu Apr  4 23:13:39 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Thu Apr  4 23:13:39 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Thu Apr  4 23:13:39 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Thu Apr  4 23:13:39 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Thu Apr  4 23:13:39 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Thu Apr  4 23:13:39 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Thu Apr  4 23:13:39 2024 user.notice firewall: Reloading firewall due to ifup of wwan (phy1-sta0)
Thu Apr  4 23:13:40 2024 daemon.warn odhcpd[1661]: No default route present, overriding ra_lifetime!
```
</details>

brada4, 2024-04-05:
Set channel auto in radio config and maybe set country code. 
I have only country code...

hashcan, 2024-04-05:
> Set channel auto in radio config

upstairs wifi is on channel 36. I set wifi channel to auto and set country code but no change.  it doesn't connect at boot.  

<details><summary>/etc/config/wireless</summary>

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi'
	option channel '1'
	option band '2g'
	option htmode 'HE20'
	option disabled '1'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/18000000.wifi+1'
	option channel 'auto'
	option band '5g'
	option htmode 'HE80'
	option cell_density '0'
	option country 'US'

config wifi-iface 'wifinet2'
	option device 'radio1'
	option mode 'sta'
	option network 'wwan'
	option ssid 'upstairs'
	option encryption 'sae'
	option key 'XXXX'

config wifi-iface 'wifinet3'
	option device 'radio1'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'sae'
	option macaddr 'random'
	option key '88888888'
	option network 'lan'

```
</details>

brada4, 2024-04-05:
I have different ordering in wireless file, ap first in file even entered later in luci.

and none of
```
daemon.notice netifd: radio1 (2113): sh: out of range
wpa_supplicant[1530]: phy1-sta0: Unknown event 37
```
in logread.

wwan is quite normal usage, cf travelmate...

hashcan, 2024-04-06:
> I have different ordering in wireless file, ap first in file even entered later in luci.

reversing the order of  wireless from sta, ap to ap, sta also did not help.

hashcan, 2024-04-06:
mentioning @nbd168 as listed for both maintainers of netifd and hostapd.

brada4, 2024-04-06:
Atypical showings in the log:
https://github.com/openwrt/openwrt/issues/15028#issuecomment-2040711004

hashcan, 2024-04-14:
Tested with OpenWrt SNAPSHOT r25915-c2f52e42b1 / LuCI Master 24.100.29615~1181915 and the issue still exists.

brada4, 2024-04-14:
Wait for dev to come by and tell issue is considered. Your config is very generic, and log indications are atypical (on top of you say it does not work)

hashcan, 2024-04-14:
> Wait for dev to come by and tell issue is considered. Your config is very generic, and log indications are atypical (on top of you say it does not work)

Think I found the underlying problem. 
Luci shows the right channel in here :
[image]
 
but when you join the network and pass this page :
[image]

you still get the wifi set to the previous channel despite checking the replace wifi configuration option. 
[image]

all will be good if I set the channel to the right one.

brada4, 2024-04-14:
You need to choose one firewall zone for the network.

hashcan, 2024-04-14:
> You need to choose one firewall zone for the network.

The issue is Luci not setting the right channel on the last photo of my previous message.

hashcan, 2024-04-29:
The issue is now resolved on latest snapshot.
Thank you so much

closed 2024-04-29 by hashcan

diogosena, 2024-09-16:
This is happening again, wifi connection is ok, but the wwan interface not getting an IP address at boot, restarting the wwan interface works as a workaround.

funnylinux, 2024-09-22:
OpenWrt SNAPSHOT r27082-b733b6acb5

I'm having exactly the same issue. I have to click on the restart button for wwan interface in Luci GUi, then wwan interface can get the IP address.