openwrt/openwrt #15030: dnsmasq: dnsmasq fails to run after installing dnsmasq-full on LXC

JoverZhang, 2024-03-31:
### Describe the bug

In the LXC. When I delete the default `dnsmasq` and reinstall `dnsmasq-full`, the `dnsmasq` becomes unable to run properly.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

x86/64

### Device

Default string Default string

### Image kind

Official downloaded image

### Steps to reproduce

1. Create CT on the Proxmox. (It should also be reproduced in LXC)
2. Remove the default dnsmasq. `opkg remove dnsmasq`
3. Install the dnsmasq-full. `opkg install dnsmasq-full`
4. Restart the dnsmasq. `/etc/init.d/dnsmasq restart`
5. Check the dnsmasq info. `/etc/init.d/dnsmasq info`.

You can see some like this:

```text
{
        "dnsmasq": {
                "instances": {
                        "cfg01411c": {
                                "running": false, // Failed to run
                                "command": [
                                        "/usr/sbin/dnsmasq",
                                        "-C",
                                        "/var/etc/dnsmasq.conf.cfg01411c",
                                        "-k",
                                        "-x",
                                        "/var/run/dnsmasq/dnsmasq.cfg01411c.pid"
                                ],
                                ......
```

### Actual behaviour

I found some error messages by the `logread`, like below:

```
Sun Mar 31 13:40:23 2024 user.err : jail: failed to mount /dev/log /tmp/ujail-IFMDJL/dev/log: Permission denied
Sun Mar 31 13:40:23 2024 user.err : jail: mount_all() failed
Sun Mar 31 13:40:23 2024 user.err : jail: failed to build jail fs
```

Then I found this script in `lxc-ci`:

https://github.com/lxc/lxc-ci/blob/a7a0d8d9408ac6ead221e368de8b1df5adcbb756/images/openwrt.yaml#L351C1-L352C57

```
    # Disable process isolation to make dnsmasq work
    sed -i 's/procd_add_jail/: \0/g' /etc/init.d/dnsmasq
```

And then, when I commented out the following code in `dnsmasq.init`, the `dnsmasq` could run normally.

```
	procd_add_jail dnsmasq ubus log
	procd_add_jail_mount $CONFIGFILE $DHCPBOGUSHOSTNAMEFILE $DHCPSCRIPT $DHCPSCRIPT_DEPENDS
	procd_add_jail_mount $EXTRA_MOUNT $RFC6761FILE $TRUSTANCHORSFILE
	procd_add_jail_mount $dnsmasqconffile $dnsmasqconfdir $resolvdir $user_dhcpscript
	procd_add_jail_mount /etc/passwd /etc/group /etc/TZ /etc/hosts /etc/ethers
	procd_add_jail_mount_rw /var/run/dnsmasq/ $leasefile
	case "$logfacility" in */*)
		[ ! -e "$logfacility" ] && touch "$logfacility"
		procd_add_jail_mount_rw "$logfacility"
	esac
	[ -e "$hostsfile" ] && procd_add_jail_mount $hostsfile

```

### Expected behaviour

It can run normally on LXC without any modifications.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-31:
> Then I found this script 

You need to  remove dnsmasq and install dnsmasq-full in place where sudo gets installed for config file patching at the end of preparation script to apply.
Alternative would be to make whole jail setup section to be `[ -x /sbin/ujail ]` conditional like in most other scripts, then yml script could get away with uninstalling ujail.

brada4, 2024-04-23:
My question was if making sandboxing conditional of presence of ujail makes uninstalling ujail permit your container to proceed.

chris6868, 2024-11-17:
LXC安装后在容器里点选项-功能-嵌套，把嵌套勾上就可以启动了，不需要改代码
After LXC installation, you can click the option in the container-function-nested, you can start the nested hook, you don’t need to change the code（英文不好，谷歌翻译的，抱歉）

MalikWilliams1greher, 2024-11-27:
What directory is dnsmasq.init in please?

Bazo0Z0o, 2025-02-02:
Thank you for finding solution!