openwrt/openwrt #15039: IPv6 DNS Server for DHCP Tag

svenopdehipt, 2024-04-01:
### Describe the bug

The IPv6 server settings for an tag doesn't overwrite the network specific settings.

I did edit `/etc/config/dhcp` and set an IPv4 and IPv6 dns server for an tag with

```
config tag 'tag1'
	list dhcp_option '6,<ip>'
        list dns '<ipv6>'
```

and apply the tag with

```
config host
	option name '<name>'
	option duid '<duid>'
	list mac '<mac>'
	list tag 'tag1'
```

After that I reloaded the config with `service dnsmasq restart`.

This setting overwrite the dns server for the client, but the IPv6 dns server is the one specified in the settings of the interface.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2024-04-01:
Normally, odhcpd is providing internal DHCPv6, not dnsmasq. Is this true for your setup?

svenopdehipt, 2024-04-01:
I haven't changed much of the default setup. Can I submit a specific IPv6 dns server to a client (or a group of clients) with odhcpd?

dave14305, 2024-04-12:
> Can I submit a specific IPv6 dns server to a client (or a group of clients) with odhcpd?

No, it doesn’t offer custom DNS options per-client. 

See all the options at 

https://github.com/openwrt/odhcpd/blob/master/README

svenopdehipt, 2024-04-15:
Ok, thanks for the information. It would be nice if this would be supported in some future release. I will close this issue now.

closed 2024-04-15 by svenopdehipt