openwrt/openwrt #15045: 5GHz radio issue on Comfast EW72 V2

aswanthk07, 2024-04-02:
### Describe the bug

I've encountered a persistent issue with my Comfast EW72-V2 router regarding the 5GHz WiFi network. After a reboot, the 5GHz WiFi fails to come up, and sometimes, along with this issue, the router also fails to provide DHCP services. Interestingly, a simple reboot resolves both issues temporarily. However, upon subsequent reboots, the same problems recur. I've attempted to address this by testing different firmware versions, including OpenWrt 23.05.2 Stable and 23.05.3 Stable, but unfortunately, both versions exhibited the same issue. It appears that the problem may be related to the loading of the mt7663pr2h_rebb.bin firmware file. Despite trying different firmware versions, the issue persists, suggesting a potential hardware or configuration issue.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3, 23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

COMFAST CF-EW72 V2

### Image kind

Official downloaded image

### Steps to reproduce

1.     Reboot the router.
2.     Observe that the 5GHz WiFi network does not come up.
3.     Additionally, DHCP services may not be available.
4.     Reboot the router again to temporarily resolve the issues.

### Actual behaviour

Sometimes 5GHz radio is not coming up on the boot. A reboot will fix the issue.

### Expected behaviour

Expects both 2.4GHz and 5GHz radios to come up while booting the device

### Additional info

`[    9.567121] kmodloader: loading kernel modules from /etc/modules.d/*
[    9.830728] Loading modules backported from Linux version v6.1.24-0-g0102425ac76b
[    9.838236] Backport generated by backports.git v5.15.92-1-44-gd6ea70fafd36
[   10.121935] pci 0000:00:00.0: enabling device (0006 -> 0007)
[   10.127645] mt7603e 0000:01:00.0: enabling device (0000 -> 0002)
[   10.133945] mt7603e 0000:01:00.0: ASIC revision: 76030010
[   10.811635] urngd: v1.0.2 started.
[   11.170544] mt7603e 0000:01:00.0: Firmware Version: ap_pcie
[   11.176147] mt7603e 0000:01:00.0: Build Time: 20160107100755
[   11.203263] mt7603e 0000:01:00.0: firmware init done
[   11.389058] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   11.401297] pci 0000:00:01.0: enabling device (0000 -> 0003)
[   11.407012] mt7615e 0000:02:00.0: enabling device (0000 -> 0002)
[   11.441182] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   11.441237] ieee80211 phy1: copying sband (band 1) due to VHT EXT NSS BW flag
[   11.518151] PPP generic driver version 2.4.2
[   11.526658] NET: Registered PF_PPPOX protocol family
[   11.545893] kmodloader: done loading kernel modules from /etc/modules.d/*
[   11.585303] random: crng init done
[   11.588854] random: 30 urandom warning(s) missed due to ratelimiting
[   13.508340] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[   20.546688] mtk_soc_eth 1e100000.ethernet eth0: Link is Down
[   20.573358] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   20.581817] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   20.590922] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   20.599548] mt7530-mdio mdio-bus:1f lan: configuring for phy/gmii link mode
[   20.609276] br-lan: port 1(lan) entered blocking state
[   20.614505] br-lan: port 1(lan) entered disabled state
[   20.622686] device lan entered promiscuous mode
[   20.627279] device eth0 entered promiscuous mode
[   20.667507] mt7530-mdio mdio-bus:1f wan: configuring for phy/gmii link mode
[   22.224472] br-lan: port 2(phy0-ap0) entered blocking state
[   22.230203] br-lan: port 2(phy0-ap0) entered disabled state
[   22.237571] device phy0-ap0 entered promiscuous mode
[   22.259831] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[   22.266997] br-lan: port 2(phy0-ap0) entered blocking state
[   22.272761] br-lan: port 2(phy0-ap0) entered forwarding state
[   22.280580] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   23.488582] mt7530-mdio mdio-bus:1f lan: Link is Up - 1Gbps/Full - flow control rx/tx
[   34.438547] mt7615e 0000:02:00.0: Message 00000010 (seq 1) timeout
[   34.444751] mt7615e 0000:02:00.0: Failed to get patch semaphore
[   34.450750] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[   34.735625] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[   54.918547] mt7615e 0000:02:00.0: Message 00000010 (seq 2) timeout
[   54.924750] mt7615e 0000:02:00.0: Failed to get patch semaphore
[   54.930745] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[   55.216252] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[   75.398569] mt7615e 0000:02:00.0: Message 00000010 (seq 3) timeout
[   75.404783] mt7615e 0000:02:00.0: Failed to get patch semaphore
[   75.410827] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[   75.696054] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[   95.878545] mt7615e 0000:02:00.0: Message 00000010 (seq 4) timeout
[   95.884762] mt7615e 0000:02:00.0: Failed to get patch semaphore
[   95.890803] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[   96.178604] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  116.358538] mt7615e 0000:02:00.0: Message 00000010 (seq 5) timeout
[  116.364754] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  116.370798] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  116.656097] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  136.838541] mt7615e 0000:02:00.0: Message 00000010 (seq 6) timeout
[  136.844756] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  136.850830] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  137.136214] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  157.318537] mt7615e 0000:02:00.0: Message 00000010 (seq 7) timeout
[  157.324749] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  157.330787] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  157.616022] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  177.798531] mt7615e 0000:02:00.0: Message 00000010 (seq 8) timeout
[  177.804764] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  177.810801] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  178.096150] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  198.278544] mt7615e 0000:02:00.0: Message 00000010 (seq 9) timeout
[  198.284780] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  198.290850] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  198.575995] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  218.758584] mt7615e 0000:02:00.0: Message 00000010 (seq 10) timeout
[  218.764901] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  218.770950] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  219.055944] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  239.238529] mt7615e 0000:02:00.0: Message 00000010 (seq 11) timeout
[  239.244848] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  239.250915] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  239.262428] br-lan: port 1(lan) entered blocking state
[  239.267687] br-lan: port 1(lan) entered forwarding state
[  239.851589] br-lan: port 2(phy0-ap0) entered disabled state
[  239.859682] device phy0-ap0 left promiscuous mode

`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aswanthk07, 2024-04-02:
> Any improvement with the snapshot?

Tried snapshot, same issue persists.

brada4, 2024-04-02:
@nbd168 ?

brada4, 2024-04-03:
yes.

aswanthk07, 2024-04-04:
> yes.

Any hopes for a fix?

brada4, 2024-04-04:
It gets better over time.