openwrt/openwrt #15047: IGMP queries should use interface's source address

mnlipp, 2024-04-02:
### Describe the bug

The RFCs for all versions of IGMP allow the use of 0.0.0.0 but recommend to use an IP address to support querier selection. Currently, it is not possible to enable the use of the interface's address except by writing to `/sys/devices/virtual/net/*/bridge/multicast_query_use_ifaddr` directly. A setting for this should be added and probably default to "use interface address".

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ZyXEL NBG6817

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dannil, 2024-04-02:
`A setting for this should be added and probably default to "use interface address".`

I assume you mean this should be possible through LuCI since it works if you edit the file directly? Then you should post the issue in https://github.com/openwrt/luci instead.

mnlipp, 2024-04-02:
LuCI would be nice, but I'd already be happy if I could configure this in some file. But which one? I cannot find anything about setting this in e.g. `etc/config/network`.