openwrt/openwrt #15048: Help fixing channel support of WHW03 V2, help porting to WHW03 V1

Lanchon, 2024-04-02:
### Describe the bug

@hauke,
@vtremblay,

the purpose of this thread is twofold:
- getting help fixing channel support of WHW03 V2 (NAND-based)
- getting help porting openwrt to WHW03 V1 (eMMC-based, but otherwise seems identical)

## the channel problem

there are a number of tri-band routers based on IPQ4019, which seem to be very much alike:
- Linksys EA8300/MR8300 (EA8250/MR8250)
- Linksys WHW03 V1/V2
- ASUS Lyra (MAP-AC2200)

of these, all of them except WHW03 V2 (including [this WHW03 V1 out of tree port originally by @flipy](https://github.com/openwrt/openwrt/pull/14350)) support these channel ranges on their separate 5 GHz interfaces:
- 36 ~ 64: supported
- 100 ~ 165: supported

however, the official WHW03 V2 port only supports a much reduced range, which is suspect (i think these channels are the ones that do not require DFS in the USA):
- 36 ~ 48: supported
- **52 ~ 64: unsupported**
- **100 ~ 144: unsupported**
- 149 ~ 165: supported

i've also seen that the official WHW03 V2 port seems to ignore the regulatory country setting. it clearly uses DFS, as 5 GHz networks are not enabled for a minute or two while the devices does CAC (Channel Availability Check). when the regulatory region is set to Panama, which does not use DFS, the CAC period remains active. (other QCA routers i own correctly disable DFS in this case.) also, the device shouldn't be using DFS at all for the restricted channel range it uses, AFAIK.

i have a single V2 to play with, currently with stock firmware. when running old 2.1.9.195026 firmware, the device indeed allows the same restricted channel range as the openwrt port. however in current 2.1.19.209783 firmware, channel selection has been removed from the GUI so there is no way to know the range. (during testing my device chose channels 40 and 149.)

i am assuming the RF section of all these routers is the same, and thus all should support the same extended range channel range... unless they cut corners on the RF filters for V2. (note: they use different FCC IDs.) how can i push openwrt WHW03 V2 to use the full range in order to test? (id don't really care if i damage my hardware, this restricted channel range make the routers unusable.)

my uninformed guess is that maybe this info is encoded in the [board data files](https://github.com/openwrt/firmware_qca-wireless/commit/711d8ea2e3b47b03d74d6be8d09797051ecb3ade). is this true? the OoT V1 port uses [different files](https://github.com/openwrt/openwrt/pull/14350/commits/e030c835c43ecc0f3453dbb7414c888ba5099285#diff-51c24c1a7b736127c0fa1b2e44b2a0cbba17d70a998209c5287965919fa7c16c).

where did these files come from? i downloaded the latest [stock firmwares](https://www.linksys.com/support-article?articleNum=47907) for V1 and V2, unpacked the ext4 and ubifs files, and uploaded them [here](https://github.com/Lanchon/whw03-stock-firmware). firmwares for V1 and V2 seem to be in sync. qca bin files are [here](https://github.com/Lanchon/whw03-stock-firmware/tree/main/whw03v2_2.1.19.209783_2022-06-20/rootfs/lib/firmware). i compared `/lib/firmware` directories of V1 and V2 firmwares and they match bit for bit, so each firmware seem to contain the bin files for both hardware versions.

next i checked the 4 files from openwrt V1 and V2 ports, and they do not match any of the files in stock `/lib/firmware`. so where did they come from? earlier versions of stock? or am i barking up the wrong tree? why does stock seem to have different bin files for different regions and openwrt does not? should i try V1 port bin files in a V2 build to unlock the extra channels?

## the WHW03 V1 port

i want to build on flipy's port and get it merged into openwrt but i really can't continue until i understand where the qca bin files included in the port came from.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys WHW03 V2 (Velop)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-02:
That indicates wifi driver does not support DFS. 
Best to start or join porting thread at forum.

Lanchon, 2024-04-02:
> That indicates wifi driver does not support DFS. Best to start or join porting thread at forum.

this involves an officially supported device.

brada4, 2024-04-02:
https://openwrt.org/toh/linksys/whw03_v2?s[]=linksys&s[]=whw03&s[]=v2#limitations

brada4, 2024-04-02:
i.e if you are super-lucky and discover more generic caldata on flash you can fix it.

Lanchon, 2024-04-02:
> [https://openwrt.org/toh/linksys/whw03_v2?s[]=linksys&s[]=whw03&s[]=v2#limitations](https://openwrt.org/toh/linksys/whw03_v2?s%5B%5D=linksys&s%5B%5D=whw03&s%5B%5D=v2#limitations)

i'm aware of that text because i wrote it... unfortunately after buying some WHW03 V2s.

brada4, 2024-04-02:
bring radios down, reboot device,
iw reg set US
iw list (the frequency maps per adapter)
iw reg set BE
iw list
iw reg set AU
iw list

You can attach 3 outputs but you can put frequency maps side by side and discover what is common to missing channels. Likely your assessment is 99% correct.
dont start out of order radio, reboot again and start AP-s back.

Lanchon, 2024-04-02:
> i.e if you are super-lucky and discover more generic caldata on flash you can fix it.

so you say channel limitations are not in the firmware image, but rather in the factory-burnt partition that contains caldata? also, i suppose this caldata is tailored to each particular device after measurements and thus cannot be transplanted to a different device? or could i try a V1 to V2 transplant?

brada4, 2024-04-02:
caldata is in factory partitions usually. near MAC address, see dts file.

Lanchon, 2024-04-02:
> bring radios down, reboot device, iw reg set US iw list (the frequency maps per adapter) iw reg set BE iw list iw reg set AU iw list
>
> You can attach 3 outputs but you can put frequency maps side by [...]

[here it is](https://gist.github.com/Lanchon/fc5ecb708a004577deeec88a750f8f01). BE actually diminishes power, so country is not being ignored.

brada4, 2024-04-02:
"normal" BE signal map:
```
                        * 5180 MHz [36] (23.0 dBm)
                        * 5200 MHz [40] (23.0 dBm)
                        * 5220 MHz [44] (23.0 dBm)
                        * 5240 MHz [48] (23.0 dBm)
                        * 5260 MHz [52] (20.0 dBm) (radar detection)
                        * 5280 MHz [56] (20.0 dBm) (radar detection)
                        * 5300 MHz [60] (20.0 dBm) (radar detection)
                        * 5320 MHz [64] (20.0 dBm) (radar detection)
                        * 5500 MHz [100] (26.0 dBm) (radar detection)
                        * 5520 MHz [104] (26.0 dBm) (radar detection)
                        * 5540 MHz [108] (26.0 dBm) (radar detection)
                        * 5560 MHz [112] (26.0 dBm) (radar detection)
                        * 5580 MHz [116] (26.0 dBm) (radar detection)
                        * 5600 MHz [120] (26.0 dBm) (radar detection)
                        * 5620 MHz [124] (26.0 dBm) (radar detection)
                        * 5640 MHz [128] (26.0 dBm) (radar detection)
                        * 5660 MHz [132] (26.0 dBm) (radar detection)
                        * 5680 MHz [136] (26.0 dBm) (radar detection)
                        * 5700 MHz [140] (26.0 dBm) (radar detection)
                        * 5720 MHz [144] (13.0 dBm) (radar detection)
                        * 5745 MHz [149] (13.0 dBm)
                        * 5765 MHz [153] (13.0 dBm)
                        * 5785 MHz [157] (13.0 dBm)
                        * 5805 MHz [161] (13.0 dBm)
                        * 5825 MHz [165] (13.0 dBm)
                        * 5845 MHz [169] (13.0 dBm)
                        * 5865 MHz [173] (13.0 dBm)
                        * 5885 MHz [177] (disabled)
```

I can only back your finding that radios are rather limited. AUTO-BW is ok, DFS is not supported or channels are totally masked by oem caldata.

Lanchon, 2024-04-02:
> Can you revert to OEM and check the available channels there?

please read the OP

brada4, 2024-04-02:
My question is more - if the only channel allowed is same as was once selected by OEM iphone app and then left forever.

testuser7, 2024-04-02:
Script from the OEM firmware to set the frequency depending on the region: https://gist.github.com/testuser7/81c4810803d6b9e661a3aa2eb413e564

Lanchon, 2024-04-02:
> Frequencies are limited in dts. This model likely had different limitations depending on the market in which it was sold. You can change the limits in dts and check if the other channels also work:
> 
>     1. wifi0 - `ieee80211-freq-limit = <2402000 2482000>;` (or remove the limit)
> 
>     2. wifi1 - `ieee80211-freq-limit = <5170000 5350000>;`
> 
>     3. wifi2 - `ieee80211-freq-limit = <5470000 5895000>;`

thanks! you are absolutely right!!

the restricted WHW03 V2 port has:
wifi0: ieee80211-freq-limit = <2401000 2473000>; # 1 to 11 DSSS (802.11 b)
wifi1: ieee80211-freq-limit = <5170000 5250000>; # 36 to 48
wifi2: ieee80211-freq-limit = <5735000 5835000>; # 149 to 165

you propose:
wifi0: ieee80211-freq-limit = <2402000 2482000>; # 1 to 13 OFDM (802.11 g/n; b not supported) # or deleted, which i prefer
wifi1: ieee80211-freq-limit = <5170000 5350000>; # 36 to **68**
wifi2: ieee80211-freq-limit = <5470000 5895000>; # **96** to 177

flipy's port of V1 has:
wifi0: ieee80211-freq-limit = <2401000 2473000>; # 1 to 11 DSSS (802.11 b) # i will delete this for my port
wifi1: ieee80211-freq-limit = <5170000 5330000>; # 36 to **64**
wifi2: ieee80211-freq-limit = <5490000 5835000>; # **100** to 165

EA8300/MR8300 have:
wifi0: # no limits
wifi1: ieee80211-freq-limit = <5170000 5330000>; # 36 to **64**
wifi2: ieee80211-freq-limit = <5490000 5835000>; # **100** to 165

ASUS Lyra (MAP-AC2200) has:
wifi0: # no limits
wifi2: ieee80211-freq-limit = <5170000 5350000>; # 36 to **68**
wifi1: ieee80211-freq-limit = <5470000 5875000>; # **96** to 173

some thoughts:

- i will remove the 2.4 GHz limits. i want chan 14 (DSSS-only) available if applicable and support for both ODFM and DSSS in chans 1 to 13.
- for all i know, all these devices share 2 antennas for both 5GHz 2x2 radios, which requires a pair of crossover filters. if i enable chans 68 and 96 as you suggested, the bandwidth available for the crossover filters gets reduced. depending on filter design, there might be too much crossover on those frequencies for the devices to work properly (and in a potentially very extreme case, even damage hardware). all linksys devices in this family avoid chans 68 and 96, however the ASUS device allows them. do you think this is just a case of the DTSes being pulled from region restricted devices? or a genuine limitation of the crossovers? i think it's likely that all devices share QCA ref design (including crossovers), including the ASUS device.
- i'm tempted to include chans 169 to 177, as you suggested. i don't see a technical reason for them being left out. (maybe antenna design?) ASUS only leaves out chan 177, possibly because its DTS was pulled from region restricted device such as [Europe or UK](https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/n/ac/ax)). the QCA firmware should limit itself to supported chans anyway.
- by the same token, i'm also temped to include chan 32 (5150000–5170000), which you did not suggest. why? why is it left out from most routers?

## the bin files

i'm still left not knowing where the bin files in the openwrt ports came from. i need to know in order to work on the V1 port.

Lanchon, 2024-04-02:
> Script from the OEM firmware to set the frequency depending on the region: https://gist.github.com/testuser7/81c4810803d6b9e661a3aa2eb413e564

thanks!

wtf? why are channels below 36 and above 165 not supported in any region? the seem to be perfectly legal in many:
https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/n/ac/ax)

Lanchon, 2024-04-02:
so chans 32, 68, 96, 169, 173 and 177 are not available anywhere. why is this?

testuser7, 2024-04-02:
> so chans 32, 68, 96, 169, 173 and 177 are not available anywhere. why is this?

Channels 2, 68 and 96 are not really used. Channels 169, 173 and 177 are new, available recently.

brada4, 2024-04-02:
Those low would be 20MHz, likely not implemented in any ac+ silicon because ac mandates 40Mhz min. 
Upper were not open until just very recent, along 6GHz band. Probaly same limitation as above.

brada4, 2024-04-02:
https://gist.github.com/testuser7/81c4810803d6b9e661a3aa2eb413e564#file-wifi_info_set-sh-L10
Actually EU ch 52 (radar) is excluded while 100+ (all radar) are included.
I think you need to find maxima of what each radio supports, but there is no caldata to back it. Maybe you need a script to adjust firmware. What is under syscfg_set? I suspect it patches official caldata somehow. (based on BE has no ch 13)

Lanchon, 2024-04-02:
> Actually EU ch 52 (radar) is excluded while 100+ (all radar) are included. 

i think that is because 52 to 64 are DFS in the EU, but 100+ are not:
https://gist.github.com/testuser7/81c4810803d6b9e661a3aa2eb413e564#file-wifi_info_set-sh-L51-L53

> Maybe you need a script to adjust firmware. What is under syscfg_set? I suspect it patches official caldata somehow. (based on BE has no ch 13)

i think this is just chans 12 and 13 being filtered out in the current DTS:
https://github.com/openwrt/openwrt/blob/d1eb0bec390a3bb77c3b3f5f792469a780578bbf/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4019-whw03v2.dts#L493

Lanchon, 2024-04-02:
> Script from the OEM firmware to set the frequency depending on the region: https://gist.github.com/testuser7/81c4810803d6b9e661a3aa2eb413e564

where is that code in openwrt? i can't find it

testuser7, 2024-04-02:
> where is that code in openwrt? i can't find it

https://github.com/openwrt/openwrt/blob/main/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4019-whw03v2.dts#L493

Lanchon, 2024-04-03:
i think you misunderstood my question

testuser7, 2024-04-03:
Yeah, the code responsible for setting the limits is located in the kernel sources: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/net/wireless/of.c?h=v6.6.23

Lanchon, 2024-04-03:
i made a build with the extra channels, works great:

[openwrt-ipq40xx-generic-linksys_whw03v2-EXTRA-CHANNELS-squashfs-sysupgrade.bin.gz](https://github.com/openwrt/openwrt/files/14849109/openwrt-ipq40xx-generic-linksys_whw03v2-EXTRA-CHANNELS-squashfs-sysupgrade.bin.gz)

source: https://github.com/Lanchon/openwrt/tree/90d556d29ee466665e9c960c297c7e98e9fa9910

will create a PR soon...

Lanchon, 2024-04-03:
@testuser7... THANK YOU!!

brada4, 2024-04-03:
515...535 547... the 3 extra 20mhz channels need to be tested

Lanchon, 2024-04-03:
> 515...535 547... the 3 extra 20mhz channels need to be tested

i have not included them in the PR.

openwrt doesn't seem able to use them anyway. when that changes, we can revisit those channels for all implementations of this reference design.

i did try a build with those channels though. as expected, even if switching to 802.11n and 20 MHz bandwidth, the UI ignored their existence. this is logical, as for other devices that do not limit their channels at all in their DTS (they have a single 5 GHz adapter), those channels do not show either.

brada4, 2024-04-03:
It is of little use, in most cases you need to push parameter via command line.

Lanchon, 2024-04-11:
closing as i wait for the commit being reviewed...

closed 2024-04-11 by Lanchon

Lanchon, 2024-04-18:
# news!

**the changes were committed to both the master branch and the 23.05 branch, so all official images after 23.05.3 should support all channels :)**

in the meantime, i made a 23.05.3 image that *should* be compatible with all the official packages including kernel modules. (this was not the case with my previous image.)

the problem with this newer image is that i could not test it as my router is elsewhere performing real life duties. so please, if you test it, report here.

the image *should* be equivalent to the official one with all the same packages, but it surprises me that it is quite a bit smaller, so maybe something is amiss. but the kernel's magic is for sure equal to the official build, so if any packages are missing, you should be able to install them for the official repo.

[openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-EXTRA_CHANNELS-squashfs-sysupgrade.bin.gz](https://github.com/openwrt/openwrt/files/15030992/openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-EXTRA_CHANNELS-squashfs-sysupgrade.bin.gz)
(i gzipped the file because of github's filetype upload restrictions.)

flipy, 2024-05-01:
> in the meantime, i made a 23.05.3 image that _should_ be compatible with all the official packages including kernel modules. (this was not the case with my previous image.)
>
> the problem with this [...]

Looks like this latest build does not have the extra channels enabled

Lanchon, 2024-05-01:
> Looks like this latest build does not have the extra channels enabled

hi flipy, what do you mean by "latest build"?

flipy, 2024-05-02:
Hi Lanchon!
Looks like the latest 23.05 build
<https://github.com/openwrt/openwrt/files/15030992/openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-EXTRA_CHANNELS-squashfs-sysupgrade.bin.gz>
your shared on OpenWRT github issue
<https://github.com/openwrt/openwrt/issues/15048> with the extra channels
does not have your DTB changes to enable new frequencies.
Nice find though, I'll check my extracted DTB files and see if there is any
mention that might be useful -- it has been a while since I worked on those
devices.

Lanchon, 2024-05-02:
> Hi Lanchon! Looks like the latest 23.05 build <https://github.com/openwrt/openwrt/files/15030992/openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-EXTRA_CHANNELS-squashfs-sysupgrade.bin.gz> your shared on OpenWRT github issue <#15048> with the extra channels does not have your DTB changes to enable new frequencies. Nice find though, I'll check my extracted DTB files and see if there is any mention that might be useful -- it has been a while since I worked on those devices.
> […](#)
> -- Atentamente, Andreas Calvo

i thought you were referring to official builds.

ok, got you now... but what you are saying is unlikely. i ssh'd into TWO production V2's which are running 23.05.3 and did this:

1. i attended-sysupgrade each router twice, to make sure both copies of the firmware had the same working setup. so if i messed up, i could have someone do the "3 interrupted boots" dance to have the router revert to the working setup.
2. using the command line, i flashed only the kernel of this latest build (the one you say it does not have the relevant changes) and rebooted.
3. using LuCI, which now showed the extra channels, i configured the devices to use the new channels.

both devices are working properly, each acting as simultaneously as an AP and a client using the two 5G adapters, and all 4 adapters are operating on frequencies previously unavailable.

so i have to ask:

> this latest build does not have the extra channels enabled

how did you arrive at this conclusion?

dwardor, 2024-05-03:
Just tested https://github.com/openwrt/openwrt/files/15030992/openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-EXTRA_CHANNELS-squashfs-sysupgrade.bin.gz on one of my A03 (Marketing name of the WHW03 when aiming Apple customers it seems) and can confirm it had the extra 49-64 channels on radio2 (5GHz)

However my A03 is not reporting a radio0 (5GHz) with it so couldn't check if channels 100-148 were there :-( !

I noticed that once gunzipped, openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-EXTRA_CHANNELS-squashfs-sysupgrade.bin was somewhat smaller (6.7Mb) than the official openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-squashfs-sysupgrade.bin (7.4Mb if I recall correctly). Why so and could something be missing ? explaining why no radio0 is reported [it was definitely there under Linksys firmware :-)] !

What can I do to help figure out why radio0 is not present ?

Lanchon, 2024-05-03:
> Just tested https://github.com/openwrt/openwrt/files/15030992/openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-EXTRA_CHANNELS-squashfs-sysupgrade.bin.gz on one of my A03 (Marketing name of [...]

yes i noticed that in the post itself.

but the kernel is totally fine and compatible with official 23.05.3.

you can untar my file and flash only my kernel over a 23.05.3 install using the command line. or you can use my kernel and official 23.05.3 to make a hybrid sysupgrade file.

i did a hybrid factory file using my kernel and the 23.05.3 factory image. here it is:
[openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-HYBRID-EXTRA_CHANNELS-squashfs-factory.bin.zip](https://github.com/openwrt/openwrt/files/15204411/openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-HYBRID-EXTRA_CHANNELS-squashfs-factory.bin.zip)

that file was made using this command:
```
{ cat kernel-LANCHON /dev/zero | dd bs=1K count=6144 status=none; dd if=openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-squashfs-factory.bin bs=1k skip=6144 status=none; } >factory.bin
```

keep in mind that if you flash a factory file you will loose all settings, so make a backup before!

dwardor, 2024-05-04:
> i did a hybrid factory file using my kernel and the 23.05.3 factory image. here it is: [...]

Reverted to stock and double checked both 5GHz wireless bands worked --> YES !!! So no hardware issue.

Tried this to flash your hybrid factory image... but stock firmware updater rejected it saying it is incorect :-1:  !
[image]

So I downloaded yesterday's snapshot factory image from ([https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ipq40xx%2Fgeneric&id=linksys_whw03v2](https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ipq40xx%2Fgeneric&id=linksys_whw03v2)) and flashed that
all 3 radios work with that, but the extra channels are missing (although I thought your patches were merged) because scan with radio0 is not seeing my channel 116 AP and scan with radio2 is not seeeing my channel 52 AP
[image]

So then I flashed your [https://github.com/openwrt/openwrt/files/15030992/openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-EXTRA_CHANNELS-squashfs-sysupgrade.bin.gz](https://github.com/openwrt/openwrt/files/15030992/openwrt-23.05.3-ipq40xx-generic-linksys_whw03v2-LANCHON-EXTRA_CHANNELS-squashfs-sysupgrade.bin.gz) again (and chosing to wipe all configuration in the process which I had not done when sysupgrading to it the very first time) and now I'm being told I only have 1 5GHz radio (which does see my channel 52 AP)
[image]

So there is definitely an issue with your images with regard to my missing 5GHz radio (that with channels >= 100)

Lanchon, 2024-05-04:
> Reverted to stock and...

LOL, i didn't say because i never thought no one would think of going back to stock to flash the image!

so here it goes: no! you are supposed to flash my factory image already from openwrt, not stock. it must have CRCs which i won't care to update.

also, flash at your own risk. my expectation is that it works, but i don't have V2 hardware to test: i returned the V2 i used to investigate the channel issue to its owner before i created the 23.05.3 compatible build. however, i do know that the kernel of that build (which contains the changes) is totally fine, as it has been tested and is in use.

btw, if you flash it and it doesn't work, the device should get back to its previous working firmware after 3 boot attempts, so just be patient and give it some time and it should recover.

Lanchon, 2024-05-08:
> it's supposed to have a Flash1 Size of 4 GiB

openwrt only uses 136 MB as explained in the commit msg

i plan to publish a procedure to repartition the emmc so that more space is available to openwrt.

> Today, I tested the new firmware, Version SNAPSHOT (r26235-0d74b2a1e5) dated 2024-05-08 10:51:12.

official builds are missing BDFs and wifis might not work well. im waiting for commits to be merged in the proper repos.

> The Ethernet speed doesn't exceed 400 Mb/s, so I reverted back to...

pls explain in detail how you tested both builds and the results of both.
thanks!

Lanchon, 2024-05-08:
@puyoooooo 

> it's supposed to have a Flash1 Size of 4 GiB

openwrt only uses 136 MB as explained in the commit msg

i plan to publish a procedure to repartition the emmc so that more space is available to openwrt.

> Today, I tested the new firmware, Version SNAPSHOT (r26235-0d74b2a1e5) dated 2024-05-08 10:51:12.

official builds are missing BDFs and wifis might not work well. im waiting for commits to be merged in the proper repos.

> The Ethernet speed doesn't exceed 400 Mb/s, so I reverted back to...

pls explain in detail how you tested both builds and the results of both.
thanks!