openwrt/openwrt #15058: AW7916-NPD too low tx power

trippleflux, 2024-04-03:
### Describe the bug

I am getting too low 17 dBm / 50 mW transmit power when using 2.4 GHz as an AP : 
```
# iwinfo 
phy0-ap0  ESSID: "Test"
          Access Point: FF:FF:FF:FF:FF:FF
          Mode: Master  Channel: 4 (2.427 GHz)  HT Mode: HE20
          Center Channel 1: 4 2: unknown
          Tx-Power: 17 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -84 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: 14C3:7906 14C3:7906 [MediaTek MT7916AN]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```

```
# iwinfo phy0-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
* 17 dBm (  50 mW)
```

From 5 GHz : 
```
# iwinfo phy1-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
  25 dBm ( 316 mW)
  26 dBm ( 398 mW)
  27 dBm ( 501 mW)
  28 dBm ( 630 mW)
  29 dBm ( 794 mW)
* 30 dBm (1000 mW)
```

### OpenWrt version

r25735-004c853019

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

generic

### Image kind

Self-built image

### Actual behaviour

2.4 Ghz tx power stuck at 17 dBm

### Expected behaviour

Higher 2.4 GHz tx power, probably the same as 5 GHz tx power, which is 30 dBm.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-04-03:
What is the limit for your country and what is the gain of your antenna? Once gain and antenna gain are added, this maximum cannot be exceeded.

trippleflux, 2024-04-03:
I think it's about 30 dBm maxed
```
(2400 - 2483 @ 40), (N/A, 30), (N/A)
```

While my antenna is around 16 dBi, probably 12 dBi optimal. 

The thing is just for test whatever country that I have chosen, 2.4 GHz seems stuck at 17 dBm.

Looking at the specs at the manufacturer site : 
```
11ac HE80: 14.5+/-1.5dBm@MCS911ax HE20: 20+/-1.5dBm@MCS911ax HE40: 17+/-1.5dBm@MCS911ax HE80: 13.5+/-1.5dBm@MCS1111ax HE160: 13.5+/-1.5dBm@MCS11
```

namiltd, 2024-04-03:
Change country to Taiwan and check.

trippleflux, 2024-04-03:
The same : 
```
global
country TW: DFS-FCC
        (2400 - 2483 @ 40), (N/A, 30), (N/A)
```

In luci also only maxed out at 30 dBm

brada4, 2024-04-04:
check module certification declaration:
https://asiarf.com/product/wi-fi-6-11ax-2x2-dbdc-1800mbps-mini-pcie-module-mt7915-aw7915-npd/
What 1w you are talking about?

trippleflux, 2024-04-04:
@brada4 

You are mistaken, please re-read again @namiltd reply, this link https://w.wol.ph/2015/08/28/maximum-wifi-transmission-power-country/.  I also misread, should be 200 mW max for TW.

Your link also should be https://asiarf.com/product/aw7916-npd-wi-fi-6e-ax3000-module-precision-coaxial-cables-and-wi-fi-7-antennas/ .

I think I should close this, as is probably within intended module certification.

closed 2024-04-04 by trippleflux

brada4, 2024-04-04:
tx power by country
https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/tree/db.txt
and `iw list`  shows that overlapped with hw capabilities.