openwrt/openwrt #15059: Wifi AC radio disappeared from 23.X.X on Tp-Link EAP225 Wall V2

cthulux, 2024-04-03:
### Describe the bug

Wifi AC radio disappeared from fresh install 22.X.X to fresh install 23.X.X (see screenshots):

- "Qualcomm Atheros **QCA9886** 802.11acn": This is not shown anymore, instead I can see "Qualcomm Atheros **QCA9560** 802.11b/g/n"
- "Qualcomm Atheros **QCA9560** 802.11bgn": This is not shown anymore and there is **nothing that replaces it**.

### On 22.03.6 version Wifi AC radio was shown:
[image]

### From 23.X.X version Wifi AC radio is not shown anymore:
[image]

### Logs from 22.03.06:
[22.03.6_log_kernel.txt](https://github.com/openwrt/openwrt/files/14852743/22.03.6_log_kernel.txt)
[22.03.6_log_system.txt](https://github.com/openwrt/openwrt/files/14852744/22.03.6_log_system.txt)

### Logs from 23.05.3:
[23.05.3_log_kernel.txt](https://github.com/openwrt/openwrt/files/14852747/23.05.3_log_kernel.txt)
[23.05.3_log_system.txt](https://github.com/openwrt/openwrt/files/14852748/23.05.3_log_system.txt)

### Screenshots from "Status --> Overview", "Network --> Wireless" for 22.03.06:
[image]
[image]
[image]

### Screenshots from "Status --> Overview", "Network --> Wireless" for 23.05.03:
[image]
[image]
[image]

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link EAP225-Wall v2

### Image kind

Official downloaded image

### Steps to reproduce

Flash new version of firmware from 22.X.X to 23.X.X

### Actual behaviour

Wifi Radio AC is not shown and is not posible to create it. 
Seems like a regresion.

### Expected behaviour

Have the 2 wifi radios that are shown on 22.X.X firmware versions:

- Qualcomm Atheros QCA9886 802.11acn
- Qualcomm Atheros QCA9560 802.11bgn

### Additional info

As shown above, there was a change from QCA9886 to QCA9560 for bgn and for acn from QCA9560 to nothing

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stman, 2024-10-17:
Hello to all. I'll spend time to help fixing this issue too.

cthulux, 2026-10-28:
For anyone experiencing this: a [patch](https://lists.openwrt.org/pipermail/openwrt-devel/2026-October/045210.html) has been sent to the OpenWrt mailing list.

cthulux, 2026-10-30:
Closing this issue due to it is fixed now in main branch with the mentioned patch.-

closed 2026-10-30 by cthulux