openwrt/openwrt #15066: ubox: logread 100% cpu usage in log_handle_fd()

a1bert01, 2024-04-04:
### Describe the bug

we are randomly experiencing logread 100% cpu usage:

logread forwards msgs to remote syslog:
```

 /sbin/logd -S 64
 /sbin/logread -f -h my-hostname -r 192.168.0.1 514 -p /var/run/logread.2.pid -u
```

sometimes (randomly) logread starts consuming 100% of cpu core , probably because there is fd with false eof that is not handled by log_handle_fd():
```

epoll_wait(3, [{EPOLLIN, {u32=4272836, u64=4272836}}], 10, -1) = 1
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11133703, tv_nsec=335917111}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11133703, tv_nsec=336098172}) = 0
epoll_wait(3, [{EPOLLIN, {u32=4272836, u64=4272836}}], 10, -1) = 1

```

```
gdb) print *u
$13 = {
cb = 0x401511 <log_handle_fd>,
fd = 8,
eof = false,
error = false,
registered = true,
flags = 1 '\001'
}

```

after restarting logread everything is ok

### OpenWrt version

r25665-79e9ce354e

### OpenWrt release

23.05

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

SolidRun Clearfog Pro A1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-04:
Can you repeat the issue on Openwrt, not vendor firmware
https://openwrt.org/releases/23.05/notes-23.05.0-rc1

brada4, 2024-04-04:
I mean supported release like 23.05.3 from firmware-selector. I cannot reproduce under common conditions 2-3 messages per second, restart logd, close remote udp port to send icmp  unreach. 
Like what (and how intense) are the last log messages when it flips?

a1bert01, 2024-04-04:
yes, we cannot reproduce it either. 

TL;DR
we are operating about 1000+ various openwrt routers on 2 different hw platforms (mvebu,ramips), some of them more than 7 years old. 6-7 months ago we started to experience this 'bug'. We know that:

it's not dependent on
-openwrt version (affects all from 17.01 - we are not runnig older)
-hw 
-uptime
-type/content of syslog message (no message is lost on remote syslog server - local buffer shows the same content as remote server)

BUT:
- it affects multiple routers at the same time often
- it started to affect old routers, that were not touched for several years 

from that   we assume it CAN be triggered by EXTERNAL EVENT

nbd168, 2024-04-04:
Please test if this patch helps: https://nbd.name/p/75a7fae0

nbd168, 2024-04-05:
Another hint based on reading the code: I'd expect that this can be triggered by getting a response data packet from the remote syslog server.

a1bert01, 2024-04-05:
yes, we thought so too, but we use stateless UDP and we were also monitoring incoming packets from remote syslog server to openwrt host but nothing was captured.

what we did not monitor is ICMP packets from routers on path to syslog server, they can emmit ICMP messages too, this is what we are preparing to do next

BTW: we have deployed patched logread to our routers and are waiting for result, thanks