openwrt/openwrt #15069: `pkg_hash_check_unresolved: cannot find dependency libubox20240126 for libuci20130104`

paulmenzel, 2024-04-04:
### Describe the bug

I think https://github.com/openwrt/openwrt/commit/89c594e8b295dbbcf32144d725f06cf1b90599c1introduced a build error:

```
$ git fetch
$ git rebase origin/main
$ git log --oneline --no-decorate -9
4ef13c4a49 libxml2: add host build dependency on libiconv-full
5aab43f933 hostapd: slightly clean up patches
7b9996d107 hostapd: replace "argument list too long" fix with a simpler version
6e391325af hostapd: remove workaround for broken WPA IEs in ancient devices
92379080ea hostapd: adjust patches to work with git am
3f5a9e80da hostapd: remove unused fix
24d0e74627 hostapd: bump to 2024-03-09
89c594e8b2 libubox: update to Git HEAD (2024-03-29)
b01d9bcc9d ipq-wifi: fixup hash
$ make -j32
[…]
Configuring ca-bundle.
Configuring fwtool.
Configuring urngd.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20240126 for libuci20130104
 * pkg_hash_check_unresolved: cannot find dependency libubox20240126 for libubus20231128
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 *      libubox20240126
 * opkg_install_cmd: Cannot install package base-files.
 * pkg_hash_check_unresolved: cannot find dependency libubox20240126 for libuclient20201210
 * pkg_hash_check_unresolved: cannot find dependency libubox20240126 for libustream-wolfssl20201210
make[2]: *** [package/Makefile:71: package/install] Error 255
make[2]: Leaving directory '/home/build/src/openwrt'
make[1]: *** [package/Makefile:124: /home/build/src/openwrt/staging_dir/target-mips_4kec_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/build/src/openwrt'
make: *** [/home/build/src/openwrt/include/toplevel.mk:233: world] Error 2
```

### OpenWrt version

4ef13c4a49

### OpenWrt release

4ef13c4a49

### OpenWrt target/subtarget

`CONFIG_TARGET_realtek_rtl838x_DEVICE_zyxel_gs1900-24e=y`

### Device

zyxel_gs1900-24e

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

paulmenzel, 2024-04-04:
cc: @nbd168

robimarko, 2024-04-05:
Fix is already merged:
https://github.com/openwrt/openwrt/commit/b906a9c78ef51cdc2f97dc6c39f92b9295f48d4e

paulmenzel, 2024-04-05:
Thank you. I saw that, and rebased. But I still get the error. Is there a way to just clean out that component to avoid cleaning all artifacts?

robimarko, 2024-04-05:
Maybe the buildbots have not yet caught up, it takes some time for them to rebuild

paulmenzel, 2024-04-06:
> Maybe the buildbots have not yet caught up, it takes some time for them to rebuild

There seems to be a misunderstanding. I am building on my machine, and don’t rely on buildbots.

robimarko, 2024-04-06:
Ah, then you can try `make package/libs/libubox/clean` first, but most likely cleaning everything with make clean would be a better idea

paulmenzel, 2024-04-06:
Neither of the tries below fixed the build error. Only `make clean`. :/

    $ make package/libs/libubox/clean
    $ make package/system/uci/clean
    $ make package/base-files/clean

closed 2024-04-06 by paulmenzel

Neustradamus, 2024-12-22:
Good news, there is a new merged PR linked to Zyxel GS1900:
- https://github.com/openwrt/openwrt/pull/16442