openwrt/openwrt #15071: ipq40xx: Colliding OTP patches with CONFIG_TESTING_KERNEL=y / kernel 6.6

hardfalcon, 2024-04-04:
### Describe the bug

When trying to build OpenWRT for an `ipq40xx` target with `CONFIG_TESTING_KERNEL=y` (which enables kernel 6.6.x), two patches collide on `drivers/mtd/mtdcore.c`, because they both try to change the `config.add_legacy_fixed_of_cells = true;` line in `static struct nvmem_device *mtd_otp_nvme`.

### OpenWrt version

main branch, commit id 000d25c31257e981309ea5d57c6c90337736f0bf

### OpenWrt release

main branch, commit id 000d25c31257e981309ea5d57c6c90337736f0bf

### OpenWrt target/subtarget

ipq40xx/mikrotik

### Device

mikrotik_sxtsq-5-ac

### Image kind

Self-built image

### Steps to reproduce

Select `CONFIG_TARGET_ipq40xx_mikrotik_DEVICE_mikrotik_sxtsq-5-ac=y` and `CONFIG_TESTING_KERNEL=y`, then run `make -j1 V=sc target/linux/compile`.

### Actual behaviour

The build fails at an early stage due to these two patches colliding:
- https://github.com/openwrt/openwrt/blob/main/target/linux/generic/pending-6.6/440-mtd-don-t-look-for-OTP-legacy-NVMEM-cells-if-proper-.patch
- https://github.com/openwrt/openwrt/blob/main/target/linux/ipq40xx/patches-6.6/110-mtd-limit-OTP-nvmem-to-non-nand-devices.patch

### Expected behaviour

The build shouldn't fail.

### Additional info

```
$ make -j1 V=sc target/linux/compile
```
[…]
```
Applying /tmp/openwrt/target/linux/ipq40xx/patches-6.6/104-clk-fix-apss-cpu-overclocking.patch using plaintext: 
patching file drivers/clk/qcom/gcc-ipq4019.c

Applying /tmp/openwrt/target/linux/ipq40xx/patches-6.6/110-mtd-limit-OTP-nvmem-to-non-nand-devices.patch using plaintext: 
patching file drivers/mtd/mtdcore.c
Hunk #1 FAILED at 933.
1 out of 1 hunk FAILED -- saving rejects to file drivers/mtd/mtdcore.c.rej
Patch failed!  Please fix /tmp/openwrt/target/linux/ipq40xx/patches-6.6/110-mtd-limit-OTP-nvmem-to-non-nand-devices.patch!
make[3]: *** [Makefile:34: /tmp/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_mikrotik/linux-6.6.24/.prepared_e36c3142507e08dfd1900e36a84eddac] Error 1
make[3]: Leaving directory '/tmp/openwrt/target/linux/ipq40xx'
make[2]: *** [Makefile:11: compile] Error 2
make[2]: Leaving directory '/tmp/openwrt/target/linux'
time: target/linux/compile#12.32#9.25#58.37
    ERROR: target/linux failed to build.
make[1]: *** [target/Makefile:32: target/linux/compile] Error 1
make[1]: Leaving directory '/tmp/openwrt'
make: *** [/tmp/openwrt/include/toplevel.mk:233: target/linux/compile] Error 2
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hardfalcon, 2024-04-04:
Tagging @rmilecki, author and committer of the other patch.

namiltd, 2024-04-04:
The fix for this bug is included in https://github.com/openwrt/openwrt/pull/15063