openwrt/openwrt #15072: Race condition with ccache and parallel builds

aduskett, 2024-04-04:
### Describe the bug

When using ccache, liblzo fails to compile the first time due to the following error:
`openwrt/staging_dir/host/bin/ccache: not found` when calling `make -j$(nproc)`

Calling `make -j$(nproc)` after the build failure without changing anything continues the build without issue. 

This seems like a pretty cut-and-dry race condition, although I am unsure where to start when debugging this problem.

### OpenWrt version

main branch

### OpenWrt release

main branch

### OpenWrt target/subtarget

TI-K3

### Device

AM62x-sk

### Image kind

Self-built image

### Steps to reproduce

  - Build the docker container based on Ubuntu 22.04 [here](https://github.com/aduskett/buildroot-docker-devel/tree/master/OpenWRT/Ubuntu/22.04): 
  - run `git clone https://github.com/aduskett/openwrt.git -b am62x-sk && cd openwrt && ./scripts/feeds update -a && ./scripts/feeds install -a && cd ../ && docker run --rm -v"$(pwd)":"$(pwd)":z -vopenwrt-ccache:/home/ccache --name "wrt-ubuntu22.04" -it "wrt-ubuntu22.04" /bin/bash`
  - cd to the OpenWRT source code directory
  - `mv k3_defconfig .config`
  -  `make -j$(nproc) V=s 2>&1 | tee build.log | grep -i -E "^make.*(error|[12345]...Entering dir)"`

### Actual behaviour

After ccache is built, host/lzo-2.10 fails to build with the error ``/bin/sh: 1: [...]/openwrt/staging_dir/host/bin/ccache: not found``

### Expected behaviour

ccache should be found and is indeed in the directory referenced above

### Additional info

I am porting the Texas Instruments K3 platform to OpenWRT.
  - This failure is unrelated.
  - I pull from the main openwrt branch and run `./scripts/feeds update -a && ./scripts/feeds install -a` every morning.

### Diffconfig

```text
CONFIG_TARGET_k3=y
CONFIG_TARGET_k3_cortexa53=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_k3_cortexa53_DEVICE_am62x-sk=y
CONFIG_TARGET_DEVICE_PACKAGES_k3_cortexa53_DEVICE_am62x-sk=""
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
CONFIG_BPF_TOOLCHAIN_HOST=y
CONFIG_BPF_TOOLCHAIN_HOST_PATH="/usr/bin"
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_CCACHE=y
CONFIG_CCACHE_DIR="/home/ccache/am62x"
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IB=y
CONFIG_IMAGEOPT=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
# CONFIG_KERNEL_IO_URING is not set
# CONFIG_KERNEL_KALLSYMS is not set
CONFIG_KERNEL_KEXEC=y
CONFIG_KERNEL_WERROR=y
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_PACKAGE_bnx2-firmware=m
CONFIG_PACKAGE_bnx2x-firmware=m
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_e100-firmware=m
# CONFIG_PACKAGE_kmod-pf-ring is not set
CONFIG_PACKAGE_libbpf=m
CONFIG_PACKAGE_libelf=m
CONFIG_PACKAGE_libiwinfo=y
CONFIG_PACKAGE_libiwinfo-data=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libwolfssl=y
CONFIG_PACKAGE_libxdp=m
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-ssl=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_mwifiex-pcie-firmware=m
CONFIG_PACKAGE_mwl8k-firmware=m
CONFIG_PACKAGE_px5g-mbedtls=y
CONFIG_PACKAGE_qosify=m
CONFIG_PACKAGE_r8169-firmware=m
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_rt2800-pci-firmware=m
CONFIG_PACKAGE_rtl8192ce-firmware=m
CONFIG_PACKAGE_rtl8192de-firmware=m
CONFIG_PACKAGE_rtl8192se-firmware=m
CONFIG_PACKAGE_rtl8723de-firmware=m
CONFIG_PACKAGE_rtl8821ae-firmware=m
CONFIG_PACKAGE_tc-full=m
CONFIG_PACKAGE_tc-mod-iptables=m
CONFIG_PACKAGE_uboot-envtools=m
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wil6210-firmware=m
CONFIG_PACKAGE_xdp-filter=m
CONFIG_PACKAGE_xdp-loader=m
CONFIG_PACKAGE_xdpdump=m
CONFIG_PACKAGE_zlib=m
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK=y
CONFIG_TARGET_ALL_PROFILES=y
CONFIG_TARGET_ROOTFS_TARGZ=y
CONFIG_USE_LLVM_HOST=y
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE=""
CONFIG_VERSION_CODE_FILENAMES=y
CONFIG_VERSION_DIST="OpenWrt"
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/releases/22.03.6"
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_WOLFSSL_HAS_NO_HW=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

aduskett, 2024-04-05:
Updated to use only the main branch + sunxi config.

thom24, 2025-12-17:
> Hit this with [fe635c5](https://github.com/openwrt/openwrt/commit/fe635c56acc7ae26978ce419e2fc50eb02699828).
> 
> I suspect running `make host/tools/ccache/compile` before the main build would be the workaround but I haven't done a full test.

wrong commit? 
`make tools/ccache/compile` is a valid workaround

vortexilation, 2026-10-13:
Root cause found: liblzo is in `tools-` (not `tools-y`) when only lzop is enabled, so it never gets the ccache dependency in tools/Makefile. Re-reported as #25870, fix in #25871.

vortexilation, 2026-10-13:
@aduskett 
Ping, fixes found, please check it out.