openwrt/openwrt #15079: gettext-full: build error: undefined reference to 'libiconv_open' and 'libiconv_close'

AcidSlide, 2024-04-05:
### Describe the bug

Doing fresh builds, the gettext-full package is getting errors

```
/usr/bin/ld: /home/acidslide/openwrt/openwrt/build_dir/hostpkg/gettext-0.21.1/gettext-tools/gnulib-lib/.libs/libgettextlib.a(rpl_la-encoding.o): in function `xmlIconvWrapper':
encoding.c:(.text+0xe69): undefined reference to `libiconv'
/usr/bin/ld: /home/acidslide/openwrt/openwrt/build_dir/hostpkg/gettext-0.21.1/gettext-tools/gnulib-lib/.libs/libgettextlib.a(rpl_la-encoding.o): in function `xmlFindCharEncodingHandler__internal_alias.part.0':
encoding.c:(.text+0x20c6): undefined reference to `libiconv_open'
/usr/bin/ld: encoding.c:(.text+0x20d4): undefined reference to `libiconv_open'
/usr/bin/ld: encoding.c:(.text+0x214b): undefined reference to `libiconv_open'
/usr/bin/ld: encoding.c:(.text+0x21e9): undefined reference to `libiconv_open'
/usr/bin/ld: encoding.c:(.text+0x2234): undefined reference to `libiconv_close'
/usr/bin/ld: encoding.c:(.text+0x223c): undefined reference to `libiconv_close'
/usr/bin/ld: /home/acidslide/openwrt/openwrt/build_dir/hostpkg/gettext-0.21.1/gettext-tools/gnulib-lib/.libs/libgettextlib.a(rpl_la-encoding.o): in function `xmlCharEncCloseFunc__internal_alias':
encoding.c:(.text+0x395a): undefined reference to `libiconv_close'
/usr/bin/ld: encoding.c:(.text+0x39ad): undefined reference to `libiconv_close'
```

### OpenWrt version

commit #46efc77502a811b768e9ddbf10e6cd8740abd227

### OpenWrt release

snapshot

### OpenWrt target/subtarget

rampis / mt7621 / kernel 6.6 (test kernel)

### Device

D-Team Newifi D2

### Image kind

Self-built image

### Steps to reproduce

do a `make -j1 V=sc` or `make package/libs/gettext-full V=sc` leads to error

### Additional info

Additional notes:  using current test kernel 6.6
OS used: Ubuntu 22.04 LTS

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_newifi-d2=y
CONFIG_DEVEL=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BATMAN_ADV_BATMAN_V=y
CONFIG_BATMAN_ADV_BLA=y
CONFIG_BATMAN_ADV_DAT=y
CONFIG_BATMAN_ADV_MCAST=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_BUSYBOX_CONFIG_FEATURE_DEFAULT_PASSWD_ALGO="md5"
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_HISTORY=1024
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_SAVEHISTORY=y
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_SAVE_ON_EXIT=y
CONFIG_BUSYBOX_CONFIG_FEATURE_REVERSE_SEARCH=y
CONFIG_COREMARK_ENABLE_MULTITHREADING=y
CONFIG_COREMARK_NUMBER_OF_THREADS=4
CONFIG_COREMARK_OPTIMIZE_O3=y
CONFIG_GNUTLS_ALPN=y
[... 339 lines cut ...]
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PACKAGE_zram-swap=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_TARGET_OPTIMIZATION="-O3 -pipe -mno-branch-likely -mips32r2 -mtune=24kc"
CONFIG_TARGET_OPTIONS=y
CONFIG_TESTING_KERNEL=y
CONFIG_ZRAM_DEF_COMP_LZORLE=y
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_kmod-crypto-sha256=y
# CONFIG_WPA_MBO_SUPPORT is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

AcidSlide, 2024-04-05:
#14819

AcidSlide, 2024-04-05:
confirming pr #14819 fixes the issue

robimarko, 2024-04-05:
PR #14819 merged, so closing as fixed.

closed 2024-04-05 by robimarko