openwrt/openwrt #15085: [bcm27xx|rpi4] sensors does not show 'cpu_thermal-virtual-0' but it does on raspbian

schuettecarsten, 2024-04-06:
### Describe the bug

For some reasons, `sensors` from lm-sensors package does not show `cpu_thermal-virtual-0` in OpenWRT with kernel-6.1, while it does on a standard Raspian setup. Looks like a kernel module or config value is missing in OpenWRT, but I have no idea, where...

Output in Openwrt:
```
gpio_fan-isa-0000
Adapter: ISA adapter
fan1:           0 RPM  (min =    0 RPM, max = 5000 RPM)

rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A
```

Output in Raspbian:
```
rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A

cpu_thermal-virtual-0
Adapter: Virtual device
temp1:        +57.5°C

gpio_fan-isa-0000
Adapter: ISA adapter
fan1:           0 RPM  (min =    0 RPM, max = 5000 RPM)
```

@Noltari any ideas how to fix this?

### OpenWrt version

r24944+894-906595c262

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Noltari, 2024-04-06:
@schuettecarsten which OpenWrt version are you using?
Is CONFIG_BCM2711_THERMAL enabled on your custom built image?

schuettecarsten, 2024-04-07:
I use the current snapshot and have some modifications in dnsmasq scripts which should not affect kernel settings at all.
The setting `CONFIG_BCM2711_THERMAL` is not in my config file, so it should take the default value from `/target/linux/bcm27xx/bcm2711/config-6.1`?

I use `collectd-mod-thermal` and it shows CPU temperature correctly from `/sys/class/thermal/thermal_zone0/temp`.

schuettecarsten, 2024-04-08:
I figured out that `CONFIG_HWMON` and `CONFIG_THERMAL_HWMON` are not set in kernel-6.1.
If I enable them, it works.

So maybe these should be added to kernel-6.1 default config for bcm27xx devices?

Noltari, 2024-04-08:
> I figured out that `CONFIG_HWMON` and `CONFIG_THERMAL_HWMON` are not set in kernel-6.1. If I enable them, it works.

If that's the issue, then installing `kmod-hwmon-core` and `kmod-thermal` should be enough to get it working.

> So maybe these should be added to kernel-6.1 default config for bcm27xx devices?

There's no need to enable those symbols if installing the suggested packages fixes the issue.

schuettecarsten, 2024-04-08:
`kmod-thermal` is off and cannot be set in menuconfig for bcm27xx as it depends on several ath10/ath11/mt7915e devices.

Noltari, 2024-08-20:
Reopen, since https://github.com/openwrt/openwrt/pull/16104 has been reverted due to https://github.com/openwrt/openwrt/issues/16191