openwrt/openwrt #15086: netifd: can not add phyX-mesh to bat0 when device boot up

PussAzuki, 2024-04-06:
### Describe the bug

netifd can not add phyX-mesh to bat0 when device boot up,more details here https://github.com/openwrt/routing/issues/1054

### OpenWrt version

r25589-f84ed09d2c

### OpenWrt release

SNAPSSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000 (OpenWrt U-Boot layout)

### Image kind

Self-built image

### Steps to reproduce

1.create a 802.11a mesh network with at least two devices.

2. make sure the mesh nodes have connect to each other.

3.change their config to using batman, and make sure they can still connect to each other.

4.reboot

### Actual behaviour

B.A.T.M.A.N does not transfer data until you restart the bat0 interface(?)

### Expected behaviour

they do transfer data after reboot done

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2024-04-06:
See also #14904

k3v3n, 2025-01-04:
experienced the same issue. Needed a script to reboot wifi after 60s on reboot to get the mesh working.

ecsv, 2025-09-05:
According to https://github.com/openwrt/openwrt/issues/14904#issuecomment-2649606720 , this was fixed in 24.10