openwrt/openwrt #15088: u-boot - LZMA ERROR 1

namiltd, 2024-04-06:
### Describe the bug

After installing PR https://github.com/openwrt/openwrt/pull/15078, router does not boot.
It displays message like below and restarts over and over again:
```
## Booting image at bc020000 ...

   Uncompressing Kernel Image ... LZMA ERROR 1 - must RESET board to recover

### OpenWrt version

SNAPSHOT

### OpenWrt release

r25829-8d934c1196

### OpenWrt target/subtarget

ramips/mt7620

### Device

TP-Link Archer C5 v4

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-04-06:
New PR https://github.com/openwrt/openwrt/commit/0a0d89265ee474d0784f675d65685ed011cc30b3 until the problem is resolved.

hauke, 2024-04-06:
Did the revert back to mkimage 2024.01 fix the problem?

Did this device ever boot for you with kernel 6.6? Maybe the kernel image is getting too big.

namiltd, 2024-04-06:
The router starts with the previous version of mkimage, it stops after uploading the new one (with exactly the same configuration).

namiltd, 2024-04-07:
The reasons are kernel size and dictionary size. After uploading the new mkimage, the kernel size increased only by 2kB but unfortunately it exceeded the limit.
dictionary size: 8388608 bytes, uncompressed size: 11866779 bytes
All I had to do was turn off one of the unused kernel modules, the size dropped by 4kB and the router started working.
This is not an update problem, the problem is kernel growth.
If the problem reoccurs, I will try to set the lzma parameters to make the dictionary size smaller.

closed 2024-04-07 by namiltd