openwrt/openwrt #15089: ipq40xx/generic - Lan access OK, Internet OK, WAN no access to ISP gateway

konradpieczynski, 2024-04-06:
### Describe the bug

Cant ping/accesss ISP gateway. Internet access works fine. also if i dont input wan gateway to lan bridge opkg cant access openwrt repo (in old version could leave it empty)

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys MR8300 (Dallas)

### Image kind

Official downloaded image

### Steps to reproduce

Laptop -> ISP gateway (can ping) -> Internet (All good)
Laptop -> Openwrt Router (can ping) -> ISP gateway (Laptop cant ping, OpenWRT can ping) -> Internet (All good, Laptop and OpenWRT can ping)

### Additional info

Also tried to move to same firewall zone both lan and wan. No changes.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-06:
Did you install BCP38?

konradpieczynski, 2024-04-06:
> Did you install BCP38?

Not installed.

brada4, 2024-04-06:
Please check while pinging CPE nat gateway:
`tcpdump -i wan icmp`
In principle it can be turned into pure network adapter as for openwrt to be on public internet (or their CGN)

konradpieczynski, 2024-04-06:
> Please check while pinging CPE nat gateway: `tcpdump -i wan icmp` In principle it can be turned into pure network adapter as for openwrt to be on public internet (or their CGN)

Modified network got some intresting results
Thats how was modified:
Laptop (192.168.1.50) -> OpenWRT 23.05.03 (192.168.1.1/24 with 192.168.2.212 ip by dhcp) -> wanport -> OpenWRT OpenWrt 22.03.1 (192.168.2.1) -> wan port -> ISP fiber modem (192.168.18.1)

Pinging 192.168.2.1 will result with: 
21:14:18.308390 IP 192.168.18.1 > 192.168.2.212: ICMP time exceeded in-transit, length 40
Pinging 192.168.18.1 will result with:
21:14:50.491917 IP 185-xxx-xxx-xxx.xx-xxx.xx > 192.168.2.212: ICMP host 192.168.1.50 unreachable, length 173
Pinging 8.8.8.8 will result with:
21:15:19.757362 IP 192.168.2.212 > dns.google: ICMP echo request, id 1, seq 4324, length 40
21:15:19.772569 IP dns.google > 192.168.2.212: ICMP echo reply, id 1, seq 4324, length 40

OpenWRT 22.03.1  router pings and connects to 192.168.18.1 without problems.

brada4, 2024-04-06:
`ip ro sh`

brada4, 2024-04-06:
Add 1-2ms latency for each chained nat device.

konradpieczynski, 2024-04-06:
> `ip ro sh`

/24

> Add 1-2ms latency for each chained nat device.

how?

brada4, 2024-04-06:
Openwrt as from reset acquires IP from wan using DHCP then provides own DHCP server with 192.168.1.1/24
Question - 
CPE >-< WAN(openwrt)LANX >-< PC
Ping CPE address from PC and have a look on traffic leaving/entering WAN. , it should be from WAN IP acquired via DHCP to the CPE IP, and response back.
You can observe states with `conntrack -E -p icmp`
Usually one created for all pings and close 30s later.

konradpieczynski, 2024-04-07:
> Openwrt as from reset acquires IP from wan using DHCP then provides own DHCP server with 192.168.1.1/24
> Question -
> CPE >-< WAN(openwrt)LANX >-< PC
> Ping CPE address from PC and have a look on traffic [...]

Laptop (192.168.2.50/24) -> OpenWRT(Wan dhcp 192.168.8.100/24) -> Wan (DHCP 192.168.8.1/24)

conntrack -E -p icmp

From laptop:
- Ping 192.168.8.1
contrack - empty
ping returned - Reply from 192.168.2.50: Destination host unreachable.
Ping 8.8.8.8
contrack -
    [NEW] icmp     1 30 src=192.168.2.50 dst=8.8.8.8 type=8 code=0 id=1 [UNREPLIED] src=8.8.8.8 dst=192.168.8.100 type=0 code=0 id=1
 [UPDATE] icmp     1 30 src=192.168.2.50 dst=8.8.8.8 type=8 code=0 id=1 src=8.8.8.8 dst=192.168.8.100 type=0 code=0 id=1
[DESTROY] icmp     1 src=192.168.2.50 dst=8.8.8.8 type=8 code=0 id=1 packets=4 bytes=240 src=8.8.8.8 dst=192.168.8.100 type=0 code=0 id=1 packets=4 bytes=240

from openwrt:
- Ping 192.168.8.1
Contrack -
    [NEW] icmp     1 30 src=192.168.8.100 dst=192.168.8.1 type=8 code=0 id=29589 [UNREPLIED] src=192.168.8.1 dst=192.168.8.100 type=0 code=0 id=29589
 [UPDATE] icmp     1 30 src=192.168.8.100 dst=192.168.8.1 type=8 code=0 id=29589 src=192.168.8.1 dst=192.168.8.100 type=0 code=0 id=29589
- Ping openwrt.org
    [NEW] icmp     1 30 src=192.168.8.100 dst=192.168.8.1 type=8 code=0 id=29589 [UNREPLIED] src=192.168.8.1 dst=192.168.8.100 type=0 code=0 id=29589
 [UPDATE] icmp     1 30 src=192.168.8.100 dst=192.168.8.1 type=8 code=0 id=29589 src=192.168.8.1 dst=192.168.8.100 type=0 code=0 id=29589

brada4, 2024-04-07:
Your PC somehow has .8. network cached locally so that it does not permit packets out. Flush routes there and re-run dhcp client.

konradpieczynski, 2024-04-07:
> Your PC somehow has .8. network cached locally so that it does not permit packets out. Flush routes there and re-run dhcp client.

route print

didnt return cached .8.

brada4, 2024-04-07:
The host rejects packet on its own :-S
If packet reached openwrt it would route to .8. network.
If the network is mostly silent you may tcpdump (wireshark if with windows on screen) the ARP packets caused by ping.
And if `ip ro fl ca` changes anything.

konradpieczynski, 2024-04-07:
> The host rejects packet on its own :-S If packet reached openwrt it would route to .8. network. If the network is mostly silent you may tcpdump (wireshark if with windows on screen) the ARP packets caused by ping. And if `ip ro fl ca` changes anything.

Broadcast	ARP	42	Who has 192.168.8.1? Tell 192.168.2.50
- No ARP reply.

brada4, 2024-04-07:
No way, it should seek router for address in other subnet.

konradpieczynski, 2024-04-07:
> No way, it should seek router for address in other subnet.

I built snapshoot for wrt1200ac and mr8300. EVerything works fine on wrt1200ac and it fails on mr8300. So its device specyfic problem.

brada4, 2024-04-07:
There is 0.00001 chance that device comes up with switch forwarding and PC acquires address from ISP CPE then changes to openwrt subnet. There is absolutely no way for DHCP to pass 2 subnets to a client at once.

brada4, 2024-04-07:
You need to tap seemingly broken device WAN port while rebooting multiple times just to see if PC DHCP leaks out.

konradpieczynski, 2024-04-07:
> There is 0.00001 chance that device comes up with switch forwarding and PC acquires address from ISP CPE then changes to openwrt subnet. There is absolutely no way for DHCP to pass 2 subnets to a client at once.

> You need to tap seemingly broken device WAN port while rebooting multiple times just to see if PC DHCP leaks out.

> Or CPE homepage suddenly shows dhcp lease for PC.

As i said  its device specyfic problem. Pluging m8300 to isp cant ping isp (but internet works). plugging wrt1200ac to isp can ping isp and internet works.

Also when i try to configure device eth0 connectity is broken totaly. Cant even ping openwrt on lan port.

Seems to be related to this change: "DO NOT upgrade from 22.03 to 23.05/main snapshot while keeping settings: config syntax isn't the same (due to DSA) and it will lead to a soft-brick. "

Becouse on 22.03 everything worked fine.

brada4, 2024-04-07:
There is absolutely no way that LAN PC gets any knowledge of WAN IP from DHCP via properly functioning OpenWRT.

eth0 is parent of all DSA ports where they connect to SoC CPU, you need to configure br-lan and similar.
You can check with `ip link` - it will be like lan5@eth0 for DSA children.

konradpieczynski, 2024-04-07:
> There is absolutely no way that LAN PC gets any knowledge of WAN IP from DHCP via properly functioning OpenWRT.
> 
> eth0 is parent of all DSA ports where they connect to SoC CPU, you need to configure br-lan and similar. You can check with `ip link` - it will be like lan5@eth0 for DSA children.

what got dhcp to packet routing?

- mr8300 22.03, wrt1200ac 22.03, 23.05, snapshoot - route correctly to 192.168.8.1 which is ip of wan interface
- mr8300 23.05, snapshoot - doesnt route correctly to 192.168.8.1 which is ip of wan interface

brada4, 2024-04-07:
> becomes unreachable.

not related to initial issue. any other vlan will work. obviously you need to add some ip config to br-lan.X to have any L3 connectivity.

if your pc does not emit packets how do you expect to route them.
what i susoect wan and br-lan mysteriously are not isolated.

konradpieczynski, 2024-04-07:
> > becomes unreachable.
> 
> not related to initial issue. any other vlan will work. obviously you need to add some ip config to br-lan.X to have any L3 connectivity.
> 
> if your pc does not emit packets how do you expect to route them. what i susoect wan and br-lan mysteriously are not isolated.

My pc tries to emit them. There is arp frame asking for route but mr8300 5.15/6.1 kernel dont answer.

138161, 2024-04-23:
opkg install owipcalc

/etc/hotplug.d/iface/80-reset-route6

#!/bin/sh
wan_dev="wan6"
[ "$HOTPLUG_TYPE" = "iface" ] || exit 0
[ "$INTERFACE" = "$wan_dev" ] || exit 0
RTMETRIC=127
. /lib/functions/network.sh
network_get_physdev lan_dev lan || exit 0
ifup_cb() {
        local _lan_dev="$1"
        local _metric="$2"
        local wan_subnet
        network_get_subnet6  wan_subnet  "$wan_dev" || return
        _wan_network=$(owipcalc "${wan_subnet}" network)
        ip -6 route replace "$_wan_network" dev "$_lan_dev" metric "$_metric"
}
ifdown_cb() {
        local _lan_dev="$1"
        local _metric="$2"
        ip -6 route flush dev "$_lan_dev" metric "$_metric"
}
case "$ACTION" in
        ifup)
                ifup_cb "$lan_dev" "$RTMETRIC"
                ;;
        ifdown)
                ifdown_cb "$lan_dev" "$RTMETRIC"
                ;;
        ifupdate)
                ifdown_cb "$lan_dev" "$RTMETRIC"
                sleep 1
                ifup_cb "$lan_dev" "$RTMETRIC"
                ;;
        *)
                ;;
esac
exit 0