openwrt/openwrt #15093: WAX206 - Realtek RTL8221B not working in kernel 6.6

lister-wrt, 2024-04-07:
### Describe the bug

Realtek RTL8221B detected as "Generic Clause 45 PHY" when booting kernel 6.6 and does not work (no link).
````
[    2.411591] mt7530-mdio mdio-bus:00: configuring for fixed/2500base-x link mode
[    2.420570] mt7530-mdio mdio-bus:00: Link is Up - 2.5Gbps/Full - flow control rx/tx
[    2.430087] mt7530-mdio mdio-bus:00 lan1 (uninitialized): PHY [mt7530-0:01] driver [MediaTek MT7531 PHY] (irq=133)
[    2.450659] mt7530-mdio mdio-bus:00 lan2 (uninitialized): PHY [mt7530-0:02] driver [MediaTek MT7531 PHY] (irq=134)
[    2.470869] mt7530-mdio mdio-bus:00 lan3 (uninitialized): PHY [mt7530-0:03] driver [MediaTek MT7531 PHY] (irq=135)
[    2.491066] mt7530-mdio mdio-bus:00 lan4 (uninitialized): PHY [mt7530-0:04] driver [MediaTek MT7531 PHY] (irq=136)
[    2.736934] mt7530-mdio mdio-bus:00 wan (uninitialized): PHY [mdio-bus:07] driver [Generic Clause 45 PHY] (irq=POLL)
[    2.748442] mtk_soc_eth 1b100000.ethernet eth0: entered promiscuous mode
````
When booting kernel 6.1 is is detected as "RTL8221B-VB-CG 2.5Gbps PHY" and works without issue. 
````
[    2.527082] mt7530-mdio mdio-bus:00: configuring for fixed/2500base-x link mode
[    2.536093] mt7530-mdio mdio-bus:00: Link is Up - 2.5Gbps/Full - flow control rx/tx
[    2.545596] mt7530-mdio mdio-bus:00 lan1 (uninitialized): PHY [mt7530-0:01] driver [MediaTek MT7531 PHY] (irq=133)
[    2.566339] mt7530-mdio mdio-bus:00 lan2 (uninitialized): PHY [mt7530-0:02] driver [MediaTek MT7531 PHY] (irq=134)
[    2.586594] mt7530-mdio mdio-bus:00 lan3 (uninitialized): PHY [mt7530-0:03] driver [MediaTek MT7531 PHY] (irq=135)
[    2.606760] mt7530-mdio mdio-bus:00 lan4 (uninitialized): PHY [mt7530-0:04] driver [MediaTek MT7531 PHY] (irq=136)
[    2.848736] mt7530-mdio mdio-bus:00 wan (uninitialized): PHY [mdio-bus:07] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)
[    2.860882] device eth0 entered promiscuous mode
````

6.6 full log
<details>

```[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.23 (@huon) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.2.0 r25803-b906a9c78e) 13.2.0, GNU ld (GNU Binutils) 2.42) #0 SMP Thu Apr  4 20:39:27 2024
[    0.000000] Machine model: Netgear WAX206
[    0.000000] earlycon: uart8250 at MMIO32 0x0000000011002000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
[    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] percpu: Embedded 18 pages/cpu s34536 r8192 d31000 u73728
[    0.000000] pcpu-alloc: s34536 r8192 d31000 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: earlycon=uart8250,mmio32,0x11002000 console=ttyS0,115200n8 swiotlb=512
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 2.
[    0.000000] software IO TLB: mapped [mem 0x000000005f4c0000-0x000000005f5c0000] (1MB)
[    0.000000] Memory: 499780K/524288K available (8640K kernel code, 904K rwdata, 2496K rodata, 384K init, 294K bss, 24508K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] workqueue: name exceeds WQ_NAME_LEN. Truncating to: events_freezable_power_efficien
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[... 762 lines cut ...]
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_tc-tiny=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_usteer=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mbedtls=y
CONFIG_PACKAGE_zlib=y
CONFIG_TARGET_OPTIMIZATION="-O2 -pipe -mcpu=cortex-a53"
CONFIG_TARGET_OPTIONS=y
CONFIG_TESTING_KERNEL=y
CONFIG_USE_LLVM_BUILD=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-01:
Paste "lsmod" from kernel 6.1 and kernel 6.6"

lister-wrt, 2024-05-04:
> Paste "lsmod" from kernel 6.1 and kernel 6.6"

Courtesy of jerrytouille on the forum

6.1
```# 'lsmod'
act_bpf                20480  0 
act_gact               16384  0 
act_mirred             20480  1 
act_skbedit            16384  0 
cfg80211              307200  5 mt7915e,mt7615_common,mt76_connac_lib,mt76,mac80211
chacha_neon            20480  1 libchacha20poly1305
cls_basic              16384  0 
cls_bpf                24576  0 
cls_flow               20480  0 
cls_fw                 16384  0 
cls_matchall           16384  0 
cls_route              20480  0 
cls_u32                24576  1 
compat                 16384  2 mac80211,cfg80211
em_u32                 16384  0 
gpio_button_hotplug    16384  0 
hwmon                  24576  2 mt7915e,mt7615_common
ifb                    16384  0 
ip_set                 40960 17 xt_set,ip_set_list_set,ip_set_hash_netportnet,ip_set_hash_netport,ip_set_hash_netnet,ip_set_hash_netiface,ip_set_hash_net,ip_set_hash_mac,ip_set_hash_ipportnet,ip_set_hash_ipportip,ip_set_hash_ipport,ip_set_hash_ipmark,ip_set_hash_ipmac,ip_set_hash_ip,ip_set_bitmap_port,ip_set_bitmap_ipmac,ip_set_bitmap_ip
ip_set_bitmap_ip       20480  0 
ip_set_bitmap_ipmac    20480  0 
ip_set_bitmap_port     20480  0 
ip_set_hash_ip         36864  0 
ip_set_hash_ipmac      36864  0 
ip_set_hash_ipmark     36864  0 
ip_set_hash_ipport     36864  0 
ip_set_hash_ipportip   36864  0 
ip_set_hash_ipportnet   40960  0 
ip_set_hash_mac        24576  0 
ip_set_hash_net        40960  0 
ip_set_hash_netiface   40960  0 
ip_set_hash_netnet     40960  0 
ip_set_hash_netport    40960  0 
ip_set_hash_netportnet   45056  0 
ip_set_list_set        20480  0 
ip_tables              28672  2 iptable_mangle,iptable_filter
ip_tunnel              32768  1 sit
ip6_tables             28672  2 ip6table_mangle,ip6table_filter
ip6_udp_tunnel         16384  1 wireguard
ip6t_REJECT            16384  0 
ip6table_filter        16384  0 
ip6table_mangle        16384  0 
ipt_ECN                16384  0 
ipt_REJECT             16384  0 
iptable_filter         16384  0 
iptable_mangle         16384  0 
leds_gpio              16384  0 
libchacha              16384  1 chacha_neon
libchacha20poly1305    16384  1 wireguard
libcrc32c              16384  1 nf_tables
libcurve25519_generic   40960  1 wireguard
mac80211              589824  5 mt7915e,mt7615e,mt7615_common,mt76_connac_lib,mt76
mt76                   77824  4 mt7915e,mt7615e,mt7615_common,mt76_connac_lib
mt76_connac_lib        53248  3 mt7915e,mt7615e,mt7615_common
mt7615_common          86016  1 mt7615e
mt7615e                24576  0 
mt7915e               131072  0 
nf_conntrack           77824  7 nft_redir,nft_nat,nft_masq,nft_flow_offload,nft_ct,nf_nat,nf_flow_table
nf_defrag_ipv4         16384  1 nf_conntrack
nf_defrag_ipv6         20480  1 nf_conntrack
nf_flow_table          32768  2 nft_flow_offload,nf_flow_table_inet
nf_flow_table_inet     16384  1 
nf_log_syslog          20480  0 
nf_nat                 36864  4 nft_redir,nft_nat,nft_masq,nft_chain_nat
[... 172 lines cut ...]
tun                    45056  0 
tunnel4                12288  1 sit
udp_tunnel             16384  1 wireguard
wireguard              65536  0 
x_tables               24576 29 ipt_REJECT,xt_time,xt_tcpudp,xt_tcpmss,xt_statistic,xt_multiport,xt_mark,xt_mac,xt_limit,xt_length,xt_hl,xt_ecn,xt_dscp,xt_comment,xt_TCPMSS,xt_LOG,xt_HL,xt_DSCP,xt_CLASSIFY,nft_compat,iptable_mangle,iptable_filter,ipt_ECN,ip_tables,xt_set,ip6table_mangle,ip6table_filter,ip6_tables,ip6t_REJECT
xt_CLASSIFY            12288  0 
xt_DSCP                12288  0 
xt_HL                  12288  0 
xt_LOG                 12288  0 
xt_TCPMSS              12288  0 
xt_comment             12288  0 
xt_dscp                12288  0 
xt_ecn                 12288  0 
xt_hl                  12288  0 
xt_length              12288  0 
xt_limit               12288  0 
xt_mac                 12288  0 
xt_mark                12288  0 
xt_multiport           12288  0 
xt_set                 16384  0 
xt_statistic           12288  0 
xt_tcpmss              12288  0 
xt_tcpudp              12288  0 
xt_time                12288  0
````

namiltd, 2024-05-04:
Can you revert https://github.com/openwrt/openwrt/commit/f6c27b22f6f84b13ad230dcf3afa7f11efe7a8bc and test?

dangowrt, 2024-05-10:
Please try this patch
```diff
--- a/target/linux/generic/pending-6.6/730-net-phy-realtek-detect-early-version-of-RTL8221B.patch
+++ b/target/linux/generic/pending-6.6/730-net-phy-realtek-detect-early-version-of-RTL8221B.patch
@@ -33,7 +33,7 @@ Signed-off-by: Daniel Golle <daniel@makrotopia.org>
 +	int val;
 +	u32 id;
 +
-+	if (phydev->is_c45) {
++	if (phydev->mdio.bus->read_c45) {
 +		if (phydev->c45_ids.device_ids[1])
 +			return phydev->c45_ids.device_ids[1] == RTL_8221B_VB_CG;
 +	} else {
```

RolandoMagico, 2024-05-10:
> Please try this patch

No improvement on Cudy M3000

enmaskarado, 2024-05-10:
> Please try this patch

Also does not work on Cudy TR3000. Dmesg still show Generic Clause 45 PHY.

Something new: if I connect a cable now it does appear in logread, but it always negotiates 2.5G (when the port at the other end is 100M) and there is no connection.

dangowrt, 2024-05-10:
Sad.
Please install `mdio-tools` or prepare a build including this package (from packages feed). It will allow us to introspect the MDIO bus and see if and how the PHY identifies and why the driver doesn't like it.
```
PHY_ID=7
for mmddev in $(seq 0 15); do
  echo ID for MMD $mmddev
  mdio mdio-bus mmd ${PHY_ID}:${mmddev} 2
  mdio mdio-bus mmd ${PHY_ID}:${mmddev} 3
done
```

RolandoMagico, 2024-05-10:
Output on M3000 with PHY_ID=1:
```
ID for MMD 0
0x0000
0x0000
ID for MMD 1
0x001c
0xc849
ID for MMD 2
0x0000
0x0000
ID for MMD 3
0x001c
0xc849
ID for MMD 4
0x0000
0x0000
ID for MMD 5
0x0000
0x0000
ID for MMD 6
0x0000
0x0000
ID for MMD 7
0x001c
0xc849
ID for MMD 8
0x0000
0x0000
ID for MMD 9
0x0000
0x0000
ID for MMD 10
0x0000
0x0000
ID for MMD 11
0x0000
0x0000
ID for MMD 12
0x0000
0x0000
ID for MMD 13
0x0000
0x0000
ID for MMD 14
0x0000
0x0000
ID for MMD 15
0x0000
0x0000

```

dangowrt, 2024-05-10:
> Output on M3000 with PHY_ID=1:
> 
> [all zeros]

So this looks like the PHY is not taken out from reset, as it is just not found at all. To be really sure, let please try again with the changed debugging script above (edited)

RolandoMagico, 2024-05-10:
Output of changed script with M3000:
```
ID for MMD 0
0x0000
0x0000
0x0000
0x0000
ID for MMD 1
0x001c
0xc849
0x001c
0xc849
ID for MMD 2
0x0000
0x0000
0x0000
0x0000
ID for MMD 3
0x001c
0xc849
0x001c
0xc849
ID for MMD 4
0x0000
0x0000
0x0000
0x0000
ID for MMD 5
0x0000
0x0000
0x0000
0x0000
ID for MMD 6
0x0000
0x0000
0x0000
0x0000
ID for MMD 7
0x001c
0xc849
0x001c
0xc849
ID for MMD 8
0x0000
0x0000
0x0000
0x0000
ID for MMD 9
0x0000
0x0000
0x0000
0x0000
ID for MMD 10
0x0000
0x0000
0x0000
0x0000
ID for MMD 11
0x0000
0x0000
0x0000
0x0000
ID for MMD 12
0x0000
0x0000
0x0000
0x0000
ID for MMD 13
0x0000
0x0000
0x0000
0x0000
ID for MMD 14
0x0000
0x0000
0x0000
0x0000
ID for MMD 15
0x0000
0x0000
0x0000
0x0000
```

dangowrt, 2024-05-10:
May reset needs to be moved from the PHY to the MDIO bus, ie try something like
```diff
diff --git a/target/linux/mediatek/dts/mt7622-netgear-wax206.dts b/target/linux/mediatek/dts/mt7622-netgear-wax206.dts
index 524a498740..02e2280ffc 100644
--- a/target/linux/mediatek/dts/mt7622-netgear-wax206.dts
+++ b/target/linux/mediatek/dts/mt7622-netgear-wax206.dts
@@ -140,6 +140,10 @@
 	};
 
 	mdio-bus {
+		reset-gpios = <&pio 101 GPIO_ACTIVE_LOW>;
+		reset-assert-us = <100000>;
+		reset-deassert-us = <100000>;
+
 		#address-cells = <1>;
 		#size-cells = <0>;
 
@@ -202,10 +206,7 @@
 		rtl8221b_phy: ethernet-phy@7 {
 			compatible = "ethernet-phy-ieee802.3-c45";
 			reg = <7>;
-			reset-gpios = <&pio 101 GPIO_ACTIVE_LOW>;
 			interrupts = <52 IRQ_TYPE_LEVEL_HIGH>;
-			reset-assert-us = <100000>;
-			reset-deassert-us = <100000>;
 		};
 	};
 };
```

RolandoMagico, 2024-05-10:
What I did for the M3000:
```diff
--- a/target/linux/mediatek/dts/mt7981b-cudy-m3000-v1.dts
+++ b/target/linux/mediatek/dts/mt7981b-cudy-m3000-v1.dts
@@ -102,15 +102,16 @@
 };
 
 &mdio_bus {
+    reset-gpios = <&pio 39 GPIO_ACTIVE_LOW>;
+    reset-assert-us = <100000>;
+    reset-deassert-us = <100000>;
+
        rtl8221b_phy: ethernet-phy@1 {
                compatible = "ethernet-phy-ieee802.3-c45";
                reg = <1>;
 
-               reset-gpios = <&pio 39 GPIO_ACTIVE_LOW>;
-
                interrupts = <38 IRQ_TYPE_LEVEL_LOW>;
-               reset-assert-us = <100000>;
-               reset-deassert-us = <100000>;
+               
        };
 };

```

But no improvement regardng the WAN port.

yogo1212, 2024-05-10:
@RolandoMagico isn't the 2.5Gb port working for you on the M3000?

enmaskarado, 2024-05-10:
I followed the same steps for the Cudy TR3000, including a very similar patch for the DTS. It still doesn't work, and in dmesg it continues showing "Generic Clause 45 PHY".

RolandoMagico, 2024-05-10:
> @RolandoMagico isn't the 2.5Gb port working for you on the M3000?

Yes, the 2,5G port is currently not working with kernel 6.6

enmaskarado, 2024-05-10:
Out of desperation I hacked `compatible = "ethernet-phy-ieee802.3-c45";` to `compatible = "ethernet-phy-id001c.c849";`. 

Now the 2.5G port works fine, but idk if this solution is upstreamable

dangowrt, 2024-05-10:
> Idk if this solution is upstreamable but I hope that at least this will help to discover the real bug.

It's not ideal as the PHY ID *does* read correctly at a later point when trying from userspace. It most likely has to do something with the reset (either polarity or timing), and fixing that would be better imho. But also hard-coded PHY IDs are acceptable and actually better than driver patches sometimes.

namiltd, 2024-05-10:
Question: If we roll back https://github.com/openwrt/openwrt/commit/f6c27b22f6f84b13ad230dcf3afa7f11efe7a8bc then at what point does the loading error occur? Could someone include the log from the rs232 console? I'm curious which function causes the error.

yogo1212, 2024-05-10:
@namiltd patch replaced, build is running.

is the serial output important?
are you looking for something that `dmesg | grep mtk_soc_eth` doesnt contain?

yogo1212, 2024-05-10:
At any rate: Without the patch, link is detected and the interface works.

This is the output of `dmesg | grep mtk_soc_eth`:
```
[    1.437898] mtk_soc_eth 15100000.ethernet eth0: mediatek frame engine at 0xffffffc081480000, irq 75
[    1.447801] mtk_soc_eth 15100000.ethernet eth1: mediatek frame engine at 0xffffffc081480000, irq 75
[    3.955490] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)
[    3.968354] mtk_soc_eth 15100000.ethernet eth0: configuring for phy/2500base-x link mode
[    7.116102] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   23.687486] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[   23.697121] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/gmii link mode
[   23.717105] mtk_soc_eth 15100000.ethernet eth1: entered allmulticast mode
[   23.724088] mtk_soc_eth 15100000.ethernet eth1: entered promiscuous mode
[   23.732517] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[   24.145491] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)
[   24.155680] mtk_soc_eth 15100000.ethernet eth0: configuring for phy/sgmii link mode
[   24.206755] mtk_soc_eth 15100000.ethernet eth0: entered allmulticast mode
[   24.213702] mtk_soc_eth 15100000.ethernet eth0: entered promiscuous mode
[   27.355968] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```

namiltd, 2024-05-10:
Please test https://github.com/namiltd/openwrt/commit/8dda850b31f3af87cef3d276363cc108b4b0d0c9

namiltd, 2024-05-10:
@yogo1212 I wanted data from the serial console because I assumed that my correction would not work, and here's a surprise, the chip is detected correctly.

enmaskarado, 2024-05-11:
Works on a Cudy TR3000. mtk_show_eth shows now `RTL8221B-VB-CG 2.5Gbps PHY`

RolandoMagico, 2024-05-11:
Thanks @namiltd ,
the patch works on M3000 as well. I also see the change in the system log:
`PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=POLL)`

gca31, 2024-05-11:
Thanks all. Working like a charm on Netgear wax206

namiltd, 2024-05-11:
@gca31 If it works, please approve changes in PR (File changed->Review changes->Approve).

closed 2024-05-13 by openwrt-bot

dangowrt, 2024-06-05:
Please test https://github.com/openwrt/openwrt/pull/15634 and let me know if it works on WAX206.

LuisMitaHL, 2024-08-17:
Hi,
I installed `r27160-b72c4b5386` on my Cudy TR3000 with RTL8221B and i'ts broken again

`[   16.035421] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:01] driver [Generic Clause 45 PHY] (irq=POLL)`

namiltd, 2024-08-18:
The problem exists since commithttps://github.com/openwrt/openwrt/commit/ca4469045f6a4845dbf8aeabd4b04b421209f66e which removed too much.Message ID: ***@***.***>

namiltd, 2024-08-18:
The problem exists after the commit https://github.com/openwrt/openwrt/commit/ca4469045f6a4845dbf8aeabd4b04b421209f66e, in which the previous ptches were removed and replaced with new ones, apparently not fully tested.

namiltd, 2024-08-18:
File target/linux/generic/pending-6.6/730-net-phy-realtek-detect-early-version-of-RTL8221B.patch to test: 
[730-net-phy-realtek-detect-early-version-of-RTL8221B.patch](https://github.com/user-attachments/files/16651473/730-net-phy-realtek-detect-early-version-of-RTL8221B.patch)

LuisMitaHL, 2024-08-19:
I tested the patch, works fine now.

```
[   16.786301] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY (C45)] (irq=POLL)
[   16.797014] mtk_soc_eth 15100000.ethernet eth0: configuring for phy/2500base-x link mode
[   19.996548] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
```

Lines 28 and 30 of the patch did not have semicolons at the end, and I had to put them in.

namiltd, 2024-08-19:
> Lines 28 and 30 of the patch did not have semicolons at the end, and I had to put them in.

Sorry, I updated the patch.

namiltd, 2024-08-19:
Fixed: https://github.com/openwrt/openwrt/commit/052732bddb4bbe2f7c3c47700f825d8f1e7b9858

commit 76584c79 (2024-05-13): generic: 6.6: fix realtek PHY detection patch

Fixes the issue of RTL8221B-VB-CG not being detected correctly.
Reverts changes from f6c27b2, leaving only the read_c45 test.

Fixed: #15093
PR #15453 (2024-05-13): generic: fix realtek PHY detection patch
Fixes the issue of RTL8221B-VB-CG not being detected correctly.
Reverts changes from f6c27b2, leaving only the read_c45 test.
Additionally, the constant RTL_8221B_VB_CG_PHYID is used instead of a numeric value for all kernels.

Fixed: #15093