openwrt/openwrt #15097: x86-64: WAN and LAN interfaces swapped after upgrading from 23.05.2 to 23.05.3

sk2449, 2024-04-07:
### Describe the bug

eth0 was LAN (built-in NIC), eth1 was WAN (USB dongle powered by kmod-usb-net-rtl8152) before the upgrade.
Now it's the other way round (eth0 became eth1 and vice versa), so i lost access to the device temporarily and had to adapt /etc/config/network by attaching screen and keyboard.

Build command used with openwrt-imagebuilder-23.05.3-x86-64.Linux-x86_64.tar.xz:
`make image PROFILE="generic" PACKAGES="kmod-usb-net-rtl8152 openssh-server luci-ssl luci-app-ddns ddns-scripts tcpdump -dropbear -ppp -ppp-mod-pppoe -luci-proto-ppp"`

Is device detection/naming not deterministic any more? I'm using this setup since OpenWrt 21.XX and never ran into such deal breakers during upgrade before.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

x86/64

### Device

Intel Atom x5-Z8350 based x86-64 machine

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-07:
2-version jump is explicitly not supported. You had to install v22 interim.

brada4, 2024-04-09:
Can you check lspci - like USB controller has lower PCI ID than netcard, then with both drivers present eth0 will be usb, but installing driver package it will be second provided onboard card is built in kernel.
rich names are from systemd, not kernel [REF](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/networking_guide/ch-consistent_network_device_naming#sec-Naming_Schemes_Hierarchy)

sk2449, 2024-04-13:
@brada4 The USB controller indeed has a lower PCI ID:

```
00:00.0 Host bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series SoC Transaction Register (rev 36)
00:02.0 VGA compatible controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Integrated Graphics Controller (rev 36)
00:0b.0 Signal processing controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Power Management Controller (rev 36)
00:14.0 USB controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series USB xHCI Controller (rev 36)
00:1a.0 Encryption controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Trusted Execution Engine (rev 36)
00:1c.0 PCI bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series PCI Express Port #1 (rev 36)
00:1f.0 ISA bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series PCU (rev 36)
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
```
The driver of the built-in NIC is definitely included in the kernel, because i never had to install additional driver packages for that one.
Still wondering, what of this might have changed with the latest minor update though.

brada4, 2024-04-13:
It was/is racy all the time. Probably drop a line in forum wishlist to have consistent naming for wired networks, it kind of is there for wifis.