openwrt/openwrt #15103: [dnsmasq-full]: conflict setup for dnsmasq and dnsmasq-full should exist

graysky2, 2024-04-08:
### Describe the bug

Selecting `dnsmasq-full` from a `make menuconfig` does not deselect `dnsmasq` which results in conflicting files.

### OpenWrt version

r0+25861-6db44a4f62

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

HC Technology.,Ltd. HCAR5000-MI

### Image kind

Self-built image

### Steps to reproduce

1. Select `dnsmasq-full` in the menuconfig

### Actual behaviour

Upon enabling `dnsmasq-full`, `dnsmasq` remains selected.

### Expected behaviour

Enabling `dnsmasq-full` should unselect `dnsmasq`

### Additional info

Here is the resulting error when building both packages:
```
% make
...
Collected errors:
 * check_data_file_clashes: Package dnsmasq-full wants to install file /scratch/union/build_dir/target-x86_64_musl/root-x86/etc/hotplug.d/ntp/25-dnsmasqsec
        But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /scratch/union/build_dir/target-x86_64_musl/root-x86/etc/init.d/dnsmasq
        But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /scratch/union/build_dir/target-x86_64_musl/root-x86/etc/uci-defaults/50-dnsmasq-migrate-ipset.sh
        But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /scratch/union/build_dir/target-x86_64_musl/root-x86/etc/uci-defaults/50-dnsmasq-migrate-resolv-conf-auto.sh
        But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /scratch/union/build_dir/target-x86_64_musl/root-x86/usr/lib/dnsmasq/dhcp-script.sh
        But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /scratch/union/build_dir/target-x86_64_musl/root-x86/usr/sbin/dnsmasq
        But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /scratch/union/build_dir/target-x86_64_musl/root-x86/usr/share/acl.d/dnsmasq_acl.json
        But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /scratch/union/build_dir/target-x86_64_musl/root-x86/usr/share/dnsmasq/dhcpbogushostname.conf
        But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /scratch/union/build_dir/target-x86_64_musl/root-x86/usr/share/dnsmasq/rfc6761.conf
        But that file is already provided by package  * dnsmasq
 * opkg_install_cmd: Cannot install package dnsmasq-full.
make[2]: *** [package/Makefile:71: package/install] Error 255
```

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
[... 383 lines cut ...]
CONFIG_PACKAGE_zstd=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_PCRE_JIT_ENABLED=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_TARGET_EXT4_RESERVED_PCT=1
CONFIG_TARGET_KERNEL_PARTSIZE=32
CONFIG_TARGET_OPTIMIZATION="-march=znver3 -O3 -pipe"
CONFIG_TARGET_OPTIONS=y
CONFIG_TARGET_ROOTFS_PARTSIZE=512
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_TESTING_KERNEL=y
CONFIG_ZSTD_OPTIMIZE_O3=y
# CONFIG_PACKAGE_dnsmasq_full_dhcpv6 is not set
CONFIG_PACKAGE_kmod-pcs-xpcs=y
CONFIG_PACKAGE_kmod-phylink=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

graysky2, 2024-04-08:
My versions always look like that.  My fork of openwrt has a few trivial changes to the kernel config file.  Could that be what causes the invalid version?

ddubnyk, 2024-06-05:
I'm also experiencing the same error when building on a brand new setup.

Also not sure if its related to this, but:

service dnsmasq start

Does absolutely nothing and dnsmasq is never detected as running.  Simply running: 

dnsmasq

At the command line while logged in succesfully starts dnsmasq and it works as intended.

My setup:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.3 r23809-234f1a2efa'
DISTRIB_TAINTS='no-all'

[config.txt](https://github.com/user-attachments/files/15572388/config.txt)

ddubnyk, 2024-10-02:
This is still broken on the latest 23.05.5 release FYI.  I did a full fresh clone and build and it still does not deselect dnsmasq.

closed 2025-10-19 by graysky2

graysky2, 2025-10-19:
Not sure where it was fixed but I cannot reproduce.