openwrt/openwrt #15104: D-Team Newifi-d2 kernel error and call trace upon boot

spamsool, 2024-04-08:
### Describe the bug

I flashed the latest OpenWRT 23.05 from https://downloads.openwrt.org/releases/23.05-SNAPSHOT/targets/ramips/mt7621/ and I receive the errors with call trace upon every boot in dmesg.

[    1.389181] ------------[ cut here ]------------
[    1.393852] WARNING: CPU: 1 PID: 1 at drivers/mtd/spi-nor/core.c:2863 0x80470f84
[    1.401228] enabling reset hack; may not recover from unexpected reboots
[    1.407908] Modules linked in:
[    1.410963] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 5.15.153 #0
[    1.417037] Stack : 80790000 8154d800 80e8d910 80082ba0 00000000 00000000 00000000 00000000
[    1.425395]         00000000 00000000 00000000 00000000 00000000 00000001 81453908 81480620
[    1.433749]         814539a0 00000000 00000000 814537b0 00000038 8039e0c4 ffffffea 00000000
[    1.442103]         814537bc 0000009c 8081cab0 ffffffff 8073ae2c 814538e8 00000000 80470f84
[    1.450454]         00000009 80e8d910 80790000 8154d800 00000018 804102e4 00000004 809d0004
[    1.458809]         ...
[    1.461249] Call Trace:
[    1.461260] [<80082ba0>] 0x80082ba0
[    1.467165] [<8039e0c4>] 0x8039e0c4
[    1.470640] [<80470f84>] 0x80470f84
[    1.474114] [<804102e4>] 0x804102e4
[    1.477590] [<80007898>] 0x80007898
[    1.481062] [<800078a0>] 0x800078a0
[    1.484534] [<80470f84>] 0x80470f84
[    1.488009] [<80382188>] 0x80382188
[    1.491481] [<80720000>] 0x80720000
[    1.494955] [<8002cf2c>] 0x8002cf2c
[    1.498432] [<80470f84>] 0x80470f84
[    1.501903] [<8002d040>] 0x8002d040
[    1.505380] [<80470f84>] 0x80470f84
[    1.508858] [<80471818>] 0x80471818
[    1.512333] [<80472558>] 0x80472558
[    1.515809] [<80429154>] 0x80429154
[    1.519285] [<803a4d64>] 0x803a4d64
[    1.522756] [<80493d18>] 0x80493d18
[    1.526229] [<8048d86c>] 0x8048d86c
[    1.529707] [<80423454>] 0x80423454
[    1.533180] [<804dd8a4>] 0x804dd8a4
[    1.536657] [<80423988>] 0x80423988
[    1.540127] [<8048d960>] 0x8048d960
[    1.543603] [<80423fe0>] 0x80423fe0
[    1.547080] [<80423f10>] 0x80423f10
[    1.550552] [<80420ef4>] 0x80420ef4
[    1.554027] [<80423d04>] 0x80423d04
[    1.557503] [<8041f374>] 0x8041f374
[    1.560974] [<80422354>] 0x80422354
[    1.564449] [<8041f7b8>] 0x8041f7b8
[    1.567928] [<8048d5d8>] 0x8048d5d8
[    1.571401] [<80491bc8>] 0x80491bc8
[    1.574874] [<8041b18c>] 0x8041b18c
[    1.578352] [<80491d08>] 0x80491d08
[    1.581825] [<804926ac>] 0x804926ac
[    1.585298] [<803f5f14>] 0x803f5f14
[    1.588778] [<8089e094>] 0x8089e094
[    1.592248] [<804951e8>] 0x804951e8
[    1.595725] [<80425c84>] 0x80425c84
[    1.599202] [<80423454>] 0x80423454
[    1.602672] [<8089e094>] 0x8089e094
[    1.606144] [<804dd8a4>] 0x804dd8a4
[    1.609621] [<80423988>] 0x80423988
[    1.613094] [<80425d2c>] 0x80425d2c
[    1.616572] [<80424118>] 0x80424118
[    1.620045] [<8042406c>] 0x8042406c
[    1.623517] [<80420e48>] 0x80420e48
[    1.626995] [<80422628>] 0x80422628
[    1.630464] [<80721488>] 0x80721488
[    1.633937] [<808b89d4>] 0x808b89d4
[    1.637416] [<80424a80>] 0x80424a80
[    1.640887] [<808b9348>] 0x808b9348
[    1.644361] [<808b93f8>] 0x808b93f8
[    1.647837] [<8000157c>] 0x8000157c
[    1.651310] [<8004e420>] 0x8004e420
[    1.654784] [<80003631>] 0x80003631
[    1.658263] [<8089f0b4>] 0x8089f0b4
[    1.661736] [<8089e094>] 0x8089e094
[    1.665212] [<8071ab28>] 0x8071ab28
[    1.668688] [<8071ab08>] 0x8071ab08
[    1.672157] [<80002f98>] 0x80002f98
[    1.675645] 
[    1.677203] ---[ end trace 6c539f7cd178ca5b ]---

### OpenWrt version

r23821-77b7c10884

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

D-Team Newifi D2

### Image kind

Official downloaded image

### Steps to reproduce

I just flashed the fresh OpenWRT image and I get this error upon every boot. No action required.

### Actual behaviour

[d-team_newifi-d2_dmesg.txt](https://github.com/openwrt/openwrt/files/14906440/d-team_newifi-d2_dmesg.txt)

I attach a txt file with dmesg and errors.

### Expected behaviour

I suppose there should be no error and trace.

### Additional info

I was previously using Padavan firmware fot this device so I have no previous experience with other OpenWRT builds on this box.
I have just decided to switch to OpenWRT.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

spamsool, 2024-04-23:
> Invalid Release reported. `23.05-SNAPSHOT` Is this from a clean repository?

Yes

DragonBluep, 2024-04-23:
This is a harmless warning as expected. You can ignore it.

closed 2025-08-20 by spamsool