openwrt/openwrt #15113: libubox20240126  Cannot install package

openips, 2024-04-09:
### Describe the bug

with git https://github.com/openwrt/openwrt/commit/d997477775bf5447000f1d300a027878b566f5d1 and use test kernel ，build failed with log 
```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20240126 for libuci20130104
 * pkg_hash_check_unresolved: cannot find dependency libubox20240126 for libubus20231128
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 *      libubox20240126
 * opkg_install_cmd: Cannot install package base-files.
 * pkg_hash_check_unresolved: cannot find dependency libubox20240126 for libuclient20201210
 * pkg_hash_check_unresolved: cannot find dependency libubox20240126 for libustream-openssl20201210
make[2]: *** [package/Makefile:71: package/install] Error 255
make[2]: Leaving directory '/LEDE/openwrt-r4'
make[1]: *** [package/Makefile:124: /LEDE/openwrt-r4/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
```

### OpenWrt version

r0+25733-7236d4f82b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4 & X64

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

openips, 2024-04-09:
> 报告的版本无效。`r0+25733-7236d4f82b` 这是来自干净的存储库吗？

build by myself

ynezz, 2024-04-09:
Yeah, this is unfortunate fallout, sorry about that, I've hit that as well, it helped me to remove all the mentioned packages and build again, so basically `rm -fr build_dir/target-aarch64_cortex-a53_musl/{uci*,ubus*,uclient*,ustream*}` and build again.

closed 2024-04-09 by ynezz