openwrt/openwrt #15115: bcm53xx: Netgear R8500 USB not working

sbwml, 2024-04-09:
### Describe the bug

The USB port is completely non-functional.

I plugged in a USB storage device and executed the `lsusb` command.

```
Bus 002 Device 001: ID 1d6b:0002 Linux 6.6.25 ehci_hcd EHCI Host Controller
Bus 001 Device 001: ID 1d6b:0001 Linux 6.6.25 ohci_hcd Generic Platform OHCI controller
Bus 003 Device 001: ID 1d6b:0003 Linux 6.6.25 xhci-hcd xHCI Host Controller
```

<details>
<summary>dmesg ↓</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.25 (admin@cooluc.com) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 14.0.1 r23809-234f1a2efa) 14.0.1, GNU ld (GNU Binutils) 2.42) #0 SMP Mon Apr  8 05:39:45 2024
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Netgear R8500
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Hit pending asynchronous external abort (FSR=0x00001c06) during first unmask, this is most likely caused by a firmware/bootloader bug.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  [mem 0x0000000008000000-0x000000009fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   node   0: [mem 0x0000000088000000-0x000000009fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000009fffffff]
[    0.000000] percpu: Embedded 17 pages/cpu s40660 r8192 d20780 u69632
[    0.000000] pcpu-alloc: s40660 r8192 d20780 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Kernel command line: console=ttyS0,115200
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130784
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 502832K/524288K available (8350K kernel code, 833K rwdata, 5244K rodata, 1024K init, 319K bss, 21456K reserved, 0K cma-reserved, 393216K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] L2C: DT/platform modifies aux control register: 0x0a130000 -> 0x3a530000
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 16 ways, 256 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x7e530001
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000002] sched_clock: 64 bits at 700MHz, resolution 1ns, wraps every 4398046511103ns
[    0.000018] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0xa17102bcf3, max_idle_ns: 440795224838 ns
[    0.000037] Switching to timer-based delay loop, resolution 1ns
[    0.000288] Calibrating delay loop (skipped), value calculated using timer frequency.. 1400.00 BogoMIPS (lpj=7000000)
[    0.000300] CPU: Testing write buffer coherency: ok
[    0.000323] CPU0: Spectre v2: using BPIALL workaround
[    0.000328] pid_max: default: 32768 minimum: 301
[    0.000523] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000534] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001327] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002003] RCU Tasks Trace: Setting shift to 1 and lim to 1 rcu_task_cb_adjust=1.
[    0.002116] Setting up static identity map for 0x100000 - 0x10003c
[    0.002253] rcu: Hierarchical SRCU implementation.
[    0.002258] rcu: 	Max phase no-delay instances is 1000.
[    0.002627] smp: Bringing up secondary CPUs ...
[    0.003214] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... 1022 lines cut ...]
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_ZLIB_OPTIMIZE_SPEED=y
CONFIG_ZSTD_OPTIMIZE_O3=y
# CONFIG_OPENSSL_WITH_ERROR_MESSAGES is not set
# CONFIG_PACKAGE_boost is not set
CONFIG_PACKAGE_kmod-netlink-diag=m
# CONFIG_PACKAGE_libcares is not set
# CONFIG_PACKAGE_libudns is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_ShadowsocksR_Libev_Client is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Shadowsocks_Libev_Server is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_SingBox is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_Trojan_Plus is not set
# CONFIG_PACKAGE_luci-app-passwall_INCLUDE_V2ray_Plugin is not set
# CONFIG_PACKAGE_shadowsocks-libev-ss-server is not set
# CONFIG_PACKAGE_shadowsocksr-libev-ssr-local is not set
# CONFIG_PACKAGE_shadowsocksr-libev-ssr-redir is not set
# CONFIG_PACKAGE_sing-box is not set
# CONFIG_PACKAGE_trojan-plus is not set
# CONFIG_PACKAGE_v2ray-plugin is not set
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

sbwml, 2024-04-09:
List all installed packages related to USB:

`opkg list-installed | grep usb`

```
brcmfmac-firmware-usb - 20240220-1
kmod-phy-bcm-ns-usb2 - 6.6.25-1
kmod-phy-bcm-ns-usb3 - 6.6.25-1
kmod-usb-audio - 6.6.25-1
kmod-usb-bcma - 6.6.25-1
kmod-usb-core - 6.6.25-1
kmod-usb-ehci - 6.6.25-1
kmod-usb-ledtrig-usbport - 6.6.25-1
kmod-usb-net - 6.6.25-1
kmod-usb-net-cdc-ether - 6.6.25-1
kmod-usb-net-ipheth - 6.6.25-1
kmod-usb-net-rndis - 6.6.25-1
kmod-usb-ohci - 6.6.25-1
kmod-usb-printer - 6.6.25-1
kmod-usb-xhci-hcd - 6.6.25-1
kmod-usb2 - 6.6.25-1
kmod-usb3 - 6.6.25-1
libusb-1.0-0 - 1.0.26-3
libusbmuxd - 2.0.2-1
luci-app-usb-printer - 1.0
luci-i18n-usb-printer-zh-cn - git-24.076.01343-f84ee78
usbmuxd - 1.1.1-1
usbutils - 014-1
```

nasbdh9, 2024-04-09:
> R8500 cannot be found in firmware selector?

https://github.com/openwrt/openwrt/commit/3b76c7cf0bdda679ca037b88621bc8f04ee9802e
https://github.com/openwrt/openwrt/commit/504000d520ac8699137bd6b8dbd55f723f34bfbe

brada4, 2024-04-09:
@rmilecki (or better write feedback at those PR-s)

sbwml, 2024-04-10:
> R8500 cannot be found in firmware selector?

it may be because `DEFAULT := n`  https://github.com/openwrt/openwrt/blob/main/target/linux/bcm53xx/image/Makefile#L482

sbwml, 2024-04-19:
r8500 usb has been fixed now.

[image]

closed 2024-04-19 by sbwml