openwrt/openwrt #15123: Wifi password badly parsed OpenWRT 23 but ok in OpenWRT19

biboc, 2024-04-10:
### Describe the bug

I add a mesh network with: 
```
config` wifi-iface 'mesh'
        option device 'radio0'
        option ifname 'mesh'
        option mode 'mesh'
        option network 'wan'
        option mesh_id '02egc@3h2f5g2'
        option mesh_fwding '1'
        option key "0Y89D#E$2C"P/A;A4D#2x6D""
        option channel '1'
        option mesh_rssi_threshold '-70'
        option encryption 'psk2'
 ```

I run `wifi` then `wifi status` 

```
wifi status
{
        "radio0": {
                "up": false,
                ...
                "interfaces": [
                        {
                                "section": "mesh",
                                "config": {
                                        "ifname": "mesh",
                                        "mode": "mesh",
                                        "mesh_id": "02egc@3h2f5g2",
                                        "mesh_fwding": true,
                                        "key": "0Y89D#E$2CP/A",
                                        "mesh_rssi_threshold": -85,
                                        "encryption": "psk2",
                                        "mode": "mesh",
                                        "network": [
                                                "wan"
                                        ]
                                },
                                "vlans": [

                                ],
                                "stations": [

                                ]
                        }
                ]
        }
}
```

Why password is truncated?
I can also see that wpa_supplicant gets same password as wifi status

### OpenWrt version

# . /etc/openwrt_release && echo $DISTRIB_REVISION r23630-842932a63d

### OpenWrt release

# . /etc/openwrt_release && echo $DISTRIB_RELEASE 23.05.2

### OpenWrt target/subtarget

# . /etc/openwrt_release && echo $DISTRIB_TARGET ramips/mt76x8

### Device

HILINK HLK-7628N

### Image kind

Self-built image

### Steps to reproduce

Build Openwrt 23 for HILINK HLK-7628N
Connect over ssh
Add a wifi with my password
Run wifi command
See that password is not correctly parsed

### Actual behaviour

Password is truncated

### Expected behaviour

Good password

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2024-04-10:
The phrase `"0Y89D#E$2C"P/A;A4D#2x6D""` is parsed as `0Y89D#E$2C` + `P/A` + `;` + `A4D#2x6D`. A semicolon terminates the line, so the semicolon itself and the `A4D#2x6D` part are skipped by the parser. The end result, due to implicit concatenation of `"0Y89D#E$2C"` and `P/A` is `0Y89D#E$2CP/A`.

Solution: put the pass phrase in single quotes:

```
config wifi-iface 'mesh'
        option device 'radio0'
        option ifname 'mesh'
        option mode 'mesh'
        option network 'wan'
        option mesh_id '02egc@3h2f5g2'
        option mesh_fwding '1'
        option key '0Y89D#E$2C"P/A;A4D#2x6D"'
        option channel '1'
        option mesh_rssi_threshold '-70'
        option encryption 'psk2'
```

biboc, 2024-04-10:
Thanks for the quick answer

If I do this, I get from wifi status is
```
"key": "0Y89D#E$2C\"P/A;A4D#2x6D\"",
```
Ok
But wpa_supplicant does not receive right password: 
```
failed to parse sae_password '"0Y89D#E$2C"P/A;A4D'
```
With 
```
cat /var/run/wpa_supplicant-mesh.conf
        sae_password="0Y89D#E$2C"P/A;A4D#2x6D""
```

This is maybe another problem with wpa_supplicant? What do you think?

jow-, 2024-04-10:
Yes, this looks like a genuine problem

biboc, 2024-04-11:
It definitly breaks my password
I'll send a message to hostapd community. It is not related to OpenWRT directly

jow-, 2024-04-11:
I don't think the issue is in hostapd, OpenWrt improperly escapes the value when writing it into the temporary configuration files.

biboc, 2024-04-17:
It is in hostapd, the patched added 3 years ago try to detect comment with " in the line but fail with my password because I have both # and " 
If you agree, close the issue

closed 2024-04-22 by biboc