openwrt/openwrt #15125: After booting OpenWrt, attempts at console input behave erratically.

dkachel7258, 2024-04-10:
### Describe the bug

Almost all attempts at console command line input fail with some error. The command listed is not 
what was typed.

### OpenWrt version

r25871-d668c74fe6

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Official downloaded image

### Steps to reproduce

1. Install a clean Snapshot release on an SD card and boot.
2. Press return to activate the console.
3. Attempt command input.

### Actual behaviour

After booting OpenWrt, attempts at console input behave erratically.

In the console output, the message "Please press Enter to activate this console." appears 
twice. Pressing return displays the message: 

	"/bin/ash: can't access tty; job control turned off"

followed by the OpenWrt login banner.

Pressing return again displays another login banner.  Subsequent returns display just the 
shell prompt.

Almost all attempts at command line input fail with some error. The command listed is not 
what was typed.

### Expected behaviour

Pressing enter displays a login banner, activates the console, and command input is processed as expected.

### Additional info

WORKAROUND
Comment out one of the lines in the file "/etc/inittab" that mention 
"/usr/libexec/login.sh", and reboot.

ADDITIONAL INFORMATION
The behavior started somewhere around the last week of March 2024, after applying an OpenWrt firmware upgrade to a Raspberry Pi 5B running a SNAPSHOT release.

By comparison, the behavior isn't present on release 23.05.3 (installed on a Raspberry 
Pi 4B).

I stumbled across commenting out an inittab entry to stop the behavior.  (I'd occasionally
been commenting both lines out to disable console logins altogether.)

I notice that inittab on 23.05.3 also has two references to login.sh so I suspect that 
simply removing one of them is not the permanent fix.

Given the behavior, it seems like two instances of the shell might be competing for the
console tty. The inittab workaround seem to support this idea.

A process list on Snapshot and 23.05.3 each show two processes running 
"/sbin/askfirst /usr/libexec/login.sh", so multiple multiple processes doesn't 
seem to be the root cause of the behavior.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Noltari, 2024-04-25:
@dkachel7258 is this behaviour present on a RPi 4B running the snapshots?

dkachel7258, 2024-04-25:
Just tested on SNAPSHOT (r26018-84a48ce400) (2024-04-25 05:18:02) and verified that behavior is present on RPI 4B (and RPI 5B, and RPI 3B+).

SirLouen, 2024-04-28:
Same here, It was happening to me and a user referred to me to this issue, so I can confirm is happening also
https://forum.openwrt.org/t/rasberry-pi-4-model-b-keyboards-gone-wild/195594

dkachel7258, 2024-04-28:
Additional Workaround:

For those not using the Raspberry Pi serial console, modify "/boot/cmdline.txt" to remove 
"console=serial0,115200 ".

This appears to disable one of the competing login processes by removing that console device.

This has the added advantage that it can be conveniently done right after factory-imaging the SD card and before the initial system boot.

SirLouen, 2024-04-28:
> Additional Workaround:
>
> For those not using the Raspberry Pi serial console, modify "/boot/cmdline.txt" to remove "console=serial0,115200 ".
>
> This appears to disable one of the competing login [...]

I'm currently using 23.05.3 and I was installing SNAPSHOT by mistake. Unless you found some feature in SNAPSHOT I suppose that for future releases this should be patched. I wonder which of the latest updates could be causing it.

ElbertMai, 2024-04-29:
> I wonder which of the latest updates could be causing it.

I've bisected this and it turns out commit [ca8c30208d5e](https://git.openwrt.org/?p=project/procd.git;a=commit;h=ca8c30208d5e1aaa2c0e3f732c4c9944735e9850) in procd is causing this issue. That same commit also completely breaks the UART console. Swapping the order of console options in cmdline.txt fixes it both issues in my testing. I've already submitted a [pull request](https://github.com/openwrt/openwrt/pull/15333) containing the swapped consoles for Raspberry Pi builds, so take a look and see if it works for you.

dkachel7258, 2024-04-29:
Tested on SNAPSHOT r26117 on an RPI 5B and RPI 4B, and swapping the two console specifications in cmdline.txt does indeed work for me. 😀

dkachel7258, 2024-05-02:
Tested on SNAPSHOT r26144-219018185e (Thu May 2 12:37:23 2024) on RPI 5B and RPI 4B; it looks like the fix made it into today's snapshot build.

closed 2024-05-02 by Noltari