openwrt/openwrt #15129: Slow wlan--->lan wlan---wan performace. No problem wlan----->router

asvio, 2024-04-11:
### Describe the bug

There is a performace 

### OpenWrt version

r25874-98f9154316

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Zyxel NBG7815

### Image kind

Self-built image

### Steps to reproduce

Test performace between wlan and  wan/lan

### Actual behaviour

This is an iperf3 test wlan-->lan ; wlan--->device.
```
asvio@MSI-7D25:~$ iperf3 -c 192.168.1.99 -P 8 ; iperf3 -c 192.168.1.1 -P 8
Connecting to host 192.168.1.99, port 5201
[  5] local 192.168.1.160 port 45334 connected to 192.168.1.99 port 5201
[  7] local 192.168.1.160 port 45350 connected to 192.168.1.99 port 5201
[  9] local 192.168.1.160 port 45354 connected to 192.168.1.99 port 5201
[ 11] local 192.168.1.160 port 45358 connected to 192.168.1.99 port 5201
[ 13] local 192.168.1.160 port 45370 connected to 192.168.1.99 port 5201
[ 15] local 192.168.1.160 port 45384 connected to 192.168.1.99 port 5201
[ 17] local 192.168.1.160 port 45394 connected to 192.168.1.99 port 5201
[ 19] local 192.168.1.160 port 45408 connected to 192.168.1.99 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  16.8 MBytes   141 Mbits/sec   51   60.8 KBytes       
[  7]   0.00-1.00   sec  7.27 MBytes  61.0 Mbits/sec   68   48.1 KBytes       
[  9]   0.00-1.00   sec  10.8 MBytes  90.9 Mbits/sec  107   45.2 KBytes       
[ 11]   0.00-1.00   sec  10.1 MBytes  84.4 Mbits/sec   58   65.0 KBytes       
[ 13]   0.00-1.00   sec  13.1 MBytes   110 Mbits/sec   44    100 KBytes       
[ 15]   0.00-1.00   sec  10.4 MBytes  87.1 Mbits/sec   43   65.0 KBytes       
[ 17]   0.00-1.00   sec  9.07 MBytes  76.1 Mbits/sec   77   45.2 KBytes       
[ 19]   0.00-1.00   sec  10.6 MBytes  88.6 Mbits/sec   61   93.3 KBytes       
[SUM]   0.00-1.00   sec  88.1 MBytes   739 Mbits/sec  509             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   1.00-2.00   sec  9.88 MBytes  82.9 Mbits/sec   42   89.1 KBytes       
[  7]   1.00-2.00   sec  7.02 MBytes  58.9 Mbits/sec   34   55.1 KBytes       
[  9]   1.00-2.00   sec  10.9 MBytes  91.2 Mbits/sec   19    103 KBytes       
[ 11]   1.00-2.00   sec  10.2 MBytes  85.5 Mbits/sec   40   91.9 KBytes       
[ 13]   1.00-2.00   sec  13.5 MBytes   113 Mbits/sec   32   83.4 KBytes       
[ 15]   1.00-2.00   sec  8.95 MBytes  75.1 Mbits/sec   44   56.6 KBytes       
[ 17]   1.00-2.00   sec  7.77 MBytes  65.2 Mbits/sec   42   62.2 KBytes       
[ 19]   1.00-2.00   sec  12.8 MBytes   107 Mbits/sec   29    102 KBytes       
[SUM]   1.00-2.00   sec  81.0 MBytes   679 Mbits/sec  282             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   2.00-3.00   sec  10.0 MBytes  83.9 Mbits/sec   51   69.3 KBytes       
[  7]   2.00-3.00   sec  9.07 MBytes  76.1 Mbits/sec   32   72.1 KBytes       
[  9]   2.00-3.00   sec  11.6 MBytes  97.5 Mbits/sec   68   74.9 KBytes       
[ 11]   2.00-3.00   sec  12.0 MBytes   101 Mbits/sec   42   63.6 KBytes       
[ 13]   2.00-3.00   sec  14.4 MBytes   120 Mbits/sec   25    120 KBytes       
[ 15]   2.00-3.00   sec  9.07 MBytes  76.1 Mbits/sec   40   73.5 KBytes       
[ 17]   2.00-3.00   sec  9.13 MBytes  76.6 Mbits/sec   16   73.5 KBytes       
[... 491 lines cut ...]
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wifischedule=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wireless-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mbedtls=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_WPA_MSG_MIN_PRIORITY=4
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

arispap2003, 2024-04-11:
Hello,
Usually, a wan connection uses Point-to-Point Protocol (PPP).
This connection uses MTU 1492 or 1480 (and not 1500).
So most of the time a connection needs to be gone under a packet fragmentation procedure. (eg. packet 1 doesn't fit the frame, will be divided into two packets, etc.)
Could be this is a part of the explanation?

Kind regards,

asvio, 2024-04-11:
If you look at my post, you will see a wlan--->lan test, so 1500 as mtu is the default value.

My wan conexión is pppoe, and It has 1492 for mtu (what It is made as default by openwrt), but even if a change mtu to 1500 i don't think i would get a 50 % performance reduction.

brada4, 2024-04-11:
Current mssfix does not work properly in all cases.
Please add file /etc/nftables.d/99-mssfix.nft and restart iptables.
```
chain mssfix {
    type filter hook postrouting priority mangle; policy accept;
    oif $wan_devices tcp flags syn / syn,fin,rst tcp option maxseg size set rt mtu
}
```
Also add -Z before -c of iperf3.
If you change pppoe MTU to 1500 all your packets will get 8 bytes chopped off the end and you will get PMTU blackhole for free.

asvio, 2024-04-11:
> Current mssfix does not work properly in all cases. Please add file /etc/nftables.d/99-mssfix.nft and restart iptables.
> 
> ```
> chain mssfix {
>     type filter hook postrouting priority mangle; policy accept;
>     oif $wan_devices tcp flags syn / syn,fin,rst tcp option maxseg size set rt mtu
> }
> ```
Still the same.
```
root@NBG7815:~# cat /etc/nftables.d/99-mssfix.nft
chain mssfix {
    type filter hook postrouting priority mangle; policy accept;
    oif $wan_devices tcp flags syn / syn,fin,rst tcp option maxseg size set rt mtu
}
root@NBG7815:~# 
```

```
asvio@MSI-7D25:~$ iperf3 -Z -c 192.168.1.99 -P 8 ; iperf3 -Z -c 192.168.1.1 -P 8
Connecting to host 192.168.1.99, port 5201
[  5] local 192.168.1.160 port 57250 connected to 192.168.1.99 port 5201
[  7] local 192.168.1.160 port 57264 connected to 192.168.1.99 port 5201
[  9] local 192.168.1.160 port 57266 connected to 192.168.1.99 port 5201
[ 11] local 192.168.1.160 port 57278 connected to 192.168.1.99 port 5201
[ 13] local 192.168.1.160 port 57280 connected to 192.168.1.99 port 5201
[ 15] local 192.168.1.160 port 57292 connected to 192.168.1.99 port 5201
[ 17] local 192.168.1.160 port 57298 connected to 192.168.1.99 port 5201
[ 19] local 192.168.1.160 port 57312 connected to 192.168.1.99 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  14.0 MBytes   117 Mbits/sec   61   76.4 KBytes       
[  7]   0.00-1.00   sec  9.60 MBytes  80.6 Mbits/sec   82   39.6 KBytes       
[  9]   0.00-1.00   sec  8.45 MBytes  70.9 Mbits/sec   26   66.5 KBytes       
[ 11]   0.00-1.00   sec  6.64 MBytes  55.7 Mbits/sec   32   67.9 KBytes       
[ 13]   0.00-1.00   sec  3.85 MBytes  32.3 Mbits/sec   25   39.6 KBytes       
[ 15]   0.00-1.00   sec  11.9 MBytes  99.7 Mbits/sec   24    148 KBytes       
[ 17]   0.00-1.00   sec  7.85 MBytes  65.8 Mbits/sec   78   29.7 KBytes       
[ 19]   0.00-1.00   sec  18.0 MBytes   151 Mbits/sec   49   80.6 KBytes       
[SUM]   0.00-1.00   sec  80.3 MBytes   673 Mbits/sec  377             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   1.00-2.00   sec  6.45 MBytes  54.1 Mbits/sec   31   45.2 KBytes       
[  7]   1.00-2.00   sec  6.51 MBytes  54.6 Mbits/sec   28   73.5 KBytes       
[  9]   1.00-2.00   sec  7.26 MBytes  60.9 Mbits/sec   25   60.8 KBytes       
[ 11]   1.00-2.00   sec  8.07 MBytes  67.7 Mbits/sec   20   73.5 KBytes       
[ 13]   1.00-2.00   sec  3.48 MBytes  29.2 Mbits/sec   28   22.6 KBytes       
[ 15]   1.00-2.00   sec  7.75 MBytes  65.0 Mbits/sec   47   39.6 KBytes       
[ 17]   1.00-2.00   sec  5.65 MBytes  47.4 Mbits/sec   11   59.4 KBytes       
[ 19]   1.00-2.00   sec  5.75 MBytes  48.2 Mbits/sec   47   41.0 KBytes       
[SUM]   1.00-2.00   sec  50.9 MBytes   427 Mbits/sec  237             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   2.00-3.00   sec  4.84 MBytes  40.6 Mbits/sec   51   33.9 KBytes       
[  7]   2.00-3.00   sec  4.90 MBytes  41.1 Mbits/sec   48   41.0 KBytes       
[  9]   2.00-3.00   sec  6.70 MBytes  56.2 Mbits/sec   33   69.3 KBytes       
[ 11]   2.00-3.00   sec  5.96 MBytes  50.0 Mbits/sec   27   43.8 KBytes       
[ 13]   2.00-3.00   sec  2.49 MBytes  20.9 Mbits/sec   40   21.2 KBytes       
[ 15]   2.00-3.00   sec  3.41 MBytes  28.6 Mbits/sec   27   39.6 KBytes       
[ 17]   2.00-3.00   sec  5.15 MBytes  43.2 Mbits/sec   29   48.1 KBytes       
[ 19]   2.00-3.00   sec  3.67 MBytes  30.8 Mbits/sec   49   41.0 KBytes       
[SUM]   2.00-3.00   sec  37.1 MBytes   311 Mbits/sec  304             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   3.00-4.00   sec  4.03 MBytes  33.8 Mbits/sec   57   45.2 KBytes       
[  7]   3.00-4.00   sec  4.84 MBytes  40.6 Mbits/sec   64   43.8 KBytes       
[  9]   3.00-4.00   sec  6.70 MBytes  56.2 Mbits/sec   23   52.3 KBytes       
[ 11]   3.00-4.00   sec  7.14 MBytes  59.9 Mbits/sec   21   60.8 KBytes       
[ 13]   3.00-4.00   sec  3.46 MBytes  29.0 Mbits/sec   58   24.0 KBytes       
[ 15]   3.00-4.00   sec  5.34 MBytes  44.8 Mbits/sec   44   36.8 KBytes       
[ 17]   3.00-4.00   sec  5.34 MBytes  44.8 Mbits/sec   42   48.1 KBytes       
[ 19]   3.00-4.00   sec  4.20 MBytes  35.2 Mbits/sec   24   26.9 KBytes       
[SUM]   3.00-4.00   sec  41.0 MBytes   344 Mbits/sec  333             
- - - - - - - - - - - - - - - - - - - - - - - - -
[... 452 lines cut ...]
[SUM]   9.00-10.00  sec   110 MBytes   923 Mbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   118 MBytes  98.9 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   115 MBytes  96.1 Mbits/sec                  receiver
[  7]   0.00-10.00  sec   112 MBytes  93.8 Mbits/sec    0             sender
[  7]   0.00-10.01  sec   108 MBytes  90.6 Mbits/sec                  receiver
[  9]   0.00-10.00  sec   135 MBytes   113 Mbits/sec    0             sender
[  9]   0.00-10.01  sec   132 MBytes   110 Mbits/sec                  receiver
[ 11]   0.00-10.00  sec   128 MBytes   108 Mbits/sec    0             sender
[ 11]   0.00-10.01  sec   125 MBytes   105 Mbits/sec                  receiver
[ 13]   0.00-10.00  sec   133 MBytes   112 Mbits/sec    0             sender
[ 13]   0.00-10.01  sec   129 MBytes   108 Mbits/sec                  receiver
[ 15]   0.00-10.00  sec   162 MBytes   136 Mbits/sec    0             sender
[ 15]   0.00-10.01  sec   159 MBytes   133 Mbits/sec                  receiver
[ 17]   0.00-10.00  sec   122 MBytes   102 Mbits/sec    0             sender
[ 17]   0.00-10.01  sec   119 MBytes  99.5 Mbits/sec                  receiver
[ 19]   0.00-10.00  sec   126 MBytes   105 Mbits/sec    0             sender
[ 19]   0.00-10.01  sec   123 MBytes   103 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  1.01 GBytes   869 Mbits/sec    0             sender
[SUM]   0.00-10.01  sec  1009 MBytes   846 Mbits/sec                  receiver

iperf Done.
asvio@MSI-7D25:~$ 
```

brada4, 2024-04-11:
Restart firewall obviously. And optionally disable mssfix from GUI.
Can you edit your test outputs to leave only 2 last [SUM]mary lines from each?

brada4, 2024-04-11:
Also what is the WLAN mode negotiated to `iwinfo phy0-ap0 assoclist`
In common case you will get 2x2 aka NSS2 1.2Gbps AX or 866Mbps AC (or seems you got 1x1 AC actually)

asvio, 2024-04-11:
> Also what is the WLAN mode negotiated to `iwinfo phy0-ap0 assoclist` In common case you will get 2x2 aka NSS2 1.2Gbps AX or 866Mbps AC (or seems you got 1x1 AC actually)
```
root@NBG7815:~# iwinfo phy0-ap0 assoclist
C8:94:02:4A:48:A5  -52 dBm / -109 dBm (SNR 57)  0 ms ago
        RX: 1200.9 MBit/s, HE-MCS 11, 80MHz, HE-NSS 2, HE-GI 0, HE-DCM 0   1678322 Pkts.
        TX: 1080.6 MBit/s, HE-MCS 10, 80MHz, HE-NSS 2, HE-GI 0, HE-DCM 0    156936 Pkts.
        expected throughput: unknown
```

> And optionally disable mssfix from GUI.

I don't know how to do it through GUI.
What I know to do is delete it and reinsert it and restart the firewall later

asvio, 2024-04-11:
Same result. 
```
root@NBG7815:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'f.............::/48'
        option packet_steering '1'

```

brada4, 2024-04-11:
More ideas:
- `tc -s qdisc` -> any drop-s are bad
- `ethtool -S br-lan` -> any errors pauses drops on any member interface bad. (please dont post shole output ;-) )

asvio, 2024-04-20:
The problem seems to have been resolved with the latest updates.
```
asvio@MSI-7D25:~$ iperf3 -Z -c 192.168.1.99 -P8
Connecting to host 192.168.1.99, port 5201
--------------------------------------------
[SUM]   0.00-10.00  sec   828 MBytes   695 Mbits/sec   28             sender
```

closed 2024-04-20 by asvio