openwrt/openwrt #15136: fstools: CONFIG_FSTOOLS_OVL_MOUNT_COMPRESS_ZLIB adds zlib to any filesystem indiscriminately

M95D, 2024-04-11:
### Describe the bug

libfstools/overlay.c
 362 #ifdef OVL_MOUNT_COMPRESS_ZLIB
 363                 "compr=zlib"
 364 #else
 365                 NULL
 366 #endif 

This adds zlib to any filesystem, including those that don't support zlib, such as f2fs used in Turris Omnia. This causes overlay mount to fail.

### OpenWrt version

r25618

### OpenWrt release

snapshot

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Turris Omnia

### Image kind

Self-built image

### Steps to reproduce

- enable CONFIG_FSTOOLS_OVL_MOUNT_COMPRESS_ZLIB
- build the squashfs image
- boot it

### Actual behaviour

f2fs overlay mount fails because of the unsupported compr=zlib option

### Expected behaviour

compr=zlib should not be applied to f2fs, optionally use compress_algorithm=zstd instead
- or -
the build option should be unavailable on devices that use f2fs

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-12:
Vlan id 1 is used internally by dsa, use anything else.

M95D, 2024-04-12:
Wrong broser tab? :roll_eyes: