openwrt/openwrt #15144: ucode: possible regression with 'in' operator

efahl, 2024-04-12:
### Describe the bug

@jow- 

Started at https://forum.openwrt.org/t/will-a-single-nic-work-for-snort3-in-ips-mode-if-pure-wireless/194218/30

Checking list containment with `in` operator has changed behavior between 
Version: 2024.02.21~ba3855ae-r1
and
Version: 2024.04.07~5507654a-r1

With 02.21 version, string was auto-coerced to int and result was found:
```
$ ucode -p " '1' in [0, 1] "
true
```
With 04.07 version, it appears strings is not coerced, resulting in different results.
```
$ ucode -p " '1' in [0, 1] "
false
```

Is this new behavior correct?  (I.e., should I be doing explicit casts in my code?)  Or is this a regression that should be fixed in ucode itself?

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

x86

### Device

any

### Image kind

Official downloaded image

### Steps to reproduce

ucode -p " '1' in [0, 1] "

### Actual behaviour

false

### Expected behaviour

true

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2024-04-12:
This is intentional, see the commit message for https://github.com/jow-/ucode/commit/be767ae197babd656d4f5d9c2d5013e39ddbe656

efahl, 2024-04-12:
Got it!  Thanks, closing.

closed 2024-04-12 by efahl