openwrt/openwrt #15146: Small reflection issue in redirects

jorgechamorro, 2024-04-12:
### Describe the bug

Hi!

```
System
Hostname	OpenWrt
Model	Hewlett-Packard HP t610 PLUS WW Thin Client
Architecture	AMD G-T56N Processor
Target Platform	x86/64
Firmware Version	OpenWrt 23.05.3 r23809-234f1a2efa / LuCI openwrt-23.05 branch git-24.086.45142-09d5a38
Kernel Version	5.15.150
```

This is the config:

```
config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'http://'
	list proto 'tcp'
	option src 'wan'
	option src_dport '80'
	option dest_ip '192.168.10.5'
	option dest_port '8079'
	option family 'ipv4'
	option reflection_src 'external'
```

It's working fine as long as dest_ip is another machine, but when I put in dest_ip its own IP (the openwrt router's), instead of the external (pppoe) IP I get its own (the router's) LAN IP in source address.

I was trying to move my (tiny) http server into the router but won't be able to do so unless this works.

Looks like a bug to me, but maybe not, I'm not an expert. Am I doing something wrong? And in any case, is there a workaround?

Thanks!

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

x86/64

### Device

Hewlett-Packard HP t610 PLUS WW Thin Client

### Image kind

Official downloaded image

### Steps to reproduce

Choose the router's lan ip as destination for the redirect.

### Actual behaviour

The source address isn't the public address but the lan address.

### Expected behaviour

It should be the public address.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2024-04-13:
If the httpd server is on the router, the dest would not need to be lan. It could be omitted.

```
config redirect
        option target 'DNAT'
        option name 'httpd'
        option family 'ipv4'
        list proto 'tcp'
        option src 'wan'
        option src_dport '8079'
        option dest_port '80'
```

jorgechamorro, 2024-04-13:
> If the httpd server is on the router, the dest would not need to be lan. It could be omitted.

Hi Dave, and thanks!

:8079 is a filter (before the server) that I'd want to be on the LAN (to be able to move it with ease). When the filter is in another (LAN) machine this works, but as soon as I move it to the router stops working (for the LAN clients) because the source address it gets is wrong.

I'd like to know why the reflection stops working depending on the destination, and if that's a bug or my bad (something I'm misunderstanding, very likely!).

Cheers,

jorgechamorro, 2024-04-13:
Tested:

```
Hostname		openWrt
Model			LeMaker Banana Pro
Architecture		ARMv7 Processor rev 4 (v7l)
Target Platform		sunxi/cortexa7
Firmware Version	OpenWrt 23.05.3 r23809-234f1a2efa / LuCI openwrt-23.05 branch git-24.086.45142-09d5a38
Kernel Version		5.15.150
````

(same redirect config)

And it does the same: only works if :8079 is in another machine.

jorgechamorro, 2024-04-13:
I've also tested using other ports, just in case :80 had somehow a different behavior due to LuCI being there, but it makes no difference.

jorgechamorro, 2024-04-13:
Could this have something to do with it?

https://git.openwrt.org/?p=project/firewall4.git;a=commitdiff;h=1a94915b5d38c9f17b27481add5a0a62341da627;hp=5c21714dc503d9cee6a638cb8ab06fb40ae473cc