openwrt/openwrt #15159: static dhcp assignments for secondary subnets on an interface do not take effect

wez, 2024-04-13:
### Describe the bug

When two or more subnets are added to the same interface, the generated dnsmasq configuration does not include `dhcp-range` for the additional subnets.

When using static dhcp host assignments for those additional subnets, dnsmasq will ignore them because they have no corresponding `dhcp-range`.  Instead, dynamic assignments from the first subnet will be used.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

I have an interface with two subnets:

```
config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ip6assign '60'
        list ipaddr '192.168.1.3/24'
        list ipaddr '10.42.0.2/24'	
```

The `10.42.0.2/24` subnet is for some local infra that I want to de-clutter from the `192.168.1.3` subnet.
I do not want or need to use VLANs for them.

In addition to the regular dynamic assignments for the `192.168.1.3/24` subnet, I want to perform static assignments for the infra machines on the `10.42.0.2` subnet:

```
config host
        option name 'some-name'
        option mac 'XXXXXX'
        option ip '10.42.0.50'
        option dns '1'
```

### Actual behaviour

The configured `10.42.0.50` assignment does not take effect; the device is assigned an address from the `192.168.1.3/24` subnet.

### Expected behaviour

The configured `10.42.0.50` assignment should take effect.

### Additional info

I realize that resolving this is a bit tricky overall, because it implies that perhaps there should be additional config plumbing and UI to control the IP ranges for those additional subnets, and other associated options.

I think that perhaps this was less of an issue in earlier releases of openwrt because it was possible to create aliased interfaces, but those appear to not be a supported concept any more.

The workaround I am using at the moment is to add the required configuration to `/etc/dnsmasq.conf`:

```dnsmasq
dhcp-range=set:infra,10.42.0.200,static,255.255.255.0,2h
dhcp-option=infra,6,10.42.0.2
```

with that in place, my static assignments do take effect.

Using `/etc/dnsmasq.conf` is undesirable because it is harder to backup/restore.

I am aware of and following https://github.com/openwrt/openwrt/pull/14975 which would seem to resolve that concern, but it doesn't address this gap around handling multiple subnets on an interface.

Thanks for listening!

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2024-04-14:
Simply declare another logical interface (e.g. lan2) and make it use the same device as lan (e.g. br-lan) or the special notation `@lan`. This is still fully supported and should integrate with the dnsmasq configuration.

wez, 2024-04-14:
Ah, yes, that does work, thanks! It looks a little weird in the UI, because the IPv6 addresses, in particular, those that are delegated from my wan, are attributed to this new `infra` interface rather than the LAN, even though I have IPv6 disabled on the `infra` interface.  I understand that this doesn't affect anything, because under the covers these are all just addresses added to `br-lan`. It just looks surprising.

Does it show up on this interface because `infra` sorts before `lan`?

[image]

rtsmith134, 2025-06-07:
Here is my DHCP file including my attempt to get a wired client lg-gram to connect to the guest network
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/home/'
	option domain 'home'
	option expandhosts '1'
	option cachesize '1000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option localservice '1'
	option ednspacket_max '1232'

config dhcp 'lan'
	option interface 'lan'
	option start '150'
	option limit '100'
	option leasetime '12h'
        option dhcpv4 'server'

config dhcp 'guest'
	option interface 'guest'
	option start '110'
	option limit '40'
	option leasetime '12h'
	option dhcpv4 'server'
	list dhcp_option '6,9.9.9.9,149.112.112.112'

config dhcp 'vpn'
	option interface 'vpn'
	option start '50'
	option limit '60'
	option leasetime '12h'
        option dhcpv4 'server'
	list dhcp_option '6,9.9.9.9,149.112.112.112'

config host
	option name 'lg-gram'
	option ip '192.168.41.22'
	option mac '00:E0:4C:68:1B:0A'
	option leasetime '3h'
	list tag 'lg-gram'

rtsmith134, 2025-06-07:
here is my network file with lan, guest, and vpn defined

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth1.1'
	option ipv6 '0'

config device
	option name 'eth1.1'
	option macaddr 'c0:56:27:4c:b4:22'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.4.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config device
	option name 'eth0.2'
	option macaddr 'c0:56:27:4c:b4:22'

config interface 'wan'
	option device 'eth0.2'
	option proto 'dhcp'

config device
	option type 'bridge'
	option name 'br-guest'
	list ports 'eth1.3'
	option ipv6 '0'

config device
	option type 'bridge'
	option name 'br-vpn'
	list ports 'eth1.4'
	option ipv6 '0'

config interface 'guest'
	option proto 'static'
	option device 'br-guest'
	option ipaddr '192.168.41.1'
	option netmask '255.255.255.0'
	option delegate '0'
	list dns '8.8.8.8'

config interface 'vpn'
	option proto 'static'
	option device 'br-vpn'
	option ipaddr '192.168.61.1'
	option netmask '255.255.255.0'
	option delegate '0'

config interface 'wan_vpn'
	option proto 'none'
	option delegate '0'
	option device 'tun0'
	option ifname 'tun0'