openwrt/openwrt #15160: tp-link archer c6 ac1200 v4.0

sergio4r0, 2024-04-13:
### Describe the bug

This version will have tp-link archer c6 ac1200 v4.0 openwrt

[image]
[image]

### OpenWrt version

*

### OpenWrt release

*

### OpenWrt target/subtarget

*

### Device

*

### Image kind

Official downloaded image

### Steps to reproduce

*

### Expected behaviour

*

### Additional info

*

### Diffconfig

```text
*
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-14:
no need to cross-post
https://forum.openwrt.org/t/tp-link-archer-c6-v4-openwrt/194439

pkgadd, 2024-04-15:
While everything has been said in the corresponding forum thread already, this is not a bug, but a feature request for new device support. But that doesn't work this way, no one will go out and buy this device in order to work on it - so either you do the development yourself and provide a pull request - or no one will.

…but given the 4 MB flash situation, device support for this device is never going to happen, so this IR should just be closed.

TRSPSR, 2024-12-31:
Am gonna give this a try. And I will let the guys know

TRSPSR, 2024-12-31:
> While everything has been said in the corresponding forum thread already, this is not a bug, but a feature request for new device support. But that doesn't work this way, no one will go out and buy [...]

Also, It's not 4 mb. It's 8 mb