openwrt/openwrt #15161: The Intel i226-v network card cannot enable TX flow control.

yunxi993, 2024-04-14:
### Describe the bug

After the device is started, the network card's TX traffic control cannot be automatically enabled.

`[   82.690877] igc 0000:01:00.0 eth0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX`

`[   75.100959] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX`

### OpenWrt version

Latest main branch

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

x86/64

### Image kind

Self-built image

### Steps to reproduce

Network interface in use: eth0: wan, eth1: lan

```
root@N100:~# ethtool -a eth0
Pause parameters for eth0:
Autonegotiate:  on
RX:             on
TX:             off
```
```
root@N100:~# ethtool -a eth1
Pause parameters for eth0:
Autonegotiate:  on
RX:             on
TX:             off
```
***
Unused network interface: eth2, eth3
```
root@N100:~# ethtool -a eth2
Pause parameters for eth2:
Autonegotiate:  on
RX:             on
TX:             on
```
```
root@N100:~# ethtool -a eth3
Pause parameters for eth2:
Autonegotiate:  on
RX:             on
TX:             on
```

### Expected behaviour

After the OpenWrt device is started, RX/TX traffic control can be enabled normally.

### Additional info

In SSH: `ethtool -A tx on` shows that TX flow control is enabled, but it will be turned off again after 5 seconds. I have to disable auto-negotiation in order to keep TX flow control enabled persistently.
[... 306 lines cut ...]
# CONFIG_PACKAGE_libltdl is not set
# CONFIG_PACKAGE_liblua5.3 is not set
# CONFIG_PACKAGE_libnetfilter-conntrack is not set
# CONFIG_PACKAGE_libnfnetlink is not set
# CONFIG_PACKAGE_libpcre2 is not set
# CONFIG_PACKAGE_libsodium is not set
# CONFIG_PACKAGE_libudns is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_Haproxy is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_ShadowsocksR_Libev_Client is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_Shadowsocks_Libev_Client is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_Simple_Obfs is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_V2ray_Plugin is not set
# CONFIG_PACKAGE_shadowsocks-libev-config is not set
# CONFIG_PACKAGE_shadowsocks-libev-ss-local is not set
# CONFIG_PACKAGE_shadowsocks-libev-ss-redir is not set
# CONFIG_PACKAGE_shadowsocksr-libev-ssr-local is not set
# CONFIG_PACKAGE_shadowsocksr-libev-ssr-redir is not set
# CONFIG_PACKAGE_simple-obfs is not set
# CONFIG_PACKAGE_v2ray-plugin is not set
# CONFIG_SING_BOX_WITH_WIREGUARD is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-14:
Does link partner support it? Is there any benefit of not having it disabled at all ends?

yunxi993, 2024-04-14:
@brada4 

Sorry, I didn't understand what you said.
Do you need me to provide you with some kind of information?

brada4, 2024-04-14:
Yes, what other link partner says about pause frames. 
Also `ethtool eth0` will show negotiation info.

yunxi993, 2024-04-14:
@brada4 

```
root@N100:~# ethtool eth0
Settings for eth0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        MDI-X: off (auto)
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```
```
root@N100:~# ethtool eth1
Settings for eth1:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        MDI-X: off (auto)
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

brada4, 2024-04-14:
Your opposite does not announce pause mode, which means either you allow it there or disable pause frames totally.
```
ethtool wan
Settings for wan:
        Supported ports: [ TP    MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
****        Link partner advertised pause frame use: No
****        Link partner advertised auto-negotiation: Yes
****        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        master-slave cfg: preferred master
        master-slave status: master
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: external
        MDI-X: Unknown
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
```

yunxi993, 2024-04-14:
@brada4 

Thank you!
eth0 serves as the WAN interface, eth1 serves as the LAN interface and is connected to a switch.
When I enable the traffic control function on the switch, OpenWrt will automatically detect and enable RX/TX traffic control. Can it be understood in this way?

eth0 is connected to the device issued by ISP and cannot enable the traffic control function. What should I do?

brada4, 2024-04-14:
You should disable pause frames. Nothing to do with tc traffic shaping/qdisc.

yunxi993, 2024-04-14:
@brada4 

Sorry, I'm not very familiar with this aspect. Could you teach me how to disable pause frames?
Thank you!

brada4, 2024-04-14:
ethtool -A eth0 tx off rx off auto on

yunxi993, 2024-04-14:
@brada4 

I used this command not long ago and found that the information displayed by `ethtool eth0` was no different from before. 
This led me to mistakenly believe that disabling pause frames was another command.

When I disable rx/tx flow control, my local network speed cannot reach the optimal state.

Sometimes more than 900Mb, sometimes only 600Mb.

Currently, it seems that the only way is to disable autoneg for `eth0 and eth1`, which can keep TX flow control enabled.

closed 2024-04-16 by yunxi993

webysther, 2024-10-04:
> @brada4
>
> I used this command not long ago and found that the information displayed by `ethtool eth0` was no different from before. This led me to mistakenly believe that disabling pause frames was [...]

For me what fixed add this line on startup > local startup:

```sh
sleep 10 && ethtool -A eth1 autoneg off rx off tx off
```

In both cases with proxmox or baremetal the same problem, to get better performance enable network > Global network options > Packet Steering

Monitor:

```sh
opkg update; opkg install procps-ng-watch
watch -n 1 'sh -c "logread | grep eth | tail"'
```

yunxi993, 2024-10-10:
> In both cases with proxmox or baremetal the same problem, to get better performance enable network > Global network options > Packet Steering

I installed OpenWrt directly on a physical machine. My device is a 4-core intel N100 CPU.

When I enable Packet Steering, the value of my `eth0` (pppoe) interface `rps_cpus` file is `1`, I don't know why its value is not `4`.

`packet_steering='1', packet_steering='2'(all CPUs)` I'm not sure which one should be used.

Whether `steering_flows` needs to be enabled, and if so, whether its value needs to be set to `4096`.

webysther, 2024-10-11:
> I installed OpenWrt directly on a physical machine. My device is a 4-core intel N100 CPU.
>
> When I enable Packet Steering, the value of my `eth0` (pppoe) interface `rps_cpus` file is `1`, I don't [...]

In my case turns out was my fault, I don't get the blocking state is because a loop being closed by the kernel, check if your dns is point back to your router, this was my mistake in this case.