openwrt/openwrt #15165: E8450 23.05.3 USB power problem

HKGIOS, 2024-04-14:
### Describe the bug

Before update to 23.05.3, I was using the E8450 UBI with 23.05.2, And I am using an ikea home hub for smart lighting plugged in the usb port of the E8450 for power without any problem.

But after update to 23.05.3 I cannot use the usb port to power on the hub, now I have to use another power adapter to operate the smart hub.

Please check or fix this issue, thank you.

### OpenWrt version

23.05.3

### OpenWrt release

r23809-234f1a2efa

### OpenWrt target/subtarget

mediatek/mt7622

### Device

E8450 UBI

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2024-04-15:
ping @dangowrt

HKGIOS, 2024-04-15:
> Invalid Version reported. `23.05.3` Is this from a clean repository?

yes, downloaded from
https://firmware-selector.openwrt.org/

dangowrt, 2024-04-15:
I've just tried connecting a USB hub with 2 downstream devices connected and without external power source to my E8450. Works fine with both 23.05.2 and 23.05.3, I cannot reproduce this problem with my hardware (which doesn't mean that the problem doesn't exist, obviously):
```
BusyBox v1.36.1 (2024-03-22 22:09:42 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.3, r23809-234f1a2efa
 -----------------------------------------------------
root@OpenWrt:/# [   37.081330] usb 1-1: new high-speed USB device number 2 using xhci-mtk
[   43.968422] usb 1-1: USB disconnect, device number 2
[   43.974594] xhci-mtk 1a0c0000.usb: drop_ep_quirk high-speed ep7in intr, mpkt:16, interval:8/16ms
[   52.211323] usb 1-1: new high-speed USB device number 3 using xhci-mtk
[   52.395063] hub 1-1:1.0: USB hub found
[   52.399144] hub 1-1:1.0: 4 ports detected
[   52.741329] usb 1-1.4: new full-speed USB device number 4 using xhci-mtk
[   58.241326] usb 1-1.3: new high-speed USB device number 5 using xhci-mtk
```

Please let me know what kind of hub you are using and share the full bootlog of both versions, 23.05.2 and 23.05.3 with the hub and devices connected. Note that the power budget of the USB 2.0 port may not be sufficient for more than one USB 2.0 device (eg. a pendrive) connected, and one of the changes which came with the 23.05.3 release was to increase the driving current for the SPI-NAND chip -- that's a very small change in terms of available power budget, but it could make a difference.

HKGIOS, 2024-04-15:
It was the ikea tradfri gateway, after 23.05.3 it seems like not enough power to boot it up.
https://www.ikea.com/gb/en/customer-service/product-support/app-gateway/

I thought the snapshot will help, so I flashed the new installer v1.1.1 and the openwrt snapshot, but it wouldn’t. I’m so sorry I can’t provide the bootlog  of both the old version now.

dangowrt, 2024-04-15:
I've been trying to find written specification or a manual of that thing but have failed to far. IKEA seems to only provide documentation in the form of youtube videos and that exceeds my patience. However, [someone on reddit claims that it needs as little as 150mA](https://www.reddit.com/r/tradfri/comments/jjll54/comment/gadbzgj/) which is completely ok for a USB 2.0 port and should be no problem. I have just now tried drawing about 0.8A from that USB 2.0 port with 23.05.2, 23.05.3 and recent snapshot, all works just fine (I'm not recommending to draw more then 0.5A there, but at least for a short time there wasn't any problem).

Can it be that there is a different cause? Such as a broken cable? Or a broken power supply? (I'm not using the stock power supply on my devices because they came with UK and US type adapters and I where I live we got EU-style German sockets).

Also, please try if the device on USB comes up if you temporarily disable both WiFi interfaces on the E8450, as that would also hint to a problem with the power supply.

HKGIOS, 2024-04-23:
sorry for late reply, Still no hope.
But I did tested the port, I cannot charge any phone with that port too. 

So maybe only my port having problem?
It's so weird, on 23.05.2 there is no any issue. Although now I can't go back to old version

Lucky1openwrt, 2024-04-30:
any chance your plug pack has changed or gone faulty

HKGIOS, 2024-05-03:
> any chance your plug pack has changed or gone faulty

the only thing I’ve done is flashing new firmware,
so I don’t know what happened, but the port seems can’t provide any power now

Lucky1openwrt, 2024-05-03:
ok well as most openwrt people have more then one router
and 9 out of 10 routers have a 12V plug pack & most of them have a 5.1mm / 2.1mm DC Jack
most have a plug pack around that they can quickly test with
mine has a 2A one tho this seems low for external HDD's ext anyway
but 2A or more should do

alienatedsec, 2024-05-16:
I had initial problems with my UPS connecting to USB on `23.05.3` - while my `WRT32x` allows me to boot an alternative partition with older firmware, I reverted to `23.05.2` and the issue was gone. 

After booting to `23.05.3` again this morning, the USB issue is no longer there and everything works as expected.

[image]

```
Thu May 16 10:02:52 2024 daemon.notice upsd[2481]: Data for UPS [UPSLiving] is stale - check driver
Thu May 16 10:02:52 2024 daemon.notice upsd[2481]: UPS [UPSLiving] data is no longer stale
```