openwrt/openwrt #15169: Problem disassociated Wifi AX9000 / AX3600 802.11R use

rdlvm, 2024-04-16:
### Describe the bug

Problem disassociated Wifi AX9000 / AX3600 802.11R use 

Wi-Fi disconnections, and constantly this warning/error in the openwrt log:

Tue Apr 16 08:55:44 2024 daemon.notice hostapd: phy2-ap0: AP-STA-DISCONNECTED
Tue Apr 16 08:55:44 2024 daemon.info hostapd: phy2-ap0: STA IEEE 802.11: disassociated
Tue Apr 16 08:55:45 2024 daemon.info hostapd: phy2-ap0: STA  IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

### OpenWrt version

r25940-8c3dfdc46b

### OpenWrt release

r25940-8c3dfdc46b

### OpenWrt target/subtarget

r25940-8c3dfdc46b

### Device

AX9000 // AX 36000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rdlvm, 2024-04-16:
@Ansuel @robimarko

brada4, 2024-04-16:
It logged during normal operation. Deauth due to code X would be sign of active deauthentication attack.

rdlvm, 2024-04-17:
> It logged during normal operation. Deauth due to code X would be sign of active deauthentication attack.

I don't understand what you mean exactly, but I constantly have WiFi outages.

brada4, 2024-04-17:
The message is logged when device walks out of range. Please fix access point information in your post.  No 36000 supported in openwrt

rdlvm, 2024-04-17:
> The message is logged when device walks out of range. Please fix access point information in your post. No 36000 supported in openwrt

It is impossible for it to go out of range because it is only 50 centimeters from the router, another maximum 5 meters from the router without obstacles... The problem is registered with constant microcuts when having 802.11R activated

closed 2024-04-17 by rdlvm