openwrt/openwrt #15174: MT7975N - High retries during an iperf3 test between router and client via 2.4 GHz

nightcore500, 2024-04-16:
### Describe the bug

I'm a bit unsure whether I should post this problem here or rather in the mt76 repo. Let me know if it is out of place here.

I have several devices here (HTC U12 Plus, OnePlus 10 Pro, Samsung Galaxy A32) that have very low download speeds (just under 20 MBit) if they are connected to the router via the 2.4 GHz module (MT7975N). It is irrelevant whether both devices are in an inteference-free environment or not. However, if the client are connected to the 5 GHz module (MT7975P), they show a stable download speed (stable well above 80 MBit).

An iperf3 test (attached below) between router and client shows that the MT7975N module apparently has problems in the TX direction. Many retries are noticeable. These problems do not appear to occur in the RX direction. The 5 GHz module (MT7975P), on the other hand, shows a stable bandwidth over minutes with very few retries.

I first noticed this problem with OpenWRT release 23.05.2 (I have not tried older versions) and it also occurs in 23.05.3. I currently have a self-built image running on the router. The OpenWRT source is from last night (main branch). I have changed the mt76 package to the current commit "8f301a5c5fe3d998b50666bde1d35d412833be89". The problem can also be observed here.

I then ran an 802.11 dump. I noticed that during the iperf3 test the client occasionally sends QoS Null function and QoS Data frames with the power management flag set.

The whole thing becomes even stranger as soon as the client are sent to standby while the iperf3 test is running. The bandwidth stabilizes abruptly to around 100 Mbit and the retries stop immediately. There are also no more frames to be seen with the power management flag set.

So is this possibly a problem with the power saving?
On the other hand, the devices work without any problems with other chipsets.

Iperf3 test on latest source - TX path:
`
Accepted connection from 192.168.1.215, port 44074
[  5] local 192.168.1.1 port 5201 connected to 192.168.1.215 port 44076
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  2.62 MBytes  22.0 Mbits/sec   48   41.0 KBytes
[  5]   1.00-2.00   sec  3.00 MBytes  25.2 Mbits/sec    6   43.8 KBytes
[  5]   2.00-3.00   sec  3.38 MBytes  28.3 Mbits/sec   18   53.7 KBytes
[  5]   3.00-4.00   sec  9.62 MBytes  80.7 Mbits/sec   74   56.6 KBytes
[  5]   4.00-5.00   sec  10.0 MBytes  83.9 Mbits/sec   58   90.5 KBytes
[  5]   5.00-6.00   sec  10.0 MBytes  83.9 Mbits/sec   99   73.5 KBytes
[  5]   6.00-7.00   sec  9.50 MBytes  79.7 Mbits/sec   47   67.9 KBytes
[  5]   7.00-8.00   sec  10.4 MBytes  87.0 Mbits/sec   54    105 KBytes
[  5]   8.00-9.00   sec  9.50 MBytes  79.7 Mbits/sec   98   63.6 KBytes
[  5]   9.00-10.00  sec  10.6 MBytes  89.1 Mbits/sec   54   89.1 KBytes
[  5]  10.00-11.00  sec  10.4 MBytes  87.0 Mbits/sec   71    100 KBytes
[  5]  11.00-12.00  sec  9.50 MBytes  79.7 Mbits/sec  125   48.1 KBytes
[  5]  12.00-13.00  sec  10.1 MBytes  84.9 Mbits/sec   62   79.2 KBytes
[  5]  13.00-14.00  sec  10.2 MBytes  86.0 Mbits/sec   40   28.3 KBytes
[  5]  14.00-15.00  sec  8.88 MBytes  74.4 Mbits/sec   65   66.5 KBytes
[  5]  15.00-16.00  sec  10.0 MBytes  83.9 Mbits/sec   21    103 KBytes
[  5]  16.00-17.00  sec  9.88 MBytes  82.8 Mbits/sec   97   84.8 KBytes
[  5]  17.00-18.00  sec  10.6 MBytes  89.1 Mbits/sec   98   82.0 KBytes
[  5]  18.00-19.00  sec  10.4 MBytes  87.0 Mbits/sec   66    102 KBytes
[  5]  19.00-20.00  sec  9.38 MBytes  78.6 Mbits/sec   71   59.4 KBytes
[  5]  20.00-21.00  sec  10.2 MBytes  86.0 Mbits/sec   36   86.3 KBytes
[  5]  21.00-22.00  sec  10.2 MBytes  86.0 Mbits/sec   76   80.6 KBytes
[  5]  22.00-23.00  sec  10.1 MBytes  84.9 Mbits/sec   78   60.8 KBytes
[  5]  23.00-24.00  sec  10.2 MBytes  86.0 Mbits/sec   57   96.2 KBytes
[  5]  24.00-25.00  sec  10.4 MBytes  87.0 Mbits/sec   78   84.8 KBytes
[  5]  25.00-26.00  sec  10.0 MBytes  83.9 Mbits/sec   44   82.0 KBytes
[  5]  26.00-27.00  sec  10.8 MBytes  90.2 Mbits/sec   20    110 KBytes
[  5]  27.00-28.00  sec  9.38 MBytes  78.6 Mbits/sec  124   59.4 KBytes
[  5]  28.00-29.00  sec  7.00 MBytes  58.7 Mbits/sec   50   1.41 KBytes
[  5]  29.00-30.00  sec  10.0 MBytes  83.9 Mbits/sec  108   99.0 KBytes
[  5]  30.00-31.00  sec  9.25 MBytes  77.6 Mbits/sec  102   63.6 KBytes
[  5]  31.00-32.00  sec  10.2 MBytes  86.0 Mbits/sec   47    102 KBytes
[  5]  32.00-33.00  sec  9.75 MBytes  81.8 Mbits/sec   52    119 KBytes
[  5]  33.00-34.00  sec  9.75 MBytes  81.8 Mbits/sec  152   63.6 KBytes
[  5]  34.00-35.00  sec  10.2 MBytes  86.0 Mbits/sec   15   96.2 KBytes
[  5]  35.00-36.00  sec  9.25 MBytes  77.6 Mbits/sec   85   56.6 KBytes
[  5]  36.00-37.00  sec  10.0 MBytes  83.9 Mbits/sec   55   84.8 KBytes
[  5]  37.00-38.00  sec  8.75 MBytes  73.4 Mbits/sec   45    249 KBytes
[  5]  38.00-39.00  sec  9.88 MBytes  82.8 Mbits/sec  324   94.7 KBytes
[  5]  39.00-40.00  sec  10.4 MBytes  87.0 Mbits/sec   51   83.4 KBytes
[  5]  40.00-41.00  sec  10.1 MBytes  84.9 Mbits/sec   42   89.1 KBytes
[  5]  41.00-42.00  sec  10.6 MBytes  89.1 Mbits/sec   76   83.4 KBytes
[  5]  42.00-43.00  sec  10.4 MBytes  87.0 Mbits/sec   85    103 KBytes
[  5]  43.00-44.00  sec  10.5 MBytes  88.1 Mbits/sec   22   86.3 KBytes
[  5]  44.00-45.00  sec  9.62 MBytes  80.7 Mbits/sec   84   76.4 KBytes
[  5]  45.00-46.00  sec  10.2 MBytes  86.0 Mbits/sec   62    102 KBytes
[  5]  46.00-47.00  sec  9.12 MBytes  76.5 Mbits/sec  116   62.2 KBytes
[  5]  47.00-48.00  sec  10.2 MBytes  86.0 Mbits/sec   66   74.9 KBytes
[... 747 lines cut ...]
CONFIG_PACKAGE_bnx2x-firmware=m
CONFIG_PACKAGE_brcmfmac-firmware-usb=m
CONFIG_PACKAGE_carl9170-firmware=m
CONFIG_PACKAGE_e100-firmware=m
CONFIG_PACKAGE_mwifiex-pcie-firmware=m
CONFIG_PACKAGE_mwl8k-firmware=m
CONFIG_PACKAGE_r8152-firmware=m
CONFIG_PACKAGE_r8169-firmware=m
CONFIG_PACKAGE_rs9113-firmware=m
CONFIG_PACKAGE_rt2800-pci-firmware=m
CONFIG_PACKAGE_rt2800-usb-firmware=m
CONFIG_PACKAGE_rtl8192ce-firmware=m
CONFIG_PACKAGE_rtl8192cu-firmware=m
CONFIG_PACKAGE_rtl8192de-firmware=m
CONFIG_PACKAGE_rtl8192se-firmware=m
CONFIG_PACKAGE_rtl8821ae-firmware=m
CONFIG_PACKAGE_wil6210-firmware=m
CONFIG_PACKAGE_wl12xx-firmware=m
CONFIG_PACKAGE_wl18xx-firmware=m
# CONFIG_VERSION_CODE_FILENAMES is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-16:
You have pre-N client on same +/-2 channel, Set channel to auto and hopefully you get rid of them

castillofrancodamian, 2024-04-16:
@brada4 Which device you named is "pre-N"?

brada4, 2024-04-16:
Wifi-B Wifi-G or just some other noise like baby radio sets.

nightcore500, 2024-04-16:
@brada4 I can rule out inteferences as described in the post. I have already tested the router and one of the affected devices in an interference-free environment. (tested using SDR) It also makes no difference which channel I go to.

What I forgot to mention:
There are devices that work without any problems. For example, an older Surface Go 3 works perfectly with full bandwidth in TX and RX. Even retries can hardly be measured here. 

The theory that pre-N devices cause interference also does not fit with the fact that the problem disappears as soon as the connected client switches to standby.

brada4, 2024-04-16:
One legacy wifi client disables short-gi for whole radio etc.
Also AP has much better antenna and is expected to slow from much further competing signals.
You mention some Wifi4 1-antenna devices and > 72Mbps. 40MHz in 2.4 works faster than 20MHz only in greenfield conditions.

nightcore500, 2024-04-17:
@brada4 Have you even read the report? This is not about poor radio conditions or insufficient bandwidths. 
This is about massive retries in the TX path. The resulting lower bandwidth is just the side effect I mentioned. 
Please take a look at the iperf3 tests. Especially the test with the heading "Iperf3 test on latest source - TX path with standby:".

Logically, I only did the tests with one client logged in at a time. No matter which of the devices. They were always the only client connected to the 2.4 GHz module.
I also use neither 40MHz nor the Greenfield mode. More channel bandwidth does not make the problem any better and cannot be the solution. Sure, the data transmission increases a little. 
But that's not the point. The massive retries still remain.

I don't have these retries with the affected devices with other 2.4 GHz chipsets on AP side. (I have already mentioned this) Same conditions, same location, same clients, same tested channel, same channel bandwidth, same configuration. Two completely different results.
I know the 2.4 GHz spectrum is heavily loaded these days and should always be treated with caution. However, I mentioned that I have already tested this behavior in an interference-free environment.

The Surface Pro 3, which does not show any problems, is also only a Wifi 4 client.

nightcore500, 2024-04-18:
Here are some more iperf3 tests. This time with the Oneplus 10 Pro (AX client). High retries occur in TX. If the device is put into standby, the bandwidth stabilizes and the retries disappear. 
The device was once again the only connected client in this test. The test environment was also inteference-free again. (Spectrum checked via SDR)
Same behavior as with the other affected devices.

station dump:

> Station 30:bb:7d:0e:7d:91 (on phy0-ap0)
>         inactive time:  10 ms
>         rx bytes:       642313311
>         rx packets:     501770
>         tx bytes:       2251983894
>         tx packets:     1842752
>         tx retries:     153633
>         tx failed:      153751
>         rx drop misc:   418
>         signal:         -44 [-48, -46, -53, -55] dBm
>         signal avg:     -43 [-47, -46, -52, -52] dBm
>         tx bitrate:     286.7 MBit/s HE-MCS 11 HE-NSS 2 HE-GI 0 HE-DCM 0
>         tx duration:    102436198 us
>         rx bitrate:     258.0 MBit/s HE-MCS 10 HE-NSS 2 HE-GI 0 HE-DCM 0
>         rx duration:    44037308 us
>         last ack signal:-44 dBm
>         avg ack signal: -42 dBm
>         airtime weight: 256
>         authorized:     yes
>         authenticated:  yes
>         associated:     yes
>         preamble:       short
>         WMM/WME:        yes
>         MFP:            no
>         TDLS peer:      no
>         DTIM period:    2
>         beacon interval:100
>         short preamble: yes
>         short slot time:yes
>         connected time: 1137 seconds
>         associated at [boottime]:       98680.914s
>         associated at:  1713357370753 ms
>         current time:   1713358506903 ms

        
        
Iperf3 test on latest source - TX path:

> Accepted connection from 192.168.1.150, port 45614
> [  5] local 192.168.1.1 port 5201 connected to 192.168.1.150 port 45626
> [ ID] Interval           Transfer     Bitrate         Retr  Cwnd
> [  5]   0.00-1.00   sec  13.1 MBytes   110 Mbits/sec  118    188 KBytes
> [  5]   1.00-2.00   sec  2.62 MBytes  22.0 Mbits/sec  105   72.1 KBytes
> [  5]   2.00-3.00   sec  4.75 MBytes  39.8 Mbits/sec   25   49.5 KBytes
> [  5]   3.00-4.00   sec  4.00 MBytes  33.6 Mbits/sec   28   31.1 KBytes
> [  5]   4.00-5.00   sec  5.00 MBytes  41.9 Mbits/sec   36   50.9 KBytes
> [  5]   5.00-6.00   sec  8.50 MBytes  71.3 Mbits/sec   15    103 KBytes
> [  5]   6.00-7.00   sec  8.75 MBytes  73.4 Mbits/sec   79   73.5 KBytes
> [  5]   7.00-8.00   sec  8.12 MBytes  68.2 Mbits/sec    0    133 KBytes
> [  5]   8.00-9.00   sec  4.75 MBytes  39.8 Mbits/sec   68   58.0 KBytes
> [  5]   9.00-10.00  sec  5.12 MBytes  43.0 Mbits/sec   51   41.0 KBytes
> [  5]  10.00-11.00  sec  4.00 MBytes  33.6 Mbits/sec    7   60.8 KBytes
> [  5]  11.00-12.00  sec  4.00 MBytes  33.6 Mbits/sec   10   72.1 KBytes
> [  5]  12.00-13.00  sec  4.75 MBytes  39.8 Mbits/sec   36   55.1 KBytes
> [  5]  13.00-14.00  sec  7.38 MBytes  61.9 Mbits/sec    3    112 KBytes
> [  5]  14.00-15.00  sec  11.0 MBytes  92.3 Mbits/sec  129   89.1 KBytes
> [  5]  15.00-16.00  sec  5.25 MBytes  44.0 Mbits/sec   53   60.8 KBytes
> [  5]  16.00-17.00  sec  4.62 MBytes  38.8 Mbits/sec   24   56.6 KBytes
> [  5]  17.00-18.00  sec  4.00 MBytes  33.6 Mbits/sec   12   69.3 KBytes
> [  5]  18.00-19.00  sec  4.75 MBytes  39.8 Mbits/sec    7   69.3 KBytes
> [  5]  19.00-20.00  sec  4.00 MBytes  33.6 Mbits/sec   40   60.8 KBytes
> [  5]  20.00-21.00  sec  2.00 MBytes  16.8 Mbits/sec    1   65.0 KBytes
> [  5]  21.00-22.00  sec  4.12 MBytes  34.6 Mbits/sec   12   59.4 KBytes
> [  5]  22.00-23.00  sec  3.88 MBytes  32.5 Mbits/sec    7   74.9 KBytes
[... 123 lines cut ...]
>         option type 'mac80211'
>         option path 'platform/soc/18000000.wifi+1'
>         option band '5g'
>         option htmode 'HE20'
>         option cell_density '0'
>         option country 'DE'
>         option channel '120'
>         option txpower '16'
>         option disabled '1'
> 
> config wifi-iface 'wifinet0'
>         option device 'radio0'
>         option mode 'ap'
>         option ssid 'test'
>         option encryption 'psk2+ccmp'
>         option key 'XXXXXXXXXXXX'
>         option network 'lan'
>         option disassoc_low_ack '0'
> 
> config wifi-iface 'wifinet2'
>         option device 'radio0'
>         option mode 'monitor'
>         option ssid 'test'
>         option network 'lan'
>

brada4, 2024-04-18:
Srsly?
option txpower '12'
option txpower '16'

brada4, 2024-04-18:
But disparity pretty much explains difference between 5 and 2.4 ....

nightcore500, 2024-04-18:
@brada4 
Can we please stay on topic? The fact that the problems disappear as soon as the clients go into standby while an iperf3 test is running and thus the frequent frames with the power management flag set also disappear has absolutely nothing to do with any interference from other outdated B/G devices or insufficient tx performance.
It also has absolutely nothing to do with the fact that I also have n clients / non AX clients in the list of affected devices.
From a purely logical point of view, all these comments make no sense in connection with the problems described...

Despite all this, here are some more iperf3 tests with increased transmission power:

nightcore500, 2024-04-18:
@brada4 
Can we please stay on topic? The fact that the problems disappear as soon as the clients go into standby while an iperf3 test is running and thus the frequent frames with the power management flag set also disappear has absolutely nothing to do with any interference from other outdated B/G devices or insufficient tx performance.
It also has absolutely nothing to do with the fact that I also have n clients / non AX clients in the list of affected devices.
From a purely logical point of view, all these comments make no sense in connection with the problems described...

Despite all this, here are some more iperf3 tests with increased TX power. The high retries are not decreasing:
16dBm TX power:

> Accepted connection from 192.168.1.215, port 37342
> [  5] local 192.168.1.1 port 5201 connected to 192.168.1.215 port 37344
> [ ID] Interval           Transfer     Bitrate         Retr  Cwnd
> [  5]   0.00-1.00   sec  8.50 MBytes  71.2 Mbits/sec  266    151 KBytes
> [  5]   1.00-2.00   sec  9.12 MBytes  76.5 Mbits/sec  140    106 KBytes
> [  5]   2.00-3.00   sec  9.12 MBytes  76.5 Mbits/sec  121   80.6 KBytes
> [  5]   3.00-4.00   sec  9.75 MBytes  81.8 Mbits/sec   31   99.0 KBytes
> [  5]   4.00-5.00   sec  9.50 MBytes  79.7 Mbits/sec   62   74.9 KBytes
> [  5]   5.00-6.00   sec  8.62 MBytes  72.4 Mbits/sec   74   55.1 KBytes
> [  5]   6.00-7.00   sec  9.88 MBytes  82.8 Mbits/sec    0    136 KBytes
> [  5]   7.00-8.00   sec  9.25 MBytes  77.6 Mbits/sec  152   84.8 KBytes
> [  5]   8.00-9.00   sec  9.88 MBytes  82.8 Mbits/sec   63    105 KBytes
> [  5]   9.00-10.00  sec  9.25 MBytes  77.6 Mbits/sec   68    107 KBytes
> [  5]  10.00-11.00  sec  9.12 MBytes  76.5 Mbits/sec   76   65.0 KBytes
> [  5]  11.00-12.00  sec  9.12 MBytes  76.5 Mbits/sec   53   76.4 KBytes
> [  5]  12.00-13.00  sec  9.50 MBytes  79.7 Mbits/sec   37    106 KBytes
> [  5]  13.00-14.00  sec  8.62 MBytes  72.4 Mbits/sec   97   73.5 KBytes
> [  5]  14.00-15.00  sec  9.25 MBytes  77.6 Mbits/sec   57   77.8 KBytes
> [  5]  15.00-16.00  sec  9.25 MBytes  77.6 Mbits/sec   65   83.4 KBytes
> [  5]  16.00-17.00  sec  9.12 MBytes  76.5 Mbits/sec   72   82.0 KBytes
> [  5]  17.00-18.00  sec  8.75 MBytes  73.4 Mbits/sec   33   91.9 KBytes
> [  5]  18.00-19.00  sec  9.12 MBytes  76.5 Mbits/sec   18   59.4 KBytes
> [  5]  19.00-20.00  sec  9.12 MBytes  76.5 Mbits/sec   52   86.3 KBytes
> [  5]  20.00-21.00  sec  9.12 MBytes  76.5 Mbits/sec   26   97.6 KBytes
> [  5]  21.00-22.00  sec  8.00 MBytes  67.1 Mbits/sec   81   55.1 KBytes
> [  5]  22.00-23.00  sec  9.75 MBytes  81.8 Mbits/sec   30   89.1 KBytes
> [  5]  23.00-24.00  sec  9.25 MBytes  77.6 Mbits/sec   46    123 KBytes
> [  5]  24.00-25.00  sec  9.25 MBytes  77.6 Mbits/sec  108   74.9 KBytes
> [  5]  25.00-26.00  sec  9.38 MBytes  78.6 Mbits/sec   21    113 KBytes
> [  5]  26.00-27.00  sec  8.62 MBytes  72.3 Mbits/sec   48   79.2 KBytes
> [  5]  27.00-28.00  sec  8.62 MBytes  72.4 Mbits/sec   54   73.5 KBytes
> [  5]  28.00-29.00  sec  9.88 MBytes  82.8 Mbits/sec   29    105 KBytes
> [  5]  29.00-30.00  sec  8.00 MBytes  67.1 Mbits/sec   64   66.5 KBytes
> - - - - - - - - - - - - - - - - - - - - - - - - -
> [ ID] Interval           Transfer     Bitrate         Retr
> [  5]   0.00-30.00  sec   274 MBytes  76.5 Mbits/sec  2044             sender

20dBm TX power:

> Accepted connection from 192.168.1.215, port 37376
> [  5] local 192.168.1.1 port 5201 connected to 192.168.1.215 port 37378
> [ ID] Interval           Transfer     Bitrate         Retr  Cwnd
> [  5]   0.00-1.00   sec  11.9 MBytes  99.5 Mbits/sec    0    355 KBytes
> [  5]   1.00-2.00   sec  10.0 MBytes  83.9 Mbits/sec  359    148 KBytes
> [  5]   2.00-3.00   sec  10.5 MBytes  88.1 Mbits/sec  110    102 KBytes
> [  5]   3.00-4.00   sec  9.00 MBytes  75.5 Mbits/sec   59    106 KBytes
> [  5]   4.00-5.00   sec  9.00 MBytes  75.5 Mbits/sec  108   76.4 KBytes
> [  5]   5.00-6.00   sec  9.62 MBytes  80.7 Mbits/sec   18   94.7 KBytes
> [  5]   6.00-7.00   sec  9.00 MBytes  75.5 Mbits/sec   68   83.4 KBytes
> [  5]   7.00-8.00   sec  9.88 MBytes  82.8 Mbits/sec   57    116 KBytes
> [  5]   8.00-9.00   sec  9.88 MBytes  82.8 Mbits/sec   71    120 KBytes
> [  5]   9.00-10.00  sec  9.75 MBytes  81.8 Mbits/sec   92   66.5 KBytes
> [  5]  10.00-11.00  sec  9.75 MBytes  81.8 Mbits/sec   22   91.9 KBytes
> [  5]  11.00-12.00  sec  9.00 MBytes  75.5 Mbits/sec   88   77.8 KBytes
> [  5]  12.00-13.00  sec  9.00 MBytes  75.5 Mbits/sec   51   83.4 KBytes
> [  5]  13.00-14.00  sec  10.5 MBytes  88.1 Mbits/sec   36    105 KBytes
> [  5]  14.00-15.00  sec  9.00 MBytes  75.5 Mbits/sec   58   73.5 KBytes
> [  5]  15.00-16.00  sec  9.75 MBytes  81.8 Mbits/sec   86   66.5 KBytes
> [  5]  16.00-17.00  sec  10.4 MBytes  87.0 Mbits/sec   46    106 KBytes
> [  5]  17.00-18.00  sec  9.00 MBytes  75.5 Mbits/sec   76   70.7 KBytes
> [  5]  18.00-19.00  sec  9.88 MBytes  82.8 Mbits/sec   51    115 KBytes
> [  5]  19.00-20.00  sec  10.8 MBytes  90.2 Mbits/sec   87   84.8 KBytes
> [  5]  20.00-21.00  sec  8.88 MBytes  74.4 Mbits/sec   87   77.8 KBytes
> [  5]  21.00-22.00  sec  10.5 MBytes  88.1 Mbits/sec   53    122 KBytes
> [  5]  22.00-23.00  sec  9.75 MBytes  81.8 Mbits/sec  107   59.4 KBytes
> [  5]  23.00-24.00  sec  7.75 MBytes  65.0 Mbits/sec  268    103 KBytes
> [  5]  24.00-25.00  sec  9.75 MBytes  81.8 Mbits/sec    0    163 KBytes
> [  5]  25.00-26.00  sec  9.88 MBytes  82.8 Mbits/sec  216   66.5 KBytes
> [  5]  26.00-27.00  sec  9.75 MBytes  81.8 Mbits/sec   43    109 KBytes
> [  5]  27.00-28.00  sec  9.88 MBytes  82.8 Mbits/sec   79    105 KBytes
> [  5]  28.00-29.00  sec  8.38 MBytes  70.3 Mbits/sec   83   72.1 KBytes
> [  5]  29.00-30.00  sec  9.00 MBytes  75.5 Mbits/sec   50    221 KBytes
> - - - - - - - - - - - - - - - - - - - - - - - - -
> [ ID] Interval           Transfer     Bitrate         Retr
> [  5]   0.00-30.00  sec   289 MBytes  80.8 Mbits/sec  2529             sender

brada4, 2024-04-18:
Bluetooth?

nightcore500, 2024-04-18:
@brada4

nightcore500, 2024-04-18:
@brada4 You mean directly from the client concerned?
When I first noticed the problem, I also suspected that it might be due to the client's bluetooth.
Due to the identical behavior of the devices, I did most of the tests on the HTC. Deactivating the Bluetooth on the HTC did not change the situation. 
I have just tested the behavior again on the Oneplus. The Oneplus has very few retries after deactivating Bluetooth. In addition, the bandwidth has just stabilized.
In effect, like when switching to standby mode. So you can actually see a change here.
I have just cross-checked the HTC and the Oneplus again with other APs with the 2.4GHz chipsets QCA9563, MT7603E and MT7976C. The first two are only BGN capable. The third is an AX module.
The radio configuration was again identical. These problems do not occur here despite active bluetooth on the client.
The QCA9563 AP ran with OpenWRT. The other two Mediathek APs, however, ran with the manufacturer's own firmware. (UBNT Unifi AP AC Lite and Unifi U6+)

nightcore500, 2024-04-18:
I have to correct myself. If I operate the affected AP with OpenWRT release 23.05.2 / 23.05.3, it makes no difference whether I deactivate or activate Bluetooth on the client. The behavior is identical on the HTC and the Oneplus.
My current main branch build with the latest mt76 source from a few days ago behaves differently. Here the bandwidth stabilizes on both clients and the retries are significantly less as soon as Bluetooth is deactivated on the client.

brada4, 2024-04-18:
There is hidden location from bluetooth option in android. You know those pesky wireless speakers....

nightcore500, 2024-04-18:
Don't get me wrong. The clients have of course never been connected to any bluetooth device at any time. 
If Bluetooth is indeed responsible, it can only be the client itself. 
Otherwise the problem would logically not disappear if the bluetooth is only deactivated on the client.
Then only the transmission of the bluetooth beacon frames from the client can be responsible. Or am I missing something?
And this tiny frame sent at intervals (of about 100ms if I'm not wrong?) is supposed to be responsible for over 2000 retries over a period of 30 seconds? hm...

Unfortunately, it also does not explain the behavior with other 2.4GHz modules on the AP side. 
None of the chipsets I have tested have such problems with the same clients.
It only occurs with the MT7975N.

castillofrancodamian, 2024-04-18:
@nightcore500 The other user doesn't really know what to answer.

brada4, 2024-04-19:
12dBm  is transmit power of class 2 bluetoith  device. You do not have to be connected anywhere, issue is not specific to SoC but to low power transmission in presence of bt scan. Not detectable by your SDR because screens were off while you measured.

DragonBluep, 2024-04-19:
Did you select `40 MHz` BW and `Force 40MHz` mode in the WiFi page?

nightcore500, 2024-04-19:
@brada4 

> screens were off while you measured.

I don't understand now? The SDR is not connected to the client if that's what you mean. I carried out the measurement on a separate laptop. Why are you bringing up the 12dBm tx power again? Have you overlooked the latest iperf3 tests? Even with 20dBm tx power, absolutely nothing improves the situation. I also tested the other 2.4 GHz modules on the AP side with "only" 12 dBm tx power. Even a test over 600 seconds shows just 200 retries with the clients concerned and the other 2.4GHz chipsets. With active Bluetooth on the client side! But here we are talking about well over 2000 within 30-60 seconds.

By the way, I have 3 more Banana Pi R3 routers here. All of the 4 APs with the MT7975N chipset show this behavior.

@DragonBluep No, I only operate the AP at 20 MHz in the 2.4GHz band.

brada4, 2024-04-19:
Ans any no-banana routers? Like 2.4 only.

nightcore500, 2024-04-19:
@brada4 The other 2.4GHz chipsets tested are in non-banana pi routers.

brada4, 2024-04-19:
I tested in vario&s other combinations of mt76 old and new but not banana, laptop intel rtl ath9k, 2 mobiles qca screen on off, wakelock, power cable etc no impact from full (-: speed :-) bt transfer 10 to 50 
cm away.  
deeper log like lvl1 - link speed change? Powersave? 
survey dump ( tried, no change from any power tricks.

nightcore500, 2024-04-19:
@brada4 Thank you very much for your effort! 
I could ask the community in the banana pi forum to do some tests on their banana pis. The successor router (Banana Pi R4) uses the same 2.4 GHz module according to the data sheet. The problem should therefore also occur with this router if it is due to the module or its driver.

degen91, 2024-04-23:
@nightcore500 MT7976N on GL.iNet MT6000 using mt76 has (what I believe to be) a [similar (same?) issue](https://github.com/openwrt/mt76/issues/859). It was noticed that the 2.4 GHz issue is related to mobile devices and might be related to power saving modes, as you suggested. Intel AX20x/21x do not experience the issue. It has also been reported that this issue is worsened by distance, much more-so than would be expected, including when signal strength is intact.

nightcore500, 2024-04-23:
Many thanks @degen91 for this hint

oxavelar, 2024-11-22:
Can you disable AQL on the 2.4 GHz band as an experiment?

```echo 0 > /sys/kernel/debug/ieee80211/phy0/aql_enable```

NOTE: Choose your 2.4 GHz phy, my MT6000 has that in phy0. My excessive retries and woes are gone by this.