openwrt/openwrt #15180: cc1: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier

zcracker, 2024-04-17:
### Describe the bug

MKDIR    staticobjs
  CC       staticobjs/bpf.o
  CC       staticobjs/btf.o
In file included from btf.c:18:
/home/zvleaks/dev/Puzzle-M901/staging_dir/target-aarch64_cortex-a72_musl/usr/include/gelf.h:86:9: error: unknown type name 'Elf64_Relr'
   86 | typedef Elf64_Relr GElf_Relr;
      |         ^~~~~~~~~~
cc1: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
make[4]: *** [Makefile:134: staticobjs/btf.o] Error 1
make[4]: Leaving directory '/home/zvleaks/dev/Puzzle-M901/build_dir/target-aarch64_cortex-a72_musl/libbpf-1.4.0/src'
make[3]: *** [Makefile:75: /home/zvleaks/dev/Puzzle-M901/build_dir/target-aarch64_cortex-a72_musl/libbpf-1.4.0/.built] Error 2
make[3]: Leaving directory '/home/zvleaks/dev/Puzzle-M901/package/libs/libbpf'
time: package/libs/libbpf/compile#0.77#0.33#1.10
    ERROR: package/libs/libbpf failed to build.
make[2]: *** [package/Makefile:129: package/libs/libbpf/compile] Error 1
make[2]: Leaving directory '/home/zvleaks/dev/Puzzle-M901'
make[1]: *** [package/Makefile:123: /home/zvleaks/dev/Puzzle-M901/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zvleaks/dev/Puzzle-M901'
make: *** [/home/zvleaks/dev/Puzzle-M901/include/toplevel.mk:233：world] 错误 2

### OpenWrt version

mian

### OpenWrt release

mian

### OpenWrt target/subtarget

mvebu

### Device

Puzzle-M901

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

zcracker, 2024-04-17:
Main Line
clean code repository
 supported device Puzzle-M901

Y-ZD-heheka, 2024-04-17:
make[4]: Entering directory '/home/yzd/Desktop/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/libbpf-1.4.0/src'
  CC       staticobjs/btf.o
In file included from btf.c:18:
/home/yzd/Desktop/immortalwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/gelf.h:86:9: error: unknown type name 'Elf64_Relr'
   86 | typedef Elf64_Relr GElf_Relr;
      |         ^~~~~~~~~~
cc1: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
make[4]: *** [Makefile:134: staticobjs/btf.o] Error 1
make[4]: Leaving directory '/home/yzd/Desktop/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/libbpf-1.4.0/src'
make[3]: *** [Makefile:75: /home/yzd/Desktop/immortalwrt/build_dir/target-aarch64_cortex-a53_musl/libbpf-1.4.0/.built] Error 2