openwrt/openwrt #15184: Problem disassociated and constantly disconnect Wifi AX9000 802.11R use

rdlvm, 2024-04-17:
### Describe the bug

Problem disassociated Wifi AX9000 802.11R use

Wi-Fi disconnections, and constantly this warning/error in the openwrt log:

Tue Apr 16 08:55:44 2024 daemon.notice hostapd: phy2-ap0: AP-STA-DISCONNECTED
Tue Apr 16 08:55:44 2024 daemon.info hostapd: phy2-ap0: STA IEEE 802.11: disassociated
Tue Apr 16 08:55:45 2024 daemon.info hostapd: phy2-ap0: STA IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

It causes all my devices to constantly disconnect and connect (I have smart plugs, echo, firetv, etc...) and they constantly lose connection 2.4G and 5G

### OpenWrt version

r25946-9693ed6a9e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX9000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rdlvm, 2024-04-17:
@Ansuel @robimarko

oded106e, 2024-06-19:
@Ansuel @robimarko

robimarko, 2024-06-19:
Sadly, I do not know the cause of this issue.

rany2, 2024-06-19:
FWIW I don't have this issue on mt76 drivers, so it's likely a driver/firmware issue.

closed 2024-06-29 by rdlvm