openwrt/openwrt #15185: Problem disassociated and constantly disconnect Wifi AX3600 802.11R use

rdlvm, 2024-04-17:
### Describe the bug

Problem disassociated Wifi AX3600 802.11R use

Wi-Fi disconnections, and constantly this warning/error in the openwrt log:

Tue Apr 16 08:55:44 2024 daemon.notice hostapd: phy2-ap0: AP-STA-DISCONNECTED
Tue Apr 16 08:55:44 2024 daemon.info hostapd: phy2-ap0: STA IEEE 802.11: disassociated
Tue Apr 16 08:55:45 2024 daemon.info hostapd: phy2-ap0: STA IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

It causes all my devices to constantly disconnect and connect (I have smart plugs, echo, firetv, etc...) and they constantly lose connection 2.4G and 5G

### OpenWrt version

r25946-9693ed6a9e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rdlvm, 2024-04-17:
@Ansuel @robimarko

robimarko, 2024-04-17:
Why 2 same issues?

rdlvm, 2024-04-17:
> Why 2 same issues?

Same problem on different router. Better to unify?

pwned-pixel, 2024-04-28:
Did you just enable ieee80211r or did you do your own settings and/or do you have tools running like dawn? How does your config actually look like. There are a lot of settings which kind a fk up fast roaming:
https://openwrt.org/docs/guide-user/network/wifi/basic (scroll down).
I remember those issues during development phase on AX3600. Where I had to set the mobility domain manually (you can calculate it for your SSID) and to disable ft_psk_generate_local (it is enabled by default) using WPA3 only.
Maybe setting WPA2/WPA3 mixed mode solves your issue already?