openwrt/openwrt #15192: kernel: status of 6.6 version

namiltd, 2024-04-17:
### Describe the bug

- [ ] apm821xx
- [ ] archs38
- [ ] armsr
- [ ] at91
- [ ] ath25
- [ ] ath79
- [ ] bcm27xx
- [ ] bcm47xx
- [ ] bcm4908
- [ ] bcm53xx
- [ ] bcm63xx
- [ ] bmips
- [ ] gemini
- [ ] imx
- [ ] ipq40xx
- [ ] ipq806x
- [ ] ixp4xx
- [ ] kirkwood
- [ ] lantiq
- [ ] layerscape
- [ ] malta
- [ ] mediatek
- [ ] mpc85xx
- [ ] mvebu
- [ ] mxs
- [ ] octeon
- [ ] octeontx
- [ ] omap
- [ ] oxnas
- [ ] pistachio
- [ ] qoriq
- [ ] qualcommax
- [ ] ramips
- [ ] realtek
- [ ] rockchip
- [ ] sifiveu
- [ ] sunxi
- [ ] tegra
- [ ] uml
- [ ] x86
- [ ] zynq

### OpenWrt version

*

### OpenWrt release

*

### OpenWrt target/subtarget

*

### Device

*

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

castillofrancodamian, 2024-04-17:
Checkboxes should not have a check mark if they only have that kernel version as a test. Yes, it should have the check mark when it is already the default version.

CHKDSK88, 2024-04-17:
layerscape: PR ready https://github.com/openwrt/openwrt/pull/14997

tmn505, 2024-05-15:
tegra: http://lists.openwrt.org/pipermail/openwrt-devel/2024-May/042822.html (or if someone finds it more readable, link to patchwork: [tegra: kernel 6.6 introduction](https://patchwork.ozlabs.org/project/openwrt/cover/20240515171133.692414-1-tmn505@terefe.re) )

namiltd, 2024-05-16:
> tegra: http://lists.openwrt.org/pipermail/openwrt-devel/2024-May/042822.html (or if someone finds it more readable, link to patchwork: [tegra: kernel 6.6 introduction](https://patchwork.ozlabs.org/project/openwrt/cover/20240515171133.692414-1-tmn505@terefe.re) )

Generate PR for your changes so that others can take a look at them.

Gateworks, 2024-05-21:
imx PR ready for review #15540

tmn505, 2024-05-29:
> @tmn505 Generate PR for your changes (https://github.com/tmn505/openwrt/commits/tegra-next) so that others can take a look at them.

I won't send duplicate adding only noise. I assume most of reviewers/users are subscribed to ML and can reply to the message from their inbox. For those not subscribed, they can browse openwrt-devel archives, click the link with sender address (right under the message title) and well behaving mail client will open with `In-Reply-To` field filled in. Copy the lines from the patch you want to comment on and after sending, it will be indexed as reply to the patch.

PS v2 of tegra bump sent: http://lists.openwrt.org/pipermail/openwrt-devel/2024-May/042906.html

GoetzGoerisch, 2024-06-07:
bmips: https://github.com/openwrt/openwrt/pull/15653

lorand-horvath, 2024-06-17:
Report for ath79/generic: custom build (LuCI included) increased in size from `6337 kB` (build 26593 / kernel 6.1) to `6465 kB` (build 26657 / kernel 6.6). That's an absolute increase of only `128 kB` from 6.1 to 6.6, very nice!

namiltd, 2024-06-17:
It's not much, but in some situations even 64kb can run out of memory during kernel loading and cause LZMA ERROR 1. One solution is to add the lzma -d22 option

lorand-horvath, 2024-06-18:
The switch from kernel 6.1 to 6.6 broke the rtl8366s switch support on WNDRMAC v2 (WNDR3800). Probably removing too many patches/backports...
Can someone please take a look at this issue? Reported here: https://github.com/openwrt/openwrt/issues/15739

namiltd, 2024-07-04:
@Noltari Pinning this tag makes it much easier to find information about which targets already have kernel 6.6 or at what stage their implementation is.
Thanks to it, people can see which targets still need to be done and which ones need help,

Noltari, 2024-07-04:
> @Noltari Pinning this tag makes it much easier to find information about which targets already have kernel 6.6 or at what stage their implementation is. Thanks to it, people can see which targets still need to be done and which ones need help,

@namiltd my bad, it was completely unintentional...

ascendbeing, 2024-07-06:
Ramips/Rt3883 based f9k1103/f9k1109 has non working 2.4ghz first wifi on 2nd to latest snapshot. I haven't tried latest snapshot. Also, the 2nd Wi-Fi allegedly has a changeable non fixed band, but in snapshot and in 23.05-snapshot, it no longer will do anything if you change it to 2ghz.

namiltd, 2024-07-06:
@ascendbeing Create a separate issue, attach the log from the serial console.

dangowrt, 2024-07-08:
> Ramips/Rt3883 based f9k1103/f9k1109 has non working 2.4ghz first wifi on 2nd to latest snapshot.

We will need much more information to be able to address this issue. For starters, please submit
 - complete bootlog of an older build with working WiFi
 - complete bootlog of snapshot with no longer working WiFi
 - dump of `/etc/config/wireless` from both builds (do *not* keep configuration)
 - output of `iw list` from both builds

lorand-horvath, 2024-07-14:
`ath79/generic` target doesn't build at all since July 9: https://downloads.openwrt.org/snapshots/targets/ath79/generic/
Does anyone know what is going on with it?

ascendbeing, 2024-07-14:
> We will need much more information to be able to address this issue. For starters, please submit
>
>     * complete bootlog of an older build with working WiFi
>
>     * complete bootlog of snapshot with [...]

I will commit to getting you all that, probably later tonight. I'll do this for both F9K1103 and F9K1109 since I have both. They run off the same sysupgrade file which is curiously only named after 1 of them. It may be edifying to have the full logs for both, for both 23.05-SNAPSHOT (working) and snapshot (broken 2.4, last I tried).
1 of them is bricked, which i expect to fix via web UI recovery. Happened cause I tried to load the other's config, Mac addresses and all. If that doesn't unbrick, it'll be for 1 model only.

lorand-horvath, 2024-07-15:
`ath79/generic` target doesn't build at all since July 9: https://downloads.openwrt.org/snapshots/targets/ath79/generic/
It's stuck on snapshot 26899, but a custom build returns an even older version 26870, which is very strange.
Is there something specifically wrong with that target?

lorand-horvath, 2024-07-16:
@aparcar @hauke @namiltd Do you know why `ath79/generic` stopped building anymore? It's been more than a week, see comment above. Thanks guys!

lorand-horvath, 2024-07-16:
@aparcar @hauke @namiltd @DragonBluep Do you know why `ath79/generic` stopped building anymore? It's been more than a week, see comment above. Thanks guys!

DragonBluep, 2024-07-16:
> Do you know why `ath79/generic` stopped building anymore? It's been more than a week, see comment above. Thanks guys!

I guess it is because the built bot has run out of the memory or some processes have timed out while packaging the image builder. Some users are starting to complain that the image size is too large (https://github.com/openwrt/openwrt/issues/15950) after https://github.com/openwrt/openwrt/pull/12959. Unfortunately, `ath79/generic` is the most popular target. It has 360+ supported devices. Disabling `ramdisk` feature should be a workaround.

And please open another issue. It is not related to the 6.6 kernel.

namiltd, 2024-08-02:
The only things left are:
- at91
- omap
- qoriq
- realtek
- starfive

ascendbeing, 2024-08-02:
@tmn505 https://github.com/openwrt/openwrt/issues/16054 This isn't what I said id provide in full but this is what happening. I think I did save lots of this if necessary for context. There isn't a lot of relevant lines tho.

namiltd, 2024-08-07:
The only targets left are:
- ~~at91~~
- ~~omap~~
- ~~qoriq~~ 
- realtek [[*]](https://github.com/openwrt/openwrt/pull/16052)
- starfive

There is not much left and it will be possible to completely remove support for kernels 5.15 and 6.1.

peterwillcn, 2024-08-14:
Thank you everyone, congratulations！

namiltd, 2024-08-14:
Or maybe it's worth waiting for 6.11, which may also be LTS?

castillofrancodamian, 2024-08-14:
There is no time for a release this year.

T-X, 2024-08-15:
@namiltd 
> Or maybe it's worth waiting for 6.11, which may also be LTS?

I'm a fan of release early, release often. My (wishful) thinking would be, if 6.11 became LTS and patches in OpenWrt were ported accordingly, just do another, timely OpenWrt release with that, too :-).

lorand-horvath, 2024-08-15:
6.6 is fine, it just needs releasing. I bet on a 24.09 or 24.10 release.

namiltd, 2024-08-15:
> @namiltd And could you please fix this issue? https://forum.openwrt.org/t/docker-pull-fails-failed-to-register-layer-operation-not-supported/138253/59?u=g-m0n3y-2503

Offtopic, make a new issue

ascendbeing, 2024-08-15:
@namiltd 6.6 isn't okay for some ramips/rt3883 target. The 2.4ghz phy0 on 6.6 for Belkin f9k1103 and f9k1109 (same chips, different flash/RAM parts) is broken on 6.6 and theres a spammy clock warning each boot. 5.15, no such issue.

namiltd, 2024-08-15:
@ascendbeing I can only tell if something is working properly based on reported errors.

ascendbeing, 2024-08-15:
@namiltd 2.4ghz phy0 for Belkin f9k1103 and f9k1109 (same chips, different flash/RAM parts) is broken on 6.6, along with a spammy clock warning each boot.
I have an issue filed on this, with some awesome people helping, thankfully.
Maybe it's isolated cases, but those ramips/rt3883 work properly on 5.15 not 6.6. I didn't see any mention in your assessment of what is or isn't working.

ascendbeing, 2024-08-15:
@namiltd https://github.com/openwrt/openwrt/issues/16054 sorry I deleted the original reply cause it was a bit silly.

namiltd, 2024-08-15:
I see that DragonBluep has already addressed this issue, so I hope it will be resolved soon.

ascendbeing, 2024-08-17:
It's a little further afield than "push dragons pull" tho. 2.4ghz being a no-op (phy0, pcie chip) on 6.6 is, afaict, still unsolved, for ramips/rt3883-based Belkin f9k110[3,9]. 5.15 works fine.

Neustradamus, 2024-08-17:
This is an important ticket, hope that all will be good for the next OpenWrt release.

olumolu, 2024-08-30:
6.6 uses more ram in routers.

olumolu, 2024-08-31:
> > 6.6 uses more ram in routers.
> 
> This is not an excuse for stagnation.

I did not say i want to keep old kernel i use a rolling distro which is called fedora.
But i think the hugh memory can be reduced with polish.

tapper82, 2024-08-31:
> > > 6.6 uses more ram in routers.
> > 
> > 
> > This is not an excuse for stagnation.
> 
> I did not say i want to keep old kernel i use a rolling distro which is called fedora. But i think the hugh memory can be reduced with polish.

We would love to see some PRs from you to lower the memory usage.

olumolu, 2024-08-31:
> > > > 6.6 uses more ram in routers.
> > > 
> > > 
> > > This is not an excuse for stagnation.
> > 
> > 
> > I did not say i want to keep old kernel i use a rolling distro which is called fedora. But i think the hugh memory can be reduced with polish.
> 
> We would love to see some PRs from you to lower the memory usage.

I would love to but i don't have knowledge about kernel stuff sorry Alas!

lorand-horvath, 2024-09-05:
@namiltd @aparcar @robimarko @hauke There's something very wrong with the snapshot builders, almost nothing builds for the last couple of days: https://buildbot.openwrt.org/images/#/grid
Also https://firmware-selector.openwrt.org and ASU both fail when requesting custom builds.

Can someone take a look, please? Thanks!

namiltd, 2024-09-05:
It looks like there are problems with downloading and communicating with the git server.

However, please post here only posts related to this topic, and for off-topic posts, please open new threads.

robimarko, 2024-09-07:
@lorand-horvath It seems to have fixed itself magically on the buildbot side, so ASU should pick those new SDK-s once they are built.

lorand-horvath, 2024-09-08:
@robimarko Can you point to the SDK-s in question ? ASU is broken, please see https://github.com/openwrt/asu/issues/961

robimarko, 2024-09-08:
> @robimarko Can you point to the SDK-s in question ? ASU is broken, please see [openwrt/asu#961](https://github.com/openwrt/asu/issues/961)

You said that buildbots were broken, and they were, now its sorted but ASU seems unrelated to that and to this thread.

namiltd, 2024-09-20:
Kernel 6.6 migration done.

closed 2024-09-20 by namiltd

dannil, 2024-09-20:
> > Kernel 6.6 migration done.
> 
> When is the release date for the stable version? We can't wait.

The last update was http://lists.openwrt.org/pipermail/openwrt-devel/2024-August/043087.html so I assume branching in October and then release after a number of release candidates has passed validation, _if_ everything goes according to plan.

Neustradamus, 2024-09-20:
Congrats to all contributors for this :)

@namiltd: Good job!

GoetzGoerisch, 2024-09-21:
@namiltd could we please start a collection of 6.6 bugs?
From a short search

 e.g.
#16396
#15857
#15615
#15313

namiltd, 2024-09-21:
Currently, all errors in the snapshot concern kernel 6.6 because there is no other one.
Another problem is that Issues contains over 2300 reports. Many of them are outdated,

GoetzGoerisch, 2024-09-21:
I was not precise enough, I was thinking about blocking bugs for a release of 24.xx.

peterwillcn, 2024-09-22:
Kernel 6.12  PREEMPT_RT

olumolu, 2024-09-22:
> Kernel 6.12 PREEMPT_RT for a release of 25.xx.

What is the benifit of having this rt stuff

perrin4869, 2024-10-01:
I noticed the current snapshots for rockchip seem to be missing many modules such as mt76 and ath11k, despite those being available for 23.05 series. Are the modules going to be added for the next release?