openwrt/openwrt #15203: tools/cpio fails to build

cryobry, 2024-04-19:
### Describe the bug

### Describe the bug

Trying to compile HEAD of 23.05 branch:

```
checking return type of major()... configure: error: cannot determine type
make[3]: *** [Makefile:19: /home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src/build_dir/host/cpio-2.14/.configured] Error 1
make[3]: Leaving directory '/home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src/tools/cpio'
time: tools/cpio/compile#0.78#0.80#1.49
    ERROR: tools/cpio failed to build.
make[2]: *** [tools/Makefile:226: tools/cpio/compile] Error 1
make[2]: Leaving directory '/home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src'
make[1]: *** [tools/Makefile:222: /home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src/staging_dir/host/stamp/.tools_compile_nyyyyyynnnyyynynyyyyyyynyyynnynyyynynnyyyyyyyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src'
make: *** [/home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src/include/toplevel.mk:232: world] Error 2
~/bin/openwrtbuilder

```

`test x$ac_cv_rettype_major = x` is passing so $ac_cv_rettype_major must be empty during the build.

### OpenWrt version

r0+23835-9b33b74ef7

### OpenWrt release

 23.05.3

### OpenWrt target/subtarget

rockchip/armv8

### Device

friendlyarm_nanopi-r4s

### Image kind

Self-built image

### Steps to reproduce

```
./scripts/feeds update -i -f &&
./scripts/feeds update -a -f &&
./scripts/feeds install -a -f
(I modify .config manually here)
make -j1 V=s defconfig download clean world
```

### Actual behaviour

cpio configure fails: checking return type of major()... configure: error: cannot determine type

### Expected behaviour

cpio builds correctly

### Additional info

Fedora 40, gcc 14.0.1

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

closed 2024-04-21 by cryobry

ErikPelli, 2024-05-07:
I'm experiencing the same issue (`checking return type of major()... configure: error: cannot determine type`) with Fedora 40 and gcc 14.0.1 (same environment).
I'm trying to build `ramips/mt7621` using the sources.

cryobry, 2024-05-07:
Are you using HEAD of 23.05 branch (or snapshot) and not the tag? It's
fixed for me in HEAD.

ErikPelli, 2024-05-07:
I was using the tag, by the way I solved with fedora 39, thank you anyway

talpachen, 2024-06-01:
see here https://github.com/openwrt/openwrt/commit/82ba1d26357fe005da6a71c10f75d83227af299c