openwrt/openwrt #15204: GnuPG signing key for v23.05.x missing

biliwala, 2024-04-19:
### Describe the bug

The key used to sign the stable release of v23.05.x is missing from the [documentation](https://openwrt.org/docs/guide-user/security/signatures) and [keyring](https://git.openwrt.org/keyring.git), without which we cannot verify the downloaded `sha256sum` file. 

The same thing happened in [v22.03.0](/openwrt/openwrt/issues/9814#issuecomment-1117030972).

@ynezz @jow- At your convenience, would you please fix this problem? Thank you for your assistance.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2024-04-19:
> The key used to sign the stable release of v23.05.x is missing from the [documentation](https://openwrt.org/docs/guide-user/security/signatures) and [keyring](https://git.openwrt.org/keyring.git)

Yes, this one is on me. I want to improve the situation around signing keys and start [Using Nitrokey 3A Mini for build artifact signing key storage](https://openwrt.org/docs/guide-developer/releases/provision-nitrokey3).

I just need to finish an artifact signing REST API service which would allow access from multiple buildbot masters to this USB dongle (For that I'm looking for 3-4 days in a row, which I currently don't have, unfortunately).

I understand that its taking me ages, so meanwhile we should perhaps consider doing it the old way and generate 23.05 keys and replace it with the key from USB dongle when its ready.

> without which we cannot verify the downloaded sha256sum file.

Its being signed, you can verify it with the https://openwrt.org/docs/guide-user/security/signatures#pgp_key_for_unattended_snapshot_builds (and perhaps we should just make it obvious, that 23.05 releases are temporarily signed with this key?)

```shell
$ wget https://downloads.openwrt.org/releases/23.05.3/targets/armsr/armv8/sha256sums
$ wget https://downloads.openwrt.org/releases/23.05.3/targets/armsr/armv8/sha256sums.asc
$ gpg --recv-keys 0xCD84BCED626471F1
$ gpg --with-fingerprint --verify sha256sums.asc sha256sums
gpg: Signature made Sat 23 Mar 2024 06:55:40 AM UTC
gpg:                using RSA key 6D9278A33A9AB3146262DCECF93525A88B699029
gpg: Good signature from "OpenWrt Build System (PGP key for unattended snapshot builds) <pgpsign-snapshots@openwrt.org>" [unknown]
gpg:                 aka "LEDE Build System (LEDE GnuPG key for unattended build jobs) <lede-adm@lists.infradead.org>" [unknown]
```

ynezz, 2024-11-09:
We've just finished switching [all buildmasters to the new key](https://lists.openwrt.org/pipermail/openwrt-devel/2024-November/043361.html), so from now on, everything should be signed with the [Universal GPG signing key from Nitrokey 3A key](https://openwrt.org/docs/guide-user/security/signatures#universal_gpg_signing_key_from_nitrokey_3a_key).

closed 2024-11-09 by ynezz