openwrt/openwrt #15207: netifd: new packet steering script slow down wan port

pesa1234, 2024-04-19:
### Describe the bug

The new packet steering script on Gl.Inet MT6000 slow down the wan speed on ppoe connection to 1.3gb/s
Old packet steering script no issue

### OpenWrt version

r25967-ff814d7c76

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Self-built image

### Steps to reproduce

Enable on wan ppoe connection with link speed more than 1gb/s
Activate packet steering (new script) --> https://github.com/openwrt/openwrt/commit/a205a5734eda4604a19b6cb0f65909ef69730699

### Actual behaviour

Speedtest with new script of packet steering can not reach more than 1.3 gb/s.

[image]
One core reach 100% with routing on ppoe

### Additional info

Reverting to old script and enabling it solve the slow speed WAN port.

This is what I've done for MT6000
https://github.com/openwrt/openwrt/commit/1aa1b9ad26fbb71400e7e4c57455d74f4dfc8aa6

and for luci config
https://github.com/openwrt/luci/commit/f9891ce20bf6555eecd9468cbdead42f04129298

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_glinet_gl-mt6000=y
CONFIG_DEVEL=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CROND_SPECIAL_TIMES=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SYSLOG_INFO=y
CONFIG_BUSYBOX_CONFIG_FEATURE_TOP_INTERACTIVE=y
CONFIG_BUSYBOX_CONFIG_PKILL=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_HTOP_LMSENSORS=y
CONFIG_IPK_FILES_CHECKSUMS=y
CONFIG_KEEPALIVED_CHECKSUM_COMPAT=y
CONFIG_KEEPALIVED_FWMARK=y
CONFIG_KEEPALIVED_IP6TABLES=y
[... 325 lines cut ...]
CONFIG_PACKAGE_wpad-basic-openssl=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_STRIP_KERNEL_EXPORTS=y
CONFIG_TARGET_OPTIMIZATION="-O2 -pipe -mcpu=cortex-a53"
CONFIG_TARGET_OPTIONS=y
CONFIG_TESTING_KERNEL=y
# CONFIG_PACKAGE_dnsmasq_full_dhcpv6 is not set
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Xploder, 2024-04-19:
Can confirm that the new packet steering is far worse with PPPoE compared to the old script on MT-6000 Snapshot

nbd168, 2024-04-19:
I will run some tests, maybe there is a problem with flow offloading and PPPoE

brada4, 2024-04-19:
Yep. Need to relocate mss fix down as in fw4 git head and 
https://github.com/openwrt/firewall4/pull/26 and still watch conntrack -E if each packet does not bounce offload out and in.

nbd168, 2024-04-23:
I've done a lot of performance measurements related to this issue on my MT7622 board.
I think the main issue is not really the load distribution, but the fact that tons of CPU cycles are being wasted on segmenting large packets assembled by GRO. So while one core only lightly loaded by the receive path, the other core that gets the steered packets is overloaded by having to perform data copy + checksum on most of the packets it gets.
Based on that, I believe the proper solution is to stick with the current script (since it significantly reduces scheduling overhead), and to get rid of the GRO/GSO overhead. I'm almost done with the kernel patches for this, my preliminary tests show very promising results.

nbd168, 2024-04-23:
Finished the change for kernels 6.1 and 6.6 in my staging tree at https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=summary
Please test.

pesa1234, 2024-04-23:
> Finished the change for kernels 6.1 and 6.6 in my staging tree at https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=summary Please test.

Yes sure I'll let you know. Thanks

pesa1234, 2024-04-23:
> Finished the change for kernels 6.1 and 6.6 in my staging tree at https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=summary Please test.

[image]
 and the link speed hangs to : 1129.10 Mbps

brada4, 2024-04-23:
Can you add cpubdetail so we see pink softirq ie nft/qdisc out of red anything in kernel?

brada4, 2024-04-23:
> segmenting large packets assembled by GRO. 

I thought its just dev_weight

pesa1234, 2024-04-23:
> Can you add cpubdetail so we see pink softirq ie nft/qdisc out of red anything in kernel?

I didn't understand your question

rany2, 2024-04-23:
> > Can you add cpubdetail so we see pink softirq ie nft/qdisc out of red anything in kernel?
> 
> I didn't understand your question

In htop, go to "Setup (F2)" > (Left sidebar) Display options > Global options > Enable "Detailed CPU time"

nbd168, 2024-04-24:
@pesa1234 until now I had not realized you were running speedtest on the device directly. That explains why my changes did not help at all in your test. My changes significantly improve routing performance, but leave local-rx unaffected.
Going back to the approach of the previous packet steering script will improve your local speedtest case, but might make routing performance worse. I could change my script to include a mode similar to the previous one, but I'm not sure how useful that would be in practice

brada4, 2024-04-24:
@pesa1234  You can reduce load from local traffic slightly
vi /usr/share/firewall4/templates/ruleset.uc
iifname "lo" -> iif "lo"
oifname "lo" -> oif "lo"

Like in https://github.com/openwrt/firewall4/commit/a5553dae70439c7e4fa910490fcf12a1ffff5bd2
takes out interface enumeration before conntrack engine. this has no impact on forwarded traffic, only on local.

brada4, 2024-04-24:
```
cp  /usr/share/firewall4/templates/ruleset.uc  /usr/share/firewall4/templates/ruleset.uc_old
vi  /usr/share/firewall4/templates/ruleset.uc
/iifname
position cursor over "name" part and delete characters with x
repeat for oif(name)
Esc:wq
root@openwrt # fw4 check
r@o# service firewall restart
```
Not primary use of "router" btw, thus slightly unoptimal.

brada4, 2024-04-24:
@pesa1234 also worth unhiding kernel threads 
[image]
(TUF-AX6000 doing gigabit from wan to wifi using fast.com no offloads)

nbd168, 2024-04-24:
Another thing you can try for local speedtest is this: Check if the load distribution or performance changes if you write `256` to `/sys/class/net/eth0/queues/rx-0/rps_flow_cnt` and `/sys/class/net/eth1/queues/rx-0/rps_flow_cnt`

Xploder, 2024-04-24:
This is with the new changes from @nbd168 's staging repo, same build as @pesa1234 is using and just a 500mbit WAN load (not local). It's definitely not working as intended on MT6000.

Loads are virtually the same as without packet steering.

[image]

pesa1234, 2024-04-24:
> Another thing you can try for local speedtest is this: Check if the load distribution or performance changes if you write `256` to `/sys/class/net/eth0/queues/rx-0/rps_flow_cnt` and `/sys/class/net/eth1/queues/rx-0/rps_flow_cnt`

`echo 256 > /sys/class/net/eth1/queues/rx-0/rps_flow_cnt `--> 2.5gbps
[image]

brada4, 2024-04-24:
Do the firewall thing for local tests. Should half pink part avoiding needless per-packet examination.

nbd168, 2024-04-24:
@Xploder those results still matches the intention of the new script. The script tries to make one CPU responsible for NAPI polling, and one for the rest of the packet processing. Different CPUs are assigned for different devices.
The relatively high load for 500 Mbps is not necessarily an indication of how much traffic it is able to handle that way. The effects of GRO are highly nonlinear. When significantly below the CPU load limit, you typically see only very few packets processed per polling interval, so scheduling overhead absolutely dominates. When reaching the limit, each poll is able to aggregate more packets, drastically reducing the per-packet CPU overhead.
What I found in cases with very high CPU load is that this strategy typically performs better than spreading load across more CPUs, since it reduces scheduling overhead when more devices (ethernet, WLAN, etc.) are involved.

brada4, 2024-04-24:
I cannot squeeze any anomaly with new script, just bounces 40-60% all CPUs, like with old one, disabling steering altogether 1st core is a bit more loaded, but no difference in throughput or latency.

brada4, 2024-04-24:
> same CPU

same core (the one with 2 hyperthreads, whatever the modern name is) better, shares uppermost caches, does not bounce cached pages aside etc.

nbd168, 2024-04-24:
I just pushed some changes to the packet steering script in my staging tree.
If you set `option packet_steering 2`, it will steer packets to all CPUs (like the old script did).
If you set `option steering_flows 128`, it will steer up to 128 locally received flows per device to the CPU receiving them. (using `rps_flow_cnt`, which I posted earlier).

pesa1234, 2024-04-25:
> I just pushed some changes to the packet steering script in my staging tree. If you set `option packet_steering 2`, it will steer packets to all CPUs (like the old script did). If you set `option [...]

Thank you! In the next days I'll let you know.
To set-up in Luci this is the small improvement https://github.com/openwrt/luci/commit/30377780f8844780b99b17928e3b834dacff085e

nbd168, 2024-04-25:
Please import the updated versions of those commits from my staging tree - I fixed some typo regressions in the code. I will push the changes to main soon.

brada4, 2024-04-25:
If you have queues > nproc the "other queue" is sometimes on same CPU. Seems nginx reuseport uses just 1st tx queue per CPU. No idea though if hitting same CPU twice is cause of asymmetry.

pesa1234, 2024-04-26:
https://github.com/openwrt/openwrt/commit/98834a4c3f81c6e4f20329ff266f9bd85731d114 - https://github.com/openwrt/openwrt/commit/b5c53848c3792355a2dd4ff0e06bf4d230e1f5fb - https://github.com/openwrt/openwrt/commit/c4d394c6cc0facba94fda5f1cc886edaf342be3a and https://github.com/openwrt/openwrt/commit/7ebcf2fb9c5db8b75666761f5e767c91abb1f437 solve the issue and works as expected, close.
Thanks @nbd168

closed 2024-04-26 by pesa1234

Djfe, 2025-01-17:
further things just got improved on main and on the release branch ☺️
https://github.com/openwrt/openwrt/commit/67a4aeef6c63e5836bd6b82faf6cc15934869efb
https://github.com/openwrt/openwrt/commit/acce25b789d40f2d88a2326a7d9921bd1743fe54