openwrt/openwrt #15216: Build error when using external toolchain [hyperscan] [snort3]

graysky2, 2024-04-20:
### Describe the bug

When using an official external toolchain, building hyperscan or snort3 ends in failures.

### OpenWrt version

N/A

### OpenWrt release

N/A

### OpenWrt target/subtarget

x86/64

### Device

N/A

### Image kind

Self-built image

### Steps to reproduce

1. Configure the build system to use the official external toolchain for x86/64
2. Try to build the hyperscan package

### Actual behaviour

The build fails

### Expected behaviour

The build should complete without errors

### Additional info

```
% package/hyperscan/{clean,compile} package/index V=s
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/xtools-openwrt/toolchain-x86_64_gcc-13.2.0_musl/bin/x86_64-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/xtools-openwrt/toolchain-x86_64_gcc-13.2.0_musl/bin/x86_64-openwrt-linux-musl-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test ARCH_64_BIT
-- Performing Test ARCH_64_BIT - Success
-- Performing Test ARCH_32_BIT
-- Performing Test ARCH_32_BIT - Failed
-- Build type RELEASE
-- using release build
-- Boost version: 1.84.0
-- Found PythonInterp: /scratch/openwrt/staging_dir/host/bin/python (found version "3.11.8")
-- Build date: 2024-04-18
-- Building shared libraries
-- gcc version 13.2.0
CMake Error at CMakeLists.txt:198 (string):
  string begin index: -1 is out of range 0 - 0

CMake Error at CMakeLists.txt:209 (message):
  Something went wrong determining gcc tune: -mtune= not valid

-- g++ version 13.2.0
-- Looking for include file unistd.h
```

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DEVEL=y
CONFIG_BUILD_NLS=y
CONFIG_EXTERNAL_GCC_VERSION="13.2.0"
CONFIG_EXTERNAL_TOOLCHAIN=y
CONFIG_EXTERNAL_TOOLCHAIN_LIBC_USE_MUSL=y
CONFIG_LIBATOMIC_FILE_SPEC="./lib/libatomic.so.*"
CONFIG_LIBATOMIC_ROOT_DIR="/opt/xtools-openwrt/toolchain-x86_64_gcc-13.2.0_musl"
CONFIG_LIBC_FILE_SPEC="./x86_64-openwrt-linux-musl/lib/ld-musl-x86_64.so.1 ./x86_64-openwrt-linux-musl/lib/libc.so"
CONFIG_LIBC_ROOT_DIR="/opt/xtools-openwrt/toolchain-x86_64_gcc-13.2.0_musl"
CONFIG_LIBGCC_FILE_SPEC="./x86_64-openwrt-linux-musl/lib/libgcc_s.so ./x86_64-openwrt-linux-musl/lib/libgcc_s.so.1"
CONFIG_LIBGCC_ROOT_DIR="/opt/xtools-openwrt/toolchain-x86_64_gcc-13.2.0_musl"
CONFIG_LIBPTHREAD_FILE_SPEC="./lib/libpthread{-*.so,.so.*}"
CONFIG_LIBPTHREAD_ROOT_DIR="/opt/xtools-openwrt/toolchain-x86_64_gcc-13.2.0_musl"
CONFIG_LIBRT_FILE_SPEC="./lib/librt{-*.so,.so.*}"
CONFIG_LIBRT_ROOT_DIR="/opt/xtools-openwrt/toolchain-x86_64_gcc-13.2.0_musl"
CONFIG_LIBSTDCPP_FILE_SPEC="./x86_64-openwrt-linux-musl/lib/libstdc++.so ./x86_64-openwrt-linux-musl/lib/libstdc++.so.6 ./x86_64-openwrt-linux-musl/lib/libstdc++.so.6.0.32 ./x86_64-openwrt-linux-musl/lib/libstdc++.so.6.0.32-gdb.py"
CONFIG_LIBSTDCPP_ROOT_DIR="/opt/xtools-openwrt/toolchain-x86_64_gcc-13.2.0_musl"
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_OPTIMIZE_SPEED=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_gperftools-runtime=y
CONFIG_PACKAGE_hyperscan-runtime=y
CONFIG_PACKAGE_kmod-nfnetlink-queue=y
CONFIG_PACKAGE_kmod-nft-queue=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libdaq3=y
CONFIG_PACKAGE_libdnet=y
CONFIG_PACKAGE_libhwloc=y
CONFIG_PACKAGE_liblzma=y
CONFIG_PACKAGE_libnetfilter-queue=y
CONFIG_PACKAGE_libnfnetlink=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpciaccess=y
CONFIG_PACKAGE_libpcre=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libtirpc=y
CONFIG_PACKAGE_libunwind=y
CONFIG_PACKAGE_luajit=y
CONFIG_PACKAGE_snort3=y
CONFIG_PACKAGE_zlib=y
CONFIG_PCRE_JIT_ENABLED=y
CONFIG_TARGET_NAME="x86_64-openwrt-linux-musl"
CONFIG_TOOLCHAIN_BIN_PATH="./usr/bin ./bin"
CONFIG_TOOLCHAIN_INC_PATH="./usr/include ./include/fortify ./include"
CONFIG_TOOLCHAIN_LIBC="musl"
CONFIG_TOOLCHAIN_LIB_PATH="./usr/lib ./lib"
CONFIG_TOOLCHAIN_PREFIX="x86_64-openwrt-linux-musl-"
CONFIG_TOOLCHAIN_ROOT="/opt/xtools-openwrt/toolchain-x86_64_gcc-13.2.0_musl"
CONFIG_USE_EXTERNAL_LIBC=y
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

graysky2, 2024-04-20:
Bug against the build system/so openwrt versions do not apply.

brada4, 2024-04-20:
cmake tests host cpu as per commiters wish...
https://github.com/openwrt/packages/blame/master/libs/hyperscan/Makefile#L32

graysky2, 2024-04-21:
> Modify the hyperscan Makefile to delete lines [32 and 33](https://github.com/openwrt/packages/blob/master/libs/hyperscan/Makefile#L32-L33) as passing those args to CMake masks this error and introduces a new error

I get the error removing those two lines.