openwrt/openwrt #15218: procd: processing procd_add_reload_trigger on ifupdate event

brianjmurrell, 2024-04-20:
### Describe the bug

As has been detailed in tohojo/sqm-scripts#171, `procd` seems to be executing `config.change` events for `sqm` when there is an `ifupdate` event on a IPv6 WAN connection.  `/etc/config/sqm` has not been touched on this machine since Nov. so there simply cannot be any valid `config.change` events for it.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

x86/64

### Device

X86

### Image kind

Self-built image

### Steps to reproduce

Configure and connect a router to an IPv6 PPPoE ISP.

Configure SQM.

Wait and observe how many times you seem SQM reload events in the syslog.

Add a debugging script to /etc/hotplug.d/iface to log all events.

Observe how the SQM events align with `ifupdate` events on the WAN interface, even if SQM isn't configured for the interface getting the `ifupdate` events.

### Actual behaviour

`config.change` events being generated when the configuration for a given key has not changed.

### Expected behaviour

`config.change` events should not be generated when the configuration is not actually changing.

### Additional info

In my configuration, I have two WAN interfaces.  I have SQM configured on a cable (DHCP) connection and I also have a PPPoE WAN connection with both IPv4 and IPv6.  There is no SQM configured for that interface.

It is the `ifupdate` event on the PPPoE interface, that SQM is not even configured for that are causing these `config.change` events to be triggered.

### Diffconfig

```text
Self-built in this context means using ImageBuilder, not build-from-source.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2024-04-23:
Is this because `/etc/config/ucitrack` says that firewall changes affect sqm? An interface event would trigger firewall, which would in turn affect sqm?
```
config firewall
        option init 'firewall'
        list affects 'luci-splash'
        list affects 'qos'
        list affects 'miniupnpd'
        list affects 'sqm'
 ```

brianjmurrell, 2024-04-23:
> Is this because `/etc/config/ucitrack` says that firewall changes affect sqm?

I don't know TBH.  I am not familiar with _ucitrack_.  But I do not have firewall installed on the AP that is exhibiting this behaviour.

> An interface event would trigger firewall,

Does it, if _firewall_ is not even installed on the AP though?

dave14305, 2024-04-23:
Running SQM on an AP?? Maybe you need to explain your setup in more detail.

brianjmurrell, 2024-04-23:
Sorry.  I meant router.  I don't use the OpenWRT firewall package because I use a different firewall tool.

dave14305, 2024-04-23:
Try to remove the `list affects 'sqm'` line from `/etc/config/ucitrack` and see if it makes any difference.

brianjmurrell, 2024-04-25:
Once I have edited that file, is there anything I need to do/restart, etc. for that change to be effected?  I.e. normally you need to restart a service in `/etc/init.d/` once you make a UCI change.

dave14305, 2024-04-25:
I would reboot to be sure. Not sure how else to clear out the existing ubus service triggers.