openwrt/openwrt #15222: Framebuffer broken with kernel 6.6

oskarirauta, 2024-04-21:
### Describe the bug

After testing kernel 6.6 I found issues with Framebuffer drivers.
It is missing CONFIG_FB_DEVICE from config.

While at it, to support VoCore's usb display, also CONFIG_FB_VIRTUAL would be need to set. Both of these where present in kernel 6.1

So I'd like to propose that video support would set CONFIG_FB_DEVICE and CONFIG_FB_VIRTUAL

### OpenWrt version

r25942-12137cb460

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

Cerberus Interceptor

### Image kind

Self-built image

### Actual behaviour

/dev/fbX missing

### Expected behaviour

/dev/fbX present

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-08:
Did the same problem occur for kernel 6.1?

oskarirauta, 2024-05-14:
No it did not.

namiltd, 2024-05-16:
Add to add
```
CONFIG_FB_DEVICE=y
CONFIG_FB_VIRTUAL=y
```
to file target\linux\x86\64\config-6.6 and recompile the sources

oskarirauta, 2024-05-20:
I already have it there.. Was just wondering if it should be enabled in mainstream to provide functionality that there was before..

namiltd, 2024-05-20:
Generate a Pull Request for your changes (e.g. by editing target\linux\x86\64\config-6.6 directly on Github).

hauke, 2024-05-25:
Please test this PR: #15559 and report your test results in the PR.

commit 77bfe544 (2024-05-31): x86: Activate CONFIG_FB_DEVICE

This adds the legacy /dev/fb* device file for kernel 6.6 again.
Linux upstream commit 701d2054fa31 ("fbdev: Make support for
userspace interfaces configurable") made this configurable and we
deactivated this option by default for kernel 6.6. On x86 we are not
space constrained and some users need this legacy interface.

Fixes: #15222
PR #15559 (2024-05-31): x86: Activate CONFIG_FB_DEVICE
This adds the legacy /dev/fb* device file for kernel 6.6 again. Linux upstream commit 701d2054fa31 ("fbdev: Make support for userspace interfaces configurable") made this configurable and we deactivated this option by default for kernel 6.6. On x86 we are not space constrained and some users need this legacy interface.

Fixes: #15222