openwrt/openwrt #15223: update lua for mesa

esaaprillia, 2024-04-21:
### Describe the bug

please update lua package i want to update video package. the mesa package requires the latest lua

```
Run-time dependency lua54 found: NO (tried pkgconfig and cmake)
Run-time dependency lua53 found: NO (tried pkgconfig and cmake)
Run-time dependency lua52 found: NO (tried pkgconfig and cmake)
Dependency lua found: NO found 5.1.5 but need: '>=5.2'
Run-time dependency lua found: NO (tried pkgconfig and cmake)
Looking for a fallback subproject for the dependency lua54
Downloading lua source from https://www.lua.org/ftp/lua-5.4.6.tar.gz
Downloading lua patch from https://wrapdb.mesonbuild.com/v2/lua_5.4.6-4/get_patch
```

### OpenWrt version

none

### OpenWrt release

none

### OpenWrt target/subtarget

none

### Device

none

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2024-04-21:
Lua 5.3 is packaged, but in a suffixed manner. You need to figure out how to tell that package buildsystem to look for liblua5.3.so / lua5.3 executable respectively. Includes are staged in /usr/include/lua5.3/

esaaprillia, 2024-04-22:
@jow- 

i managed to build the latest version of mesa. but mesa will build lua if mesa doesn't find lua version 5.2. lua in openwrt version 5.1. here is the mesa log while building

```
2024-04-21T11:56:11.8714663Z make[2]: Entering directory '/home/runner/work/mosdns/mosdns/sdk/feeds/mosdns/libs/mesa'
2024-04-21T11:56:12.1743326Z touch /home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5/.prepared_bdb90bac31127ba45e3ddcbf07766045_6664517399ebbbc92a37c5bb081b5c53_check
2024-04-21T11:56:12.1768641Z . /home/runner/work/mosdns/mosdns/sdk/include/shell.sh; xzcat /home/runner/work/mosdns/mosdns/sdk/dl/mesa-24.0.5.tar.xz | tar -C /home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5/.. -xf -
2024-04-21T11:56:13.3706974Z [ ! -d ./src/ ] || cp -fpR ./src/. /home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5
2024-04-21T11:56:13.3769511Z 
2024-04-21T11:56:13.3770948Z Applying /home/runner/work/mosdns/mosdns/sdk/feeds/mosdns/libs/mesa/patches/001-fix-wayland-scanner-detect.patch using plaintext: 
2024-04-21T11:56:13.3811497Z patching file meson.build
2024-04-21T11:56:13.3813275Z Hunk #1 succeeded at 1987 (offset -2 lines).
2024-04-21T11:56:13.3815850Z patching file meson_options.txt
2024-04-21T11:56:13.4272509Z touch /home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5/.prepared_bdb90bac31127ba45e3ddcbf07766045_6664517399ebbbc92a37c5bb081b5c53
2024-04-21T11:56:13.4300686Z rm -f /home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5/.configured_*
2024-04-21T11:56:13.4328600Z rm -f /home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/stamp/.mesa_installed
2024-04-21T11:56:13.4384769Z /home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/sed -e "s|@CC@|'/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/gcc',|" -e "s|@CXX@|'/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/g++',|" -e "s|@PKGCONFIG@|/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/home/runner/work/mosdns/mosdns/sdk/staging_dir/hostpkg/bin/python3|" -e "s|@CFLAGS@|'-O2', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/hostpkg/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/host/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/hostpkg/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/host/include',|" -e "s|@CXXFLAGS@|'-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/hostpkg/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/host/include',|" -e "s|@LDFLAGS@|'-L/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/lib', '-L/home/runner/work/mosdns/mosdns/sdk/staging_dir/hostpkg/lib', '-L/home/runner/work/mosdns/mosdns/sdk/staging_dir/target-aarch64_generic_musl/host/lib',|" -e "s|@PREFIX@||" < /home/runner/work/mosdns/mosdns/sdk/staging_dir/host/lib/meson/openwrt-native.txt.in > /home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5/openwrt-native.txt
2024-04-21T11:56:13.4454862Z /home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/sed -e "s|@CC@|'aarch64-openwrt-linux-musl-gcc',|" -e "s|@CXX@|'aarch64-openwrt-linux-musl-g++',|" -e "s|@LD@|'bfd',|" -e "s|@AR@|aarch64-openwrt-linux-musl-gcc-ar|" -e "s|@STRIP@|aarch64-openwrt-linux-musl-strip|" -e "s|@NM@|aarch64-openwrt-linux-musl-gcc-nm|" -e "s|@PKGCONFIG@|/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/home/runner/work/mosdns/mosdns/sdk/staging_dir/hostpkg/bin/python3|" -e "s|@CFLAGS@|'-Os', '-pipe', '-mcpu=generic', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-ffile-prefix-map=/home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5=mesa-24.0.5', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-I/home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5/openwrt-build/src/broadcom/', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/toolchain-aarch64_generic_gcc-13.2.0_musl/usr/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/toolchain-aarch64_generic_gcc-13.2.0_musl/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/toolchain-aarch64_generic_gcc-13.2.0_musl/include/fortify', '-Wno-format', '-Wno-format-security',|" -e "s|@CXXFLAGS@|'-Os', '-pipe', '-mcpu=generic', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-ffile-prefix-map=/home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5=mesa-24.0.5', '-Wformat', '-Werror=format-security', '-fstack-protector', '-D_FORTIFY_SOURCE=1', '-Wl,-z,now', '-Wl,-z,relro', '-I/home/runner/work/mosdns/mosdns/sdk/build_dir/target-aarch64_generic_musl/mesa-24.0.5/openwrt-build/src/broadcom/', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/toolchain-aarch64_generic_gcc-13.2.0_musl/usr/include', '-I/home/runner/work/mosdns/mosdns/sdk/staging_dir/toolchain-aarch64_generic_gcc-13.2.0_musl/include [... cut ...]
2024-04-21T12:02:56.4090170Z [1366/1689] Compiling C object src/gallium/drivers/freedreno/libfreedreno.a.p/a4xx_fd4_query.c.o
2024-04-21T12:02:56.5868743Z [1367/1689] Compiling C object src/gallium/drivers/freedreno/libfreedreno.a.p/a4xx_fd4_rasterizer.c.o
2024-04-21T12:02:56.6912218Z [1368/1689] Compiling C object src/gallium/drivers/freedreno/libfreedreno.a.p/a4xx_fd4_resource.c.o
2024-04-21T12:02:56.8749162Z [1369/1689] Compiling C object src/gallium/drivers/freedreno/libfreedreno.a.p/a4xx_fd4_screen.c.o
2024

esaaprillia, 2024-04-22:
hy @jow- 

can you apply this patch. now mesa can find lua5.3

https://github.com/esaaprillia/openwrt/commit/9532083e7910e4d01cd0dd6ba6f45c4b037d25f8

```
Run-time dependency lua53 found: YES 5.3.5
```

esaaprillia, 2024-05-23:
@neheb 

Is there an option for how to determine the python path you want to use? This is a serious problem when building Mesa and other packages that require Python. the example package requires the gobject-introspection python path

neheb, 2024-05-23:
No. The path is fixed.

esaaprillia, 2024-05-23:
> No. The path is fixed.

https://github.com/openwrt/video/pull/34

esaaprillia, 2024-09-18:
@neheb @jefferyto 

i am trying to build python scipy. but fortran needs to be added in ```meson.mk``` can you add it

```
build_dir/target-aarch64_generic_musl/pypi/scipy-1.14.1/meson.build:84:0: ERROR: 'fortran' compiler binary not defined in cross file [binaries] section
```

https://github.com/scipy/scipy/issues/14812#issuecomment-1204896603

neheb, 2024-09-23:
You could make a PR.

esaaprillia, 2024-10-10:
@neheb 

I don't understand adding it

neheb, 2024-10-15:
Fortran requires CONFIG_INSTALL_GFORTRAN for your config AFAIK.

Also something like

```
--- a/include/meson.mk
+++ b/include/meson.mk
@@ -78,6 +78,7 @@ define Meson/CreateCrossFile
        $(STAGING_DIR_HOST)/bin/sed \
                -e "s|@CC@|$(foreach BIN,$(TARGET_CC),'$(BIN)',)|" \
                -e "s|@CXX@|$(foreach BIN,$(TARGET_CXX),'$(BIN)',)|" \
+               -e "s|@FORTRAN@|$(foreach BIN,$(TARGET_???),'$(BIN)',)|" \
                -e "s|@LD@|$(foreach FLAG,$(TARGET_LINKER),'$(FLAG)',)|" \
                -e "s|@AR@|$(TARGET_AR)|" \
                -e "s|@STRIP@|$(TARGET_CROSS)strip|" \
diff --git a/tools/meson/files/openwrt-cross.txt.in b/tools/meson/files/openwrt-cross.txt.in
index 836a0e51e4fb..d86aa7b056d0 100644
--- a/tools/meson/files/openwrt-cross.txt.in
+++ b/tools/meson/files/openwrt-cross.txt.in
@@ -3,6 +3,7 @@ c = [@CC@]
 c_ld = [@LD@]
 cpp = [@CXX@]
 cpp_ld = [@LD@]
+gfortran = [@FORTRAN@]
 ar = '@AR@'
 strip = '@STRIP@'
 nm = '@NM@'

```

A lot of the plumbing is missing unfortunately.

esaaprillia, 2024-10-16:
@neheb 

still getting errors while building

esaaprillia, 2024-10-16:
```
include $(TOPDIR)/rules.mk

PKG_NAME:=python-scipy
PKG_VERSION:=1.14.1
PKG_RELEASE:=1

PYPI_NAME:=scipy
PKG_HASH:=5a275584e726026a5699459aa72f828a610821006228e841b94275c4a7c08417

PKG_MAINTAINER:=Esa Aprilia Salsabila <esaapriliasalsabila@gmail.com>
PKG_LICENSE:=BSD-3-Clause
PKG_LICENSE_FILES:=LICENSE.txt

HOST_BUILD_DEPENDS:=python-setuptools-scm/host python-cython/host meson-python/host python-pybind11/host python-pythran
PKG_BUILD_DEPENDS:=python-setuptools-scm/host python-cython/host meson-python/host python-pybind11/host python-pythran

include $(TOPDIR)/feeds/packages/lang/python/pypi.mk
include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/host-build.mk
include $(TOPDIR)/feeds/packages/meson/meson.mk
include $(TOPDIR)/feeds/packages/lang/python/python3-package.mk
include $(TOPDIR)/feeds/packages/lang/python/python3-host-build.mk

define Package/python3-scipy
  SUBMENU:=Python
  SECTION:=lang
  CATEGORY:=Languages
  TITLE:=Fundamental algorithms for scientific computing in Python
  URL:=https://scipy.org
  DEPENDS:=+INSTALL_GFORTRAN:libgfortran +python3 +python3-numpy +openblas +python3-pooch
endef

define Package/python3-scipy/description
  SciPy (pronounced “Sigh Pie”) is an open-source software for mathematics, science, and engineering.
  It includes modules for statistics, optimization, integration, linear algebra, Fourier transforms, signal and image processing, ODE solvers
endef

ifneq ($(CONFIG_INSTALL_GFORTRAN),y)
MAKE_FLAGS += NOFORTRAN=1
endif

$(eval $(call Py3Package,python3-scipy))
$(eval $(call BuildPackage,python3-scipy))
$(eval $(call BuildPackage,python3-scipy-src))
$(eval $(call HostBuild))
```

Neustradamus, 2024-11-09:
It is possible to update Lua to 5.4.x?

At this time: 5.4.7 (2024-06-25):
- https://lua.org/news.html

5.3.x is EOL since 5.3.6 (2020-09-25).

esaaprillia, 2025-04-04:
@neheb @jow- 

can you update openssl to 3.4.1? there is one patch that needs to be updated

```
From 0000000000000000000000000000000000000000 Mon Sep 17 00:00:00 2001
From: Eneas U de Queiroz <cote2004-github@yahoo.com>
Date: Thu, 27 Sep 2018 08:29:21 -0300
Subject: Do not use host kernel version to disable AFALG

This patch prevents the Configure script from using the host kernel
version to disable building the AFALG engine on openwrt targets.

--- a/Configure
+++ b/Configure
@@ -1788,9 +1788,9 @@ $config{CFLAGS} = [ map { $_ eq '--ossl-strict-warnings'
 
 unless ($disabled{afalgeng}) {
     $config{afalgeng}="";
-    if (grep { $_ eq 'afalgeng' } @{$target{enable}}) {
+    if ($target =~ m/openwrt$/) {
         push @{$config{engdirs}}, "afalg";
-    } else {
+    } elsif (grep { $_ eq 'afalgeng' } @{$target{enable}}) {
         disable('not-linux', 'afalgeng');
     }
 }
```