openwrt/openwrt #15228: tools/cpio fails to configure

cryobry, 2024-04-21:
### Describe the bug

Trying to compile HEAD of 23.05 branch:

```
checking return type of major()... configure: error: cannot determine type
make[3]: *** [Makefile:19: /home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src/build_dir/host/cpio-2.14/.configured] Error 1
make[3]: Leaving directory '/home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src/tools/cpio'
time: tools/cpio/compile#0.78#0.80#1.49
    ERROR: tools/cpio failed to build.
make[2]: *** [tools/Makefile:226: tools/cpio/compile] Error 1
make[2]: Leaving directory '/home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src'
make[1]: *** [tools/Makefile:222: /home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src/staging_dir/host/stamp/.tools_compile_nyyyyyynnnyyynynyyyyyyynyyynnynyyynynnyyyyyyyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src'
make: *** [/home/bryan/.local/bin/openwrtbuilder/src/r4s/23.05.3-src/include/toplevel.mk:232: world] Error 2
~/bin/openwrtbuilder
```
`test x$ac_cv_rettype_major = x` indicates $ac_cv_rettype_major is empty.

### OpenWrt version

r0+23835-9b33b74ef7

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

rockchip/armv8

### Device

friendlyarm_nanopi-r4s

### Image kind

Self-built image

### Steps to reproduce

```
./scripts/feeds update -i -f &&
./scripts/feeds update -a -f &&
./scripts/feeds install -a -f
(I modify .config manually here)
make -j1 V=s defconfig download clean world
```

### Actual behaviour

cpio configure fails: checking return type of major()... configure: error: cannot determine type

### Expected behaviour

cpio builds correctly

### Additional info

Fedora 40, gcc 14.0.1

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-04-22:
Can you please try:
https://github.com/robimarko/openwrt/commit/82ba1d26357fe005da6a71c10f75d83227af299c

Updating to 2.15 fixes it for me on Fedora 40

cryobry, 2024-04-22:
Yes that fixes it, I also needed this b43-tools change for newer gcc: https://github.com/openwrt/openwrt/commit/0c96d20bf9fba6d814efa88c6fb08a5df094103a

robimarko, 2024-04-23:
> Yes that fixes it, I also needed this b43-tools change for newer gcc: [0c96d20](https://github.com/openwrt/openwrt/commit/0c96d20bf9fba6d814efa88c6fb08a5df094103a)

Can you please make an issue for that as well

robimarko, 2024-04-23:
Closing as fixed via https://github.com/openwrt/openwrt/commit/82ba1d26357fe005da6a71c10f75d83227af299c

commit 82ba1d26 (2024-04-23): tools/cpio: update to 2.15

Release Notes:
https://lists.gnu.org/archive/html/info-gnu/2024-01/msg00006.html

Fixes: #15228
PR #15240 (2024-04-23): [23.05] tools/cpio: update to 2.15
Cherry-pick from main.

Release Notes:
https://lists.gnu.org/archive/html/info-gnu/2024-01/msg00006.html

Fixes compilation on Fedora 40 with GCC14.
Fixes: #15228
closed 2024-04-23 by robimarko