openwrt/openwrt #15229: ipq40xx: ipq4019: whw03(v1): Unable to handle kernel paging request at virtual address [...]

Lanchon, 2024-04-22:
### Describe the bug

i had a transient, one time boot failure while working with an ipq4019 soc. the kernel panicked and the subsequent reboot succeeded.

this happened while porting a new device whw03 v1, but it seems to be a kernel issue for that platform unrelated to my efforts, which is why it might be of interest to someone; idk.

i captured the incident [here](https://gist.github.com/Lanchon/0c1e330617030f870d0a2910902fc4c3#file-whw03-v1-failed-boot-txt-L218-L299).

the commit upon which i am basing my work is in master:
1f3e0177d9 ("tools/bzip2: switch to cmake", 2024-04-03)

thanks!

### OpenWrt version

r25772+54-1f3e0177d9

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys WHW03 (Velop)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

LGA1150, 2024-04-22:
You may enable `CONFIG_KERNEL_KASAN` in `make menuconfig`, and see if it detects anything.

Lanchon, 2024-04-22:
i've booted an hundred times and only this one time it failed. smells like a race condition. so basically i can't reproduce. would CONFIG_KERNEL_KASAN help anyway?

Lanchon, 2024-04-22:
this seems to be a build problem, disregard, sorry!